home-assistant/operating-system #3943: Raspberry Pi 5 stuck after an OS update

sairon, 2025-03-18:
This issue serves to gather information and troubleshoot issues with updating of **Raspberry Pi 5 boards only**. Before reporting an issue, try the following:

1. Perform the update after a cold boot (unplugging the Pi from power before booting), ideally with an HDMI display connected.
2. If the update gets stuck, check the HDMI output and share the details below (camera capture is fine).
3. In the case of a stuck update, check what bootloader version is used - simply run the Pi without the boot media (SD card/NVMe) to leave it hanging in the bootloader.

One of the problems that people keep running into was fixed in OS 15.0. Unfortunately, due to the nature of the update method, it's not possible to fix that issue in the old OS version that the device is running currently. The above steps should mitigate the influence of the issue, and the report should also help us understand what exactly is wrong with your install if the cause is different.

> Updated HA OS from v. 14.2 to 15 on two RPi 5s (different locations but identical hardware). Following reboot initiated by update, the devices became unresponsive until a force reboot was initiated. Following the reboot, the devices reported OS 14.2
> 
> ### What operating system image do you use?
> 
> rpi5-64  (Raspberry Pi 5 64-bit OS)
> 
> ### What version of Home Assistant Operating System is installed?
> 
> 14.2
> 
> ### Did the problem occur after upgrading the Operating System?
> 
> Yes
> 
> ### Hardware details
> 
> Raspberry Pi 5 8GB inside generic fan enclosure
> Sonoff Zigbee 3.0 gateway attached using USB 2 extension cable
> Ethernet connection into network switch
> 
> ### Steps to reproduce the issue
> 
> 1. Update from OS 14.2 to OS 15
> 2. System does not correctly reboot and stays unresponsive indefinitely
> 3. Hard reboot Pi (power off, 10 seconds, power on)
> 4. Device reboots and OS 14.2 boots - upgrade to 15 still available
> 

 (Originally posted by @danners430 in #3937)

danners430, 2025-03-18:
Thanks @sairon for your efforts - apologies the last thread got so cluttered so rapidly... is there anything you'd like me to try this evening once I'm home to try and provide some insight?

UBvoigt, 2025-03-18:
I can confirm the same issue

sairon, 2025-03-18:
@danners430 Just follow the steps from the first post here. Hopefully it will resolve the problem or at least shed some light on it.

@UBvoigt ...

RobMeades, 2025-03-18:
Confirmed that this procedure worked for me (upgrading from v4.2 on an 8 gig Pi 5) : shut-down Home Assistant, pull the power from the Pi 5, re-apply power to the Pi 5, wait for Home Assistant start-up to complete, perform the upgrade, Home Assistant comes back all bright and breezy on v5.0.

Thanks for the advice!

sairon, 2025-03-18:
Found this one by Alan McKay in the Facebook group (I'm not sure if he's here on GH):

[image]

This shows there are some processes blocking the reboot (can be very well some misbehaving integration in the HA Core). Somewhat concerning is that the reboot doesn't proceed after Systemd requests the kernel to do so but if it's happening more often, it could also explain why some devices end up stuck after updating.

Anyway, this shows how important it is to have a display connected when troubleshooting such issues. It's either that or a crystal ball (which I don't have).

iyanmv, 2025-03-18:
> Confirmed that this procedure worked for me [upgrading from v4.2 on an 8 gig Pi 5], as opposed to the previous time: shut-down Home Assistant, pull the power from the Pi 5, re-apply power to the Pi 5, wait for Home Assistant start-up to complete, perform the upgrade, Home Assistant comes back all bright and breezy on v5.0.
> 
> Thanks for the advice!

Did you update from the web interface or via ssh? In my case, after doing that and trying to update from the web left the RPi5 unreachable again

> Anyway, this shows how important it is to have a display connected when troubleshooting such issues. It's either that or a crystal ball (which I don't have).

Yeah... but understand that not everyone has an extra display to connect a device that is supposed to work without any display at all.

Mika-999, 2025-03-18:
I have RPi5 and faced this issue after version OS14.0. I were not able to update any later versions including the 15.0 today without reboot. After I made cold reboot and then tried again the update from 14.0 to 15.0 it work!

Core 2025.3.3
Supervisor 2025.03.3
Operating System 15.0
Frontend 20250306.0

Sparkeeh, 2025-03-18:
I can confirm that I was having that issue and that updating via WebUI after a cold reboot worked perfectly to fix it.

RobMeades, 2025-03-18:
@iyanmv 

> Did you update from the web interface or via ssh? In my case, after doing that and trying to update from the web left the RPi5 unreachable again

Via the web interface worked in my case, i.e. having freshly booted seemed to do the trick.

mattbyte, 2025-03-18:
Upgrade after cold boot also worked for my RPi5. Could a warning be added to the release notes? This is what the WebUI links to when it offers the update, it might save some people the confusion.

yumbrad, 2025-03-18:
In my case (RPi5, 14.2 -> 15..0 headless via WebUI attempted), it didn't come back up. Plugged in display and rebooted a few times and it fell back to 14.2 slot. Then retried upgrade, and it booted to 15.0 successfully. So I moved it back to my network closet, but it didn't come up then. So I moved back to my desktop/display, and it came up. Then tried without display at my desktop, didn't come up - plugged in display without rebooting and got nothing. Then rebooted w/display a number of times and it comes up randomly half the time, the other times it boots but goes to emergency console. Can't seem to tie it to anything like having display / keyboard connected - just randomly works only half the time.. Switching my boot-slot back to 14.2 for now since I can't get 15.0 running reliably.

appliko, 2025-03-18:
Local web interface, shutdown, power off, power on, wait for everything online, then update from 14.2.
No issues here.

puttemanp, 2025-03-18:
> Local web interface, shutdown, power off, power on, wait for everything online, then update from 14.2. No issues here.

Something that shouldn't be necessary and moreover impossible for a lot of HA instances, either because of distance (in my case, almost 1600Km away), or because unreachable with a screen and keyboard....

DaSNyB, 2025-03-18:
It really feels like the foundation has been making some questionable decisions lately when it comes to development. 🤨 Just look at the disastrous rollout of the new Zigbee, the removal of the action, and the baffling return process (or lack thereof) for the Sky Connect key, which seems nearly impossible to exchange. 🔄❌  

Of course, I’m still loyal to my assistant for now, but an immediate and decisive course correction is absolutely necessary. ⚠️🚀

cogneato, 2025-03-18:
@DaSNyB This is not the community forum, otherwise I would go through each of these points with follow up points and questions. Please keep comments relevant to the specific issue.

trainman, 2025-03-18:
> This issue serves to gather information and troubleshoot issues with updating of **Raspberry Pi 5 boards only**. Before reporting an issue, try the following:
>
> 1. Perform the update after a cold [...]

I removed my pi5 from its normal location and brought it to my desk to try the cold boot approach. (It's connected to the official power adaptor in its proper location. At my desk, it's connected to a 96-watt Mac power adaptor that doesn't supply enough 5v power.)

Upgrade attempt one failed from a cold boot. I captured this screenshot before it restarted and automatically reverted itself to 14.2.

[image]

It tried again, forgetting to power off, and the upgrade worked, HAOS came up successfully. I placed it back in its home, and it wouldn't come back up. Unfortunately, no monitor is available to see why. I brought it back to my desk, and it got stuck here:

[image]

When connected to a monitor, I can power off and boot it or reboot it and HA will come up. For some reason w/o a monitor (even at my desk) it won't boot.

Currently, it appears stable sitting on my desk, but I'd like to get it back onto it's rack shelf, where it's headless and on a UPS.

Bootloader: 6fe0b091 2024/06/05
USB Ports: /dev/serial/by-id/usb-Zooz_800_Z-Wave_Stick_[...] and /dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_[...]-port0
Networking: Wired (same switch at desk and it's home / also tried another network cable, just to be sure)

linky67, 2025-03-18:
Too bad, previous post has been removed with screenshots.
It is still the case as I have a rasberry pi 5 16GB and did a cold restart as it was mandatory to unlock the upgrade that rollback to 14.2.
Same issue while still doing exactly the same. I also tried to upgrade in safe mode and failed the same way.

Here are the screenshots again:

[image]

[image]

[image]

yumbrad, 2025-03-18:
When mine failed to boot 15.0, the boot messages (kernel panic) and messages right before the emergency console comes up look just like linky67's and trainman's above.

linky67, 2025-03-18:
Finally decided to move forward as I am limited in time this week and do not want to be stuck while traveling.

Did a full backup (do not forget your encryption key), installed from scratch OS 15.0 on the SD card with Rasberry Imager and fully restored HA.

Worked like a charm (but do not forget to know your encryption password before erasing your data or your backup will be useless).

rouxdoo, 2025-03-19:
Following the top comment, starting from a cold boot I had a flawless upgrade so thank you for the guidance, would have been helpful to have that front-and-center on release notes.

Question - reading changelog it appears that the USB storage quirks were addressed and kernel upgraded. Can I safely remove this from my config.txt?

 # Added for compatibility with HAOSS 12.3->12.4 bug
dtoverlay=dwc2

SVH-Powel, 2025-03-19:
Just to add my experience. I have tried cold boot and upgrading 3 times now, but it will not boot v15. After 40 minutes, I pulled the power a few seconds and it booted v14.x. Rpi5 with SD card to boot from and a M.2 drive on the pci bus. I don't have a spare monitor to connect.

DaSNyB, 2025-03-19:
> [@DaSNyB](https://github.com/DaSNyB) This is not the community forum, otherwise I would go through each of these points with follow up points and questions. Please keep comments relevant to the specific issue.

Bro, I hear what you're saying, but you're going to tell me it's not a shitty job after all

2025-03-19 08:31:23.835 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd090 fail, reason -52
2025-03-19 08:31:23.835 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd095 fail, reason -52
2025-03-19 08:31:23.835 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd099 fail, reason -52
2025-03-19 08:31:23.835 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd09d fail, reason -52
2025-03-19 08:31:23.835 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd0a1 fail, reason -52
2025-03-19 08:31:23.835 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd0a5 fail, reason -52
2025-03-19 08:33:21.875 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0x100e fail, reason -52
2025-03-19 08:33:21.879 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
2025-03-19 08:33:21.987 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
2025-03-19 08:33:22.091 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
2025-03-19 08:33:22.191 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
2025-03-19 08:33:23.863 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd090 fail, reason -52
2025-03-19 08:33:23.867 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd095 fail, reason -52
2025-03-19 08:33:23.867 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd099 fail, reason -52
2025-03-19 08:33:23.867 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd09d fail, reason -52
2025-03-19 08:33:23.867 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd0a1 fail, reason -52
2025-03-19 08:33:23.867 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd0a5 fail, reason -52
2025-03-19 08:35:22.799 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0x100e fail, reason -52
2025-03-19 08:35:22.803 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
2025-03-19 08:35:22.907 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
2025-03-19 08:35:23.011 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
2025-03-19 08:35:23.115 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
2025-03-19 08:35:24.787 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd090 fail, reason -52
2025-03-19 08:35:24.787 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd095 fail, reason -52
2025-03-19 08:35:24.791 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd099 fail, reason -52
2025-03-19 08:35:24.791 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd09d fail, reason -52
2025-03-19 08:35:24.791 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd0a1 fail, reason -52
2025-03-19 08:35:24.791 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd0a5 fail, reason -52
2025-03-19 08:37:23.835 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0x100e fail, reason -52
2025-03-19 08:37:23.839 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
2025-03-19 08:37:23.943 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
2025-03-19 08:37:24.047 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
2025-03-19 08:37:24.151 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
2025-03-19 08:37:25.823 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd090 fail, reason -52
2025-03-19 08:37:25.827 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd095 fail, reason -52
2025-03-19 08:37:25.827 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd099 fail, reason -52
2025-03-19 08:37:25.827 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd09d fail, reason -52
2025-03-19 08:37:25.827 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd0a1 fail, reason -52
2025-03-19 08:37:25.827 homeassistant kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd0a5 fail, reason -52
2025-03-19 08:37:45.739 homeassistant kernel: audit: type=1334 audit(1742373465.733:593): prog-id=188 op=LOAD
2025-03-19 08:37:45.744 homeassistant systemd-timesyncd[670]: Network configuration changed, trying to establish connection.
2025-03-19 08:37:45.776 homeassistant systemd-timesyncd[670]: Contacted time server 162.159.200.123:123 (time.cloudflare.com).
2025-03-19 08:37:45.776 homeassistant systemd[1]: Started Journal Gateway Service.

Bredin76, 2025-03-19:
Here, tried to install 15.0 from 14.2 and my RPI 5 was stuck in boot.
had to unplug it, it booted in 5minutes but still in 14.2... 
Tried a new update from this "cold start" but still in 14.2... 

Will not try another time...

danez, 2025-03-19:
I see this in my logs:

<details>

<summary>Logs</summary>

```
2025-03-17 15:18:59.533 homeassistant kernel: rp1-firmware: probe of rp1_firmware failed with error -110
2025-03-17 15:18:59.536 homeassistant hassos-expand[449]: [INFO] Data partition end block 124735455
2025-03-17 15:18:59.537 homeassistant hassos-expand[449]: [INFO] No resize of data partition needed
2025-03-17 15:18:59.543 homeassistant kernel: Unable to handle kernel paging request at virtual address ffffffc0808edf04
2025-03-17 15:18:59.543 homeassistant kernel: Mem abort info:
2025-03-17 15:18:59.543 homeassistant kernel:   ESR = 0x0000000096000047
2025-03-17 15:18:59.543 homeassistant kernel:   EC = 0x25: DABT (current EL), IL = 32 bits
2025-03-17 15:18:59.543 homeassistant kernel:   SET = 0, FnV = 0
2025-03-17 15:18:59.543 homeassistant kernel:   EA = 0, S1PTW = 0
2025-03-17 15:18:59.543 homeassistant kernel:   FSC = 0x07: level 3 translation fault
2025-03-17 15:18:59.543 homeassistant kernel: Data abort info:
2025-03-17 15:18:59.543 homeassistant kernel:   ISV = 0, ISS = 0x00000047, ISS2 = 0x00000000
2025-03-17 15:18:59.543 homeassistant kernel:   CM = 0, WnR = 1, TnD = 0, TagAccess = 0
2025-03-17 15:18:59.543 homeassistant kernel:   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
2025-03-17 15:18:59.543 homeassistant kernel: swapper pgtable: 4k pages, 39-bit VAs, pgdp=00000000015c4000
2025-03-17 15:18:59.543 homeassistant kernel: [ffffffc0808edf04] pgd=100000010017c003, p4d=100000010017c003, pud=100000010017c003, pmd=1000000101c70003, pte=0000000000000000
2025-03-17 15:18:59.543 homeassistant kernel: Internal error: Oops: 0000000096000047 [#1] PREEMPT SMP
2025-03-17 15:18:59.543 homeassistant kernel: Modules linked in: brcmfmac_wcc sch_fq_codel btsdio brcmfmac hci_uart btqca btrtl btbcm brcmutil btintel spidev vc4 cfg80211 bluetooth snd_soc_hdmi_codec aes_ce_blk ecdh_generic ecc aes_ce_cipher ghash_ce libaes drm_display_helper cec drm_dma_helper drm_kms_helper rfkill snd_soc_core pisp_be sha2_ce sha256_arm64 snd_compress snd_pcm_dmaengine snd_pcm rpivid_hevc(C) spi_bcm2835 sha1_ce snd_timer snd v4l2_mem2mem rp1_pio(+) videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common v3d gpio_keys raspberrypi_hwmon gpu_sched i2c_brcmstb drm_shmem_helper pwm_fan rp1_mailbox rp1_adc raspberrypi_gpiomem rp1 nvmem_rmem uio_pdrv_genirq uio drm drm_panel_orientation_quirks backlight fuse
2025-03-17 15:18:59.543 homeassistant kernel: CPU: 1 PID: 177 Comm: (udev-worker) Tainted: G         C         6.6.74-haos-raspi #1
2025-03-17 15:18:59.543 homeassistant kernel: Hardware name: Raspberry Pi 5 Model B Rev 1.1 (DT)
2025-03-17 15:18:59.543 homeassistant kernel: pstate: 00400009 (nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
2025-03-17 15:18:59.543 homeassistant kernel: pc : __memcpy_toio+0x14/0xa0
2025-03-17 15:18:59.543 homeassistant kernel: lr : rp1_firmware_message+0x74/0x140 [rp1]
2025-03-17 15:18:59.543 homeassistant kernel: sp : ffffffc080f4b790
2025-03-17 15:18:59.543 homeassistant kernel: x29: ffffffc080f4b790 x28: ffffffc080f4bca8 x27: ffffffc080f4bc00
2025-03-17 15:18:59.543 homeassistant kernel: x26: 0000000000000001 x25: 0000000000020004 x24: ffffffc080f4b7f0
2025-03-17 15:18:59.543 homeassistant kernel: x23: 0000000000000008 x22: ffffffc080f4b7ec x21: ffffffd0144d1050
2025-03-17 15:18:59.543 homeassistant kernel: x20: 0000000000000004 x19: ffffff81074c8b00 x18: 0000000000000000
2025-03-17 15:18:59.543 homeassistant kernel: x17: 0000000000000000 x16: ffffffd084021288 x15: 0000000000000000
2025-03-17 15:18:59.543 homeassistant kernel: x14: 0000000000000000 x13: ffffffd084f1cc38 x12: 00000000dbfbb796
2025-03-17 15:18:59.543 homeassistant kernel: x11: 00000000000001c8 x10: 0000000000001a60 x9 : ffffffd0144cf07c
2025-03-17 15:18:59.543 homeassistant kernel: x8 : ffffff8104a03980 x7 : 0000000000000000 x6 : dead000000000100
2025-03-17 15:18:59.543 homeassistant kernel: x5 : dead000000000122 x4 : ffffffd0144d1060 x3 : 0000000000000020
2025-03-17 15:18:59.543 homeassistant kernel: x2 : 0000000000000004 x1 : ffffffc080f4b7ed x0 : ffffffc0808edf04
2025-03-17 15:18:59.543 homeassistant kernel: Call trace:
2025-03-17 15:18:59.543 homeassistant kernel:  __memcpy_toio+0x14/0xa0
2025-03-17 15:18:59.543 homeassistant kernel:  rp1_firmware_get_feature+0x54/0xb0 [rp1]
2025-03-17 15:18:59.543 homeassistant kernel:  rp1_pio_probe+0xf4/0x350 [rp1_pio]
2025-03-17 15:18:59.543 homeassistant kernel:  platform_probe+0x70/0xd8
2025-03-17 15:18:59.543 homeassistant kernel:  really_probe+0x150/0x2d0
2025-03-17 15:18:59.543 homeassistant kernel:  __driver_probe_device+0x80/0x140
2025-03-17 15:18:59.543 homeassistant kernel:  driver_probe_device+0x44/0x170
2025-03-17 15:18:59.543 homeassistant kernel:  __driver_attach+0x9c/0x1b0
2025-03-17 15:18:59.543 homeassistant kernel:  bus_for_each_dev+0x84/0xf8
2025-03-17 15:18:59.543 homeassistant kernel:  driver_attach+0x2c/0x40
2025-03-17 15:18:59.543 homeassistant kernel:  bus_add_driver+0xec/0x218
2025-03-17 15:18:59.543 homeassistant kernel:  driver_register+0x68/0x138
2025-03-17 15:18:59.543 homeassistant kernel:  __platform_driver_register+0x30/0x48
2025-03-17 15:18:59.543 homeassistant kernel:  rp1_pio_driver_init+0x28/0xff8 [rp1_pio]
2025-03-17 15:18:59.543 homeassistant kernel:  do_one_initcall+0x64/0x2c8
2025-03-17 15:18:59.543 homeassistant kernel:  do_init_module+0x5c/0x210
2025-03-17 15:18:59.543 homeassistant kernel:  load_module+0x1e94/0x2020
[... cut ...]
2025-03-17 15:18:59.562 homeassistant kernel: Code: f100005f d1002041 9a9f1021 b4000101 (f9400020) 
2025-03-17 15:18:59.562 homeassistant kernel: ---[ end trace 0000000000000000 ]---
2025-03-17 15:18:59.562 homeassistant kernel: note: (udev-worker)[359] exited with preempt_count 1
2025-03-17 15:18:59.569 homeassistant systemd[1]: Starting Virtual Console Setup...
2025-03-17 15:18:59.624 homeassistant systemd[1]: Finished Virtual Console Setup.
```
</details>

This was with bootloader from 2024-09-23

sairon, 2025-03-19:
@danez It looks like something that should be already fixed in the kernel in OS 15.0 by https://github.com/raspberrypi/linux/issues/6593 

However, 2024-09-23 EEPROM is indeed old and is missing support for the Linux PIO driver. I suggest updating it to the latest, either from another card with RPi OS using `rpi-eeprom-update -a` or with a card flashed with the bootloader update created in the RPi Flasher.

Same goes to @linky67 and @trainman - check the bootloader version and eventually update it.

@yumbrad What's EEPROM bootloader version in your case?

danez, 2025-03-19:
> @danez It looks like something that should be already fixed in the kernel in OS 15.0 by https://github.com/raspberrypi/linux/issues/6593
>
> However, 2024-09-23 EEPROM is indeed old and is missing [...]

I did indeed then update to some november version of the bootloader and in the process of it did cold boots and then the update worked.

LeonardSEO, 2025-03-19:
Just wanted to take a moment to say **thank you** @sairon  your effort, patience, and communication in this issue have been incredible.

The amount of chaos and pressure in those threads is huge, but somehow you manage to stay helpful, focused, and professional with every single reply. That really doesn’t go unnoticed!

It’s impressive how you keep seeing the forest for the trees while guiding everyone through troubleshooting steps, collecting logs, and still pushing this forward.

I truly appreciate your work here 🙏 it makes a massive difference to the community. Thanks again!

iyanmv, 2025-03-19:
> [@danez](https://github.com/danez) It looks like something that should be already fixed in the kernel in OS 15.0 by [raspberrypi/linux#6593](https://github.com/raspberrypi/linux/issues/6593)
> 
> However, 2024-09-23 EEPROM is indeed old and is missing support for the Linux PIO driver. I suggest updating it to the latest, either from another card with RPi OS using `rpi-eeprom-update -a` or with a card flashed with the bootloader update created in the RPi Flasher.
> 
> Same goes to [@linky67](https://github.com/linky67) and [@trainman](https://github.com/trainman) - check the bootloader version and eventually update it.
> 
> [@yumbrad](https://github.com/yumbrad) What's EEPROM bootloader version in your case?

I updated using `rpi-eeprom-update -a`:

```
CURRENT: Sat 20 Apr 10:53:30 UTC 2024 (1713610410)
UPDATE: Mon 23 Sep 13:02:56 UTC 2024 (1727096576)
```

Still the update to 15.0 failed for me. At least this time it went back on its own to 14.2 without having to unplug the power or anything.

danez, 2025-03-19:
> I updated using `rpi-eeprom-update -a`:
>
> ```
> CURRENT: Sat 20 Apr 10:53:30 UTC 2024 (1713610410)
> UPDATE: Mon 23 Sep 13:02:56 UTC 2024 (1727096576)
> ```
>
> Still the update to 15.0 failed for me. At [...]

Try running `apt update` and `apt upgrade` in raspberryOS before doing the bootloader update, you should get an even newer one and then it might work.

mcmuller, 2025-03-19:
@iyanmv same for me, I needed to flash with a card created by RPi Flasher to get the latest...

AdrianEddy, 2025-03-19:
I wasn't able to update from 14.2 to 15.0 even after cold boot. Every time it became stuck and after power cycle it reverted back to 14.2

Until I updated the bootloader using the RPi Imager via the SD Card - once the bootloader was updated I was able to successfully update HA from 14.2 to 15.0

HA should check the bootloader version somehow before update and inform user what to do, or at least there should be a clear warning about that in the release notes

sairon, 2025-03-19:
> HA should check the bootloader version somehow before update and inform user what to do, or at least there should be a clear warning about that in the release notes

Yes, that's something that's been on my backlog for a while. Automatic updates are too risky, and there is not an easy way how we can mandate it before an OS update, but easier way of updating the EEPROM from HAOS is definitely something that's needed. It wasn't that much needed in for Pi 4, but with Pi 5 things are more turbulent now and an old bootloader can be source of many troubles.

Anyway, updating the bootloader should be one of the first things to do before further troubleshooting a Pi 5. Unfortunately, as I commented in other issue, I'm having hard time reproducing any of the issues locally even when I downgrade the bootloader.

trainman, 2025-03-19:
@sairon, thank you so much for the fix. My Pi is now on the Feb 2nd bootloader, and my issues are resolved.

> Yes, that's something that's been on my backlog for a while. Automatic updates are too risky, and there is not an easy way how we can mandate it before an OS update, but easier way of updating the [...]

I am sure fixing this won't be easy, and I thank you for pressing on. Until then, can we please put something at the top of the release notes? Here's my reasoning...

I'm relatively new to HAOS and migrating from SmartThings. From how well all the other updates work and how polished they are, this caught me entirely off guard. I assumed the HAOS route meant I didn't have to maintain the underlying Pi. I may have missed it in the documentation, and that would be on me. Either way, it wasn't what I'd expect with the "*OS" naming convention; I expected it to replace Raspbian fully. A "reminder" in the release notes would help.

We should know ahead of time when we need to update the Pi firmware if HAOS can't do it for us. As a user of the platform, I don't want to be in a scenario where I need to rush to find a spare SD and flash my firmware when I'm not expecting that time commitment. If I can manage my expectations beforehand, I'm a happy camper. 

You all are doing a fantastic job with the platform. Please give us a little heads-up to help keep us enthused. I've been scared to apply updates lately, now that I understand why I'm feeling calmer, but it will take some time to regain my trust.

linky67, 2025-03-19:
> [@danez](https://github.com/danez) It looks like something that should be already fixed in the kernel in OS 15.0 by [raspberrypi/linux#6593](https://github.com/raspberrypi/linux/issues/6593) [...]

Hello, thank you for your support and feedback.

I confirm that the bootloader firmware was very old here. Created a usb stick to avoid SD card backups as I only had 1 with HAOS on it.

BOOTLOADER: update available
   CURRENT: Wed  6 Dec 18:29:25 UTC 2023 (1701887365)
    LATEST: Mon 23 Sep 13:02:56 UTC 2024 (1727096576)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader-2712/default)
            Use raspi-config to change the release.
   CURRENT: Wed  6 Dec 18:29:25 UTC 2023 (1701887365)
    UPDATE: Mon 23 Sep 13:02:56 UTC 2024 (1727096576)
    BOOTFS: /boot/firmware
UPDATE SUCCESSFUL

But as explained, I installed HAOS 15.0 from scratch and restored the backup on it. This means that the bootloader is not getting updated at all without going through the Rasberry Pi OS.
Would be a nice enhancement to be able to monitor and update the firmware from HA indeed.

Anyway, it is now tracked and clear. Thanks again for your support :-)

trainman, 2025-03-19:
> Hello, thank you for your support and feedback.
>
> I confirm that the bootloader firmware was very old here. Created a usb stick to avoid SD card backups as I only had 1 with HAOS on it.
>
> BOOTLOADER: [...]

@linky67, you and I appear to have quite different boot loaders despite just going through the same firmware update process today.

When I boot my Pi w/o a SD card, here's what I see:
```Raspberry Pi 5 - 8GB
bootloader: f788aab6    2025/02/12
update-ts: 1742405912```

Picard16, 2025-03-19:
Upgrade after cold boot also worked for my RPi5 8gb running on a 128gb SDcard.
I started the update from Android app.
I didn't try to update prior to cold boot, so it worked right away.

DavidMooreUK, 2025-03-19:
Had an issue in Raspberry Pi 3B+. Was no way to recover. Had to primate the card and do a restore from backup.

iyanmv, 2025-03-19:
> > Still the update to 15.0 failed for me. At least this time it went back on its own to 14.2 without having to unplug the power or anything.
> 
> Try running `apt update` and `apt upgrade` in raspberryOS before doing the bootloader update, you should get an even newer one and then the haos update might work.

Thank you! Exactly, now it worked! 🎊 For reference, this was the second update:

```
CURRENT: Mon 23 Sep 13:02:56 UTC 2024 (1727096576)
LATEST: Wed 12 Feb 10:51:52 UTC 2025 (1739357512)
```

trainman, 2025-03-19:
> Thank you! Exactly, now it worked! 🎊 For reference, this was the second update:
>
> ```
> CURRENT: Mon 23 Sep 13:02:56 UTC 2024 (1727096576)
> LATEST: Wed 12 Feb 10:51:52 UTC 2025 (1739357512)
> ```

@linky67, you may need to do this to get the latest firmware...

linky67, 2025-03-19:
> [@linky67](https://github.com/linky67), you and I appear to have quite different boot loaders despite just going through the same firmware update process today.
>
> When I boot my Pi w/o a SD card, [...]

Indeed i missed that, thanks. This also helped to upgrade all firmwares on the board. I used "apt full-upgrade" just in case as I didnt really care about the Rasberry OS.

yumbrad, 2025-03-19:
> 
> [@yumbrad](https://github.com/yumbrad) What's EEPROM bootloader version in your case?

is there a way to check from HAOS, or do I need to shutdown and flash Raspbian or something?

sairon, 2025-03-20:
The current version can be checked from HAOS as described here: https://github.com/home-assistant/operating-system/issues/3953#issuecomment-2736686022

There is no way to update is from HAOS though, I'd say the easiest way is to use RPi Imager with a spare SD card as described here: https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#bootloader_update_stable

elfinith, 2025-03-21:
Raspberry 3B with USB-connected SSD, stuck after upgrade to 15.
Power-cyceled a lot of times, no luck...

Every time displays stays with message:

Starting USB...
Bus usb@7e980000: USB DWC2
scanning bus usb@7e980000 for devices...

AndersRipa, 2025-03-21:
Now my system Raspberry Pi 5 have updated correctly, I updated the boot loader to the latest version and then the update to 15.0 went without problems.

avegao, 2025-03-21:
> The current version can be checked from HAOS as described here: [#3953 (comment)](https://github.com/home-assistant/operating-system/issues/3953#issuecomment-2736686022)
> 
> There is no way to update is from HAOS though, I'd say the easiest way is to use RPi Imager with a spare SD card as described here: https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#bootloader_update_stable

I can confirm that works with a rpi5, easy and fast. Thanks!

yumbrad, 2025-03-21:
> The current version can be checked from HAOS as described here: [#3953 (comment)](https://github.com/home-assistant/operating-system/issues/3953#issuecomment-2736686022)

Interestingly, on my HAOS, /proc/device-tree is a symbolic link to a non-existent /sys/firmware/devicetree/base, so I couldn't check from HAOS.

But I booted into rpi OS lite and checked (and then updated), and I was on "5 Jan 15:57:30 GMT 2024". With the  feb'25 bootloader, all seems to be working on 15.0 now.

dangelomda, 2025-03-21:
On my Dell Wyse, the installation error occurred, I even skipped this update to version 15 in the hope of another upgrade that would work without causing startup problems.

[image]

[image]

basically the same update issue.

rdeserm, 2025-03-22:
Phew, after a few hard reboots it looks like it reverted back automatically?  HomeAssistant been working flawlessly for years, now I'm a bit scared to install this update again.

Sanaki, 2025-03-22:
Having read through this issue, it seems there are two separate issues mentioned.

1. Pi5 with outdated bootloader will fail to install HAOS 15.

2. Pi5 with HAOS 15 installed will fail to boot without a monitor attached.

The first is what this issue is about. If the second is valid, someone who experienced that problem should open a new issue and post logs there.

dangelomda, 2025-03-22:
> Phew, after a few hard reboots it looks like it reverted back automatically? HomeAssistant been working flawlessly for years, now I'm a bit scared to install this update again.

Me too Im Skip this version.

YasinSuzon, 2025-03-22:
Still no solution? 😕 new HA core update is here too, let see if it can be installed

ThomasHeinrichSchmidt, 2025-03-22:
Same here, update from HA OS 14.2. to 15 fails. 
Cannot check my bootloader version from "Termial & SSH Add-on" using `$ xxd /proc/device-tree/chosen/bootloader/build-timestamp`, yields: `xxd: /proc/device-tree/chosen/bootloader/build-timestamp: No such file or directory`. Is there another way to find out the bootloader version?

iyanmv, 2025-03-22:
@YasinSuzon @ThomasHeinrichSchmidt It's been explained a few times already above, and confirmed to work by also a few of us.

1. Create an SD or USB with Raspberry Pi OS
2. Boot up and then run
```bash
sudo apt-get update
sudo apt-get dist-upgrade
sudo rpi-eeprom-update -a
```
3. Update to 15.0 will work without issues

For now it is not possible to update the Raspberry Pi bootloader EEPROM from HA. This was not an issue with RPi4, but it seems like EEPROM updates are becoming more relevant for RPi5 with several bugs fixed in the last months.

Picard16, 2025-03-22:
Why search for the version of the bootloader first and not simply update it directly? This seems to be recommended or even necessary anyway. I did it with the instructions that Sairon linked above:
_There is no way to update is from HAOS though, I'd say the easiest way is to use RPi Imager with a spare SD card as described here: https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#bootloader_update_stable _

Takes no longer than 10 minutes if you still have a microSD card lying around. The flashing itself was a matter of a few seconds.
Raspi 5 with 8GB and HA on a 128GB SDcard

jasipsw, 2025-03-22:
I ran into this problem on my rpi5, I worked through it and was able to get to OS 15.0. 

What I did was disconnect the RPI from the wired network and connect a physical monitor and keyboard. I then did a number of "unplug and then wait" for cycles until the RPI booted to the prior os and the CLI was active.  "os info" showed one of the slots as having os 15.0, but it is marked bad.  

I then used the " os boot-slot other" command. The rpi rebooted and shows os 15.0 as active. Now all is well with HA again.

This was done with the sdcard installed.

yoabau, 2025-03-22:
I ran into this problem as well. HA OS 14 was not able to upgrade to v15.
I updated the bootloader on my raspberry Pi 5 and the upgrade was successful.
Thanks to @sairon for the advices.

ThomasHeinrichSchmidt, 2025-03-22:
@Picard16 - yes, you are right, no need to know the current bootloader. But for me it's more work to upgrade the bootloader than having a "microSD card lying around". I need to detach the RaspPI from its case, find an HDMI monitor and a spare keyboard, lay a network cable through the apartment to the router, and then I'm ready for the "10 minutes".

Picard16, 2025-03-22:
Hm, I am not quite sure what you mean.
I flashed the SD card with my Laptop first, then shut down the HA System, unplugged it from electricity, changed the SD card in my RasPi5, HA SD card out, put the bootloader SD card in the slot, plug the RasPi5 again, waited a minute until the LED flashes regularly (quite fast), unplug RasPi5 again, put the HA SDcard back in and boot the RasPi5 again. That's it. No network cable, no keyboard, no HDMI monitor needed.
I only doublecheck the LED flashing with the green screen, but the flashing LED is enough from my point of view.

revbem5000, 2025-03-22:
> @YasinSuzon @ThomasHeinrichSchmidt It's been explained a few times already above, and confirmed to work by also a few of us.
>
> 1. Create an SD or USB with Raspberry Pi OS
> 2. Boot up and then run [...]

That worked perfectly. 
My Bootloader was from September 2024, after update from February 2025.
[image]
Thx a lot (RPi5, 8GB, SSD 256GB).

ThomasHeinrichSchmidt, 2025-03-22:
Yes, but for your method to work, you need to connect a monitor and keyboard and have an Ethernet cable with Internet handy.
If one can't or don't want to do this, [Use Raspberry Pi Imager to update the bootloader](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#imager) is a better way.

[13 more replies]

closed 2025-03-31 by sairon