home-assistant/operating-system #3944: HP T620 stuck after HAOS15 update

peat-puck, 2025-03-18:
### Describe the issue you are experiencing

device boot stalls during/after update OS15

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

HP-T620, 8GB ram, 256GB SSD
ETH on UTP cable
Sonoff Zigbee 3.0 USB Dongle plus attached to USB via extension cord

### Steps to reproduce the issue

1. started OS 15 (from 14.2) update via settings menu
2. System does not correctly reboot and stays unresponsive indefinitely
3. monitor attached to HA server (T620)
4. console shows: warn home assistant cli not starting jump into emergency console
5.  waited 30 minutes
6. typed: login
7. console shows: warn home assistant cli not starting jump into emergency console
8. waited 30 minutes
9. typed: reboot
10. waited 30 minutes
11. performed a hard power cycle
12. back to step 2
13. system keeps stalled

### Anything in the Supervisor logs that might be useful for us?

```txt
not accessable, server stalls
```

### Anything in the Host logs that might be useful for us?

```txt
not accessable, server stalls
```

sairon, 2025-03-18:
> console shows: warn home assistant cli not starting jump into emergency console

Does it jump into the emergency console at this point? Your report seems to be a bit confusing about that, saying the system is stalled. If you can type any commands at that point, `login` would do nothing (it serves to switch from HA CLI to OS shell). In the OS shell/emergency console, you can check what fails to start by running `systemctl --failed` and `docker ps -a`.

peat-puck, 2025-03-18:
hello Sairon,
sorry that I'm unclear, my tech knwoledge is a bit under powered.
Yes it jumps into emergy console at that point. Tried "login"a few times as I found on the Internet that this could solve the issue.
I could also enter "reboot"and the system response that is goed into reboot.
Screen shows unmounting
last line says: rebooting  but nothing happens afterwards
waited for approx. 15 minutes and performed a powercycle
did that several times and now the system is back up with the previous OS 14.2 and shows an available update to 15 again.

hope this helps, thanks for your help and understanding!

DaHype, 2025-03-18:
I have exact the sane problem. But i use another HP , but its Same issue.

Knillis, 2025-03-18:
Same issue here. Update from 14.2 > 15.0 on a HP T620.

iVolt1, 2025-03-18:
I have a Dell Wyse 5060 that I thought was toast after the upgrade last night and a few reboots.  I restarted this morning and everything came up with HAOS 14.2 still installed and the15.0 upgrade still offered.

DaHype, 2025-03-18:
I really hope for an Fix. Which Logs i have to provide? Or any other that could be helpful?
I´m newbie to that, so please understand. I´m trying my best to help that the issue can be fixed very soon.

Bjk8kds, 2025-03-18:
> on my T620 too 
> 
> [image]
> 
> Edit, it came back after 3rd manual reboot attempt, automatically rolled back to 14.2
> 
> [image] 

 _Originally posted by @Bjk8kds in [#3937](https://github.com/home-assistant/operating-system/issues/3937#issuecomment-2730963328)_

spanxter, 2025-03-18:
I have the exact same issue… so if I powercycle the machine a few times it will automatically go back to the previous version?

Bluesw, 2025-03-18:
im using dell wyze can also confirm i was unable to install update had to restart her 3 times to revert back and get up n running

sairon, 2025-03-18:
To get the logs from broken 15.0, you can follow these steps:
1. Update to 15.0 again.
1. When it "jumps to the emergency console", check that data folder is mounted: `test -d /mnt/data/supervisor/homeassistant/ || echo "fail"`
1. If the command didn't return "fail", run `(systemctl --failed; systemctl status docker; docker ps -a; dmesg) > /mnt/data/supervisor/homeassistant/debug.txt` (be sure to get the parenthesis correct)
1. Run `ha os boot-slot other` to return back to the working slot, or reboot and select the other slot in the GRUB menu. When HA comes up, you should be able to fetch the `debug.txt` file from the HA configuration folder (e.g. using the File editor addon).

kridness, 2025-03-18:
> Same issue here. Update from 14.1 > 15.0 on a HP T620.
> 
> Edit: just rebooted for the 5th time, used another LAN port and I'm back up. But doesn't seem to be updated:
> 
> ```
> Core 2025.1.4
> Supervisor 2025.03.3
> Operating System 14.1
> Frontend 20250109.2
> ```
> 
>

I got my system live again after unplugging the LAN cable and a hard reboot. I saw the HA OS was still v 14.2.

compuguy, 2025-03-18:
> To get the logs from broken 15.0, you can follow these steps:
> 
> 1. Update to 15.0 again.
> 2. When it "jumps to the emergency console", check that data folder is mounted: `test -d /mnt/data/supervisor/homeassistant/ || echo "fail"`
> 3. If the command didn't return "fail", run `(systemctl --failed; systemctl status docker; docker ps -a; dmesg) > /mnt/data/supervisor/homeassistant/debug.txt` (be sure to get the parenthesis correct)
> 4. Run `ha os boot-slot other` to return back to the working slot, or reboot and select the other slot in the GRUB menu. When HA comes up, you should be able to fetch the `debug.txt` file from the HA configuration folder (e.g. using the File editor addon).

Was able to pull logs following your instructions (minus the boot slot change; no access to ha cli). The dmesg log mentions a kernel bug/oops: `kernel BUG at drivers/gpu/drm/amd/amdgpu/amdgpu_device.c:1174!` at line 899. See attached file @sairon: 

[debug.txt](https://github.com/user-attachments/files/19324866/debug.txt)

sairon, 2025-03-18:
From other sources, it seems this is a bug in the Linux `amdgpu` driver. Could anyone please at least provide a screen capture of the output of `dmesg -l warn+` (mind the plus sign) when the device boots in the emergency console?

compuguy, 2025-03-18:
Yeah, that's what I thought @sairon. Upstream kernel bug with `amdgpu`. Though a temporarily workaround *could be* to blacklist the `amdgpu` driver and fallback to the framebuffer or whatever fallback for console display? Not sure that's worth the effort when a rollback to 14.2 works as well....

sairon, 2025-03-18:
Actually yes, just returned here to suggest that - since the `amdgpu` driver is loaded as module, adding `modprobe.blacklist=amdgpu` should work. You'll lose modesetting but console should still be shown. To try it, you can press `e` in the GRUB menu and add it at the end of the arguments (last one would be probably probably `rauc.slot=X`), then run it with Ctrl+x. If it works, it can be persisted in `cmdline.txt` in the boot partition.

KeiJeiYoon, 2025-03-19:
I had the same issue after updating to 15.0.
It kept getting stuck and not showing the CLI screen. So I manually changed to another boot slot and booted.

BuzatuCristian, 2025-03-19:
Can we hope that you will release a functional update for the unfortunate owners of the HP T620 that will not destroy it and possibly repair the slot already marked as defective?

hakjoh1955, 2025-03-19:
Same issue, hp t620. After hardreboot it jumped back to 14.2 and works ok

RvV1979, 2025-03-19:
Same issue here on HP T520

PNXPH, 2025-03-19:
Same issue. Hp Thinclient 8GB ram, 128GB SSD.

Eraser127, 2025-03-19:
Same on HP T520

sairon, 2025-03-19:
A user on the community forums confirmed that adding `amdgpu` to the blacklisted modules can be used as a workaround. To apply it permanently, connect to the device (either directly with a display or through developer SSH at port 22222, if you have it enabled) and run:
```
sed -i.bak "s/$/ modprobe.blacklist=amdgpu/" /mnt/boot/cmdline.txt
```
For a proper solution, we need to wait for an upstream driver fix.

sairon, 2025-03-19:
Bug report upstream: https://gitlab.freedesktop.org/drm/amd/-/issues/4050

If you have more information to add or if somethings is not right in the report, please add it to that report directly. If you need help reverting the update, applying the workaround or anything else, please use [Discussions](https://github.com/home-assistant/operating-system/discussions) or [Community forums](https://community.home-assistant.io/).

sairon, 2025-03-19:
Here's a test build that might fix these issues without the need for disabling the `amdgpu` driver: https://github.com/home-assistant/operating-system/actions/runs/13952565640/artifacts/2783001415

I'd appreciate any feedback on it. To test it, you can flash it to a USB drive (e.g. using Rufus or any other preferred utility) and use it as a temporary boot device. Be aware the downloaded image is a ZIP archive that needs to be extracted first (unlike standard OS builds, which are `.img.xz` that can be usually used directly; that's unfortunately limitation of GitHub build artifacts).

If it boots correctly, let me know in https://github.com/home-assistant/operating-system/pull/3957. If not, check `dmesg -l warn+` as before and post it there too. Thanks! :heart:

closed 2025-03-20 by frenck