home-assistant/operating-system #3948: Update to os 15 fails, falls back to 14.2

bruce607, 2025-03-18:
### Describe the issue you are experiencing

Update to os 15 fails, falls back to 14.2

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

14.2

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Intel NUC

Sonoff Zigbee dongle

ZBT-1 running as matter

### Steps to reproduce the issue

1. attempt to update to os 15
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2025-03-18 07:34:38.342 WARNING (MainThread) [supervisor.addons.addon] Watchdog found addon Network UPS Tools is failed, restarting...
2025-03-18 07:34:38.344 INFO (SyncWorker_6) [supervisor.docker.manager] Cleaning addon_a0d7b954_nut application
2025-03-18 07:34:38.540 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/nut/amd64 with version 0.14.1
2025-03-18 07:34:39.424 INFO (MainThread) [supervisor.core] Start Home Assistant Core
2025-03-18 07:34:39.437 INFO (SyncWorker_1) [supervisor.docker.manager] Starting homeassistant
2025-03-18 07:34:39.527 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready
2025-03-18 07:34:43.009 WARNING (MainThread) [supervisor.addons.addon] Watchdog found addon Network UPS Tools is failed, restarting...
2025-03-18 07:34:43.012 INFO (SyncWorker_7) [supervisor.docker.manager] Cleaning addon_a0d7b954_nut application
2025-03-18 07:34:43.245 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/nut/amd64 with version 0.14.1
2025-03-18 07:34:43.556 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state startup
2025-03-18 07:34:43.556 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2025-03-18 07:34:44.534 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2025-03-18 07:34:44.537 INFO (MainThread) [supervisor.homeassistant.core] Home Assistant Core state changed to APIState(core_state='NOT_RUNNING', offline_db_migration=False)
2025-03-18 07:34:47.790 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-2285' coro=<Addon.watchdog_container() done, defined at /usr/src/supervisor/supervisor/addons/addon.py:1579> exception=AddonsJobError('Rate limit exceeded, more than 10 calls in 0:30:00')>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/addons/addon.py", line 1593, in watchdog_container
    await self._restart_after_problem(event.state)
  File "/usr/src/supervisor/supervisor/jobs/decorator.py", line 291, in wrapper
    raise on_condition(
        f"Rate limit exceeded, more than {self.throttle_max_calls} calls in {self.throttle_period(group_name)}",
    )
supervisor.exceptions.AddonsJobError: Rate limit exceeded, more than 10 calls in 0:30:00
2025-03-18 07:35:11.147 INFO (MainThread) [supervisor.homeassistant.core] Home Assistant Core state changed to APIState(core_state='RUNNING', offline_db_migration=False)
2025-03-18 07:35:11.147 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
2025-03-18 07:35:11.147 INFO (MainThread) [supervisor.addons.manager] Phase 'application' starting 1 add-ons
2025-03-18 07:35:11.368 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-configurator with version 5.8.0
2025-03-18 07:35:41.479 INFO (MainThread) [supervisor.misc.tasks] All core tasks are scheduled
2025-03-18 07:35:41.479 INFO (MainThread) [supervisor.host.info] Updating local host information
2025-03-18 07:35:41.480 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
2025-03-18 07:35:41.481 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
2025-03-18 07:35:41.481 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
2025-03-18 07:35:41.481 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2025-03-18 07:35:41.482 INFO (MainThread) [supervisor.resolution.checks.base] Run check for no_current_backup/system
2025-03-18 07:35:41.482 INFO (MainThread) [supervisor.resolution.module] Create new suggestion create_full_backup - system / None
2025-03-18 07:35:41.482 INFO (MainThread) [supervisor.resolution.module] Create new issue no_current_backup - system / None
[... 178 lines cut ...]

### System information

<html>
<body>

Version | core-2025.3.3
-- | --
Installation type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
User | root
Virtual environment | false
Python version | 3.13.2
Operating system family | Linux
Operating system version | 6.6.73-haos
CPU architecture | x86_64

</body>
</html>

### Additional information

after attemoting to upgrade, system stays unavailabel for 10 to 15 minutes then reverts to 14.2

sairon, 2025-03-18:
What "Intel NUC" exactly? Can you try upgrading with a display connected and/or USB devices disconnected?

aperezva, 2025-03-18:
I 

> What "Intel NUC" exactly? Can you try upgrading with a display connected and/or USB devices disconnected?

I have more or less the same NUc in Prxmox, I don´t know if it´s safely upgarde. Do you have any Root cause regarding these different issues in diferent devices. 

do you recomend wait some days to give you time to investigate?.

Thanks for your efforts

avdb35, 2025-03-18:
Same problem here. HP NUC intel 
update and no response after that. 
monitor connected and startup (after installation) runs till cfi
keeps waiting and finally  falls back in safe mode,
version 14.1 installed and (lucky me) backup uploaded
all's fine now

BartRoels, 2025-03-18:
Same issue, have booted into the 14.2 again. Need to investigate the logs further, but needed to get HA up and running again before my household got angry :)

BernieHD, 2025-03-19:
same issue here running

Home Assistant

Core 2025.3.3
Supervisor 2025.03.3
Operating System 14.2
Frontend 20250306.0

on thin client HPt620, upgraded to 8GB RAM & 128GB SSD

Eraser127, 2025-03-19:
Here the same as @PNXPH with a HP T520 with 240GB SSD and HA15 image.

sairon, 2025-03-19:
@avdb35 @BartRoels same question as above, what device is that exactly?

Would it be possible to connect an HDMI display and update with it connected to see where it gets stuck? If it falls into the emergency console, check `dmesg -l warn+` for errors. Eventually, there might be something in the [Host logs](https://my.home-assistant.io/redirect/logs/?provider=host) for previous boots (click three dots and select previous boot in "Show HAOS startups").

@BernieHD @PNXPH @Eraser127 Please use search before reporting, those HP devices are discussed in #3944.

Corvus71, 2025-03-19:
Hi,

I have HA (OS15) running on a Lenovo laptop G50 which also reverting to emergency console since it has an AMD GPU,

check for line:
[drm:amdgpu_invalid_rreg [amdgpu]] *ERROR* Invalid callback to read register 0x58184

Here is the log: Hope it helps

  UNIT LOAD ACTIVE SUB DESCRIPTION
0 loaded units listed.
● docker.service - Docker Application Container Engine
     Loaded: loaded (/usr/lib/systemd/system/docker.service; enabled; preset: enabled)
    Drop-In: /usr/lib/systemd/system/docker.service.d
             └─failure.conf, hassos.conf, resource.conf
     Active: active (running) since Wed 2025-03-19 18:20:50 UTC; 4min 39s ago
TriggeredBy: ● docker.socket
       Docs: https://docs.docker.com
    Process: 556 ExecStartPre=/usr/libexec/docker-disk-check (code=exited, status=0/SUCCESS)
   Main PID: 558 (dockerd)
      Tasks: 30
     Memory: 134.9M
        CPU: 3.014s
     CGroup: /system.slice/docker.service
             ├─558 /usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock
             ├─856 /usr/bin/docker-proxy -proto tcp -host-ip 0.0.0.0 -host-port 4357 -container-ip 172.30.32.6 -container-port 80 -use-listen-fd
             └─864 /usr/bin/docker-proxy -proto tcp -host-ip :: -host-port 4357 -container-ip 172.30.32.6 -container-port 80 -use-listen-fd

Mar 19 18:20:52 HomeAsslaptop hassio_supervisor[558]: [18:20:52] INFO: Starting local supervisor watchdog...
Mar 19 18:20:52 HomeAsslaptop hassio_supervisor[558]: s6-rc: info: service legacy-services successfully started
Mar 19 18:20:55 HomeAsslaptop hassio_supervisor[558]: 2025-03-19 18:20:55.316 INFO (MainThread) [__main__] Initializing Supervisor setup
Mar 19 18:20:55 HomeAsslaptop hassio_supervisor[558]: 2025-03-19 18:20:55.480 INFO (MainThread) [supervisor.coresys] Setting up coresys for machine: generic-x86-64
Mar 19 18:20:55 HomeAsslaptop hassio_supervisor[558]: 2025-03-19 19:20:55.496 INFO (MainThread) [supervisor.docker.supervisor] Attaching to Supervisor ghcr.io/home-assistant/amd64-hassio-supervisor with version 2025.03.3
Mar 19 18:20:55 HomeAsslaptop hassio_supervisor[558]: 2025-03-19 19:20:55.496 INFO (MainThread) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
Mar 19 18:20:55 HomeAsslaptop hassio_supervisor[558]: 2025-03-19 19:20:55.685 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state initialize
Mar 19 18:20:55 HomeAsslaptop hassio_supervisor[558]: 2025-03-19 19:20:55.691 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
Mar 19 18:20:55 HomeAsslaptop hassio_supervisor[558]: 2025-03-19 19:20:55.692 INFO (MainThread) [__main__] Setting up Supervisor
Mar 19 18:20:55 HomeAsslaptop hassio_supervisor[558]: 2025-03-19 19:20:55.918 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
CONTAINER ID   IMAGE                                                          COMMAND               CREATED                  STATUS                       PORTS                                     NAMES
7485a96897b5   ghcr.io/home-assistant/amd64-hassio-observer:2023.06.0         "/usr/bin/observer"   Less than a second ago   Up 4 minutes                 0.0.0.0:4357->80/tcp, [::]:4357->80/tcp   hassio_observer
376148c0f52e   ghcr.io/home-assistant/amd64-hassio-supervisor:latest          "/init"               4 minutes ago            Up 4 minutes                                                           hassio_supervisor
a2cd93e10613   homeassistant/amd64-addon-configurator:5.8.0                   "/init"               6 minutes ago            Exited (143) 5 minutes ago                                             addon_core_configurator
fb761187be7a   ghcr.io/home-assistant/amd64-hassio-multicast:2024.03.0        "/init"               7 minutes ago            Exited (0) 5 minutes ago                                               hassio_multicast
99727a625795   ghcr.io/home-assistant/amd64-hassio-audio:2023.12.0            "/init"               7 minutes ago            Exited (0) 5 minutes ago                                               hassio_audio
2721b78bb738   ghcr.io/home-assistant/amd64-hassio-dns:2025.02.0              "/init"               7 minutes ago            Exited (0) 5 minutes ago                                               hassio_dns
16368538bd4b   ghcr.io/home-assistant/amd64-hassio-cli:2025.02.0              "/init"               7 minutes ago            Exited (0) 5 minutes ago                                               hassio_cli
cc0a319348d6   ghcr.io/home-assistant/generic-x86-64-homeassistant:2025.3.3   "/init"               4 days ago               Exited (0) 6 minutes ago                                               homeassistant
[    0.000000] Linux version 6.12.18-haos (builder@644e17d3fe20) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -g3914f8ca) 12.4.0, GNU ld (GNU Binutils) 2.40) #1 SMP PREEMPT_DYNAMIC Mon Mar 17 11:16:39 UTC 2025
[    0.000000] Command line: BOOT_IMAGE=(hd0,gpt2)/bzImage root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootwait zram.enabled=1 zram.num_devices=3 systemd.machine_id=a36555edd7b742468987d48e72a48e5c fsck.repair=yes console=tty0 rauc.slot=A
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000006dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000006e000-0x000000000006ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000070000-0x0000000000087fff] usable
[... cut ...]
[   42.941439] docker0: port 1(vethfba8f8c) entered blocking state
[   42.941456] docker0: port 1(vethfba8f8c) entered forwarding state
[   46.822263] hassio: port 2(vethc4c149e) entered blocking state
[   46.822291] hassio: port 2(vethc4c149e) entered disabled state
[   46.822351] vethc4c149e: entered allmulticast mode
[   46.823204] vethc4c149e: entered promiscuous mode
[   46.912900] eth1: renamed from veth4dd1c6b
[   46.914463] hassio: port 2(vethc4c149e) entered blocking state
[   46.914476] hassio: port 2(vethc4c149e) entered forwarding state

reini-1, 2025-03-19:
Same problem here.

After a few power off/on it booted into previous 14.2.

I tried to update via web and command line (ha os update), both with the same result.

System: T8plus Intel N100 16GB RAM, 512GB m.2 SSD (https://vi.aliexpress.com/item/1005005234838380.html)

[dmesg.log from the booted system (14.2)](https://github.com/user-attachments/files/19348831/dmesg.log)

```shell
# ha os info
board: generic-x86-64
boot: B
boot_slots:
  A:
    state: inactive
    status: bad
    version: "15.0"
  B:
    state: booted
    status: good
    version: "14.2"
data_disk: N900-512-AA000000000000003217
update_available: true
version: "14.2"

# ha host info
agent_version: 1.6.0
apparmor_version: 3.1.2
boot_timestamp: 1742408459545288
broadcast_llmnr: true
broadcast_mdns: true
chassis: embedded
cpe: cpe:2.3:o:home-assistant:haos:14.2:*:production:*:*:*:generic-x86-64:*
deployment: production
disk_free: 421
disk_life_time: null
disk_total: 468.7
disk_used: 28.6
dt_synchronized: true
dt_utc: "2025-03-19T18:22:27.911664+00:00"
features:
- reboot
- shutdown
- services
- network
- hostname
- timedate
- os_agent
- haos
- resolved
- journal
- disk
- mount
hostname: homeassistant
kernel: 6.6.73-haos
llmnr_hostname: homeassistant
operating_system: Home Assistant OS 14.2
startup_time: 0.740659
timezone: Etc/UTC
use_ntp: true
virtualization: ""

# ha cli info
update_available: false
version: 2025.02.0
version_latest: 2025.02.0
```

sairon, 2025-03-19:
@reini-1 Thanks for the details, unfortunately, that's not much helpful at this point - I'll need logs from the failing OS as I asked in my previous post.

AnthonyWrather, 2025-03-20:
I had the same problem on a Raspberry Pi 5 upgrading from 14.2 to 15.0
In my case it was the MotionEye add-on that was installed but not configured. I'm not sure if it was the lack of config or the add-on itself but once I had removed it the upgrade worked fine.

BeJaHome, 2025-03-20:
Updated from 14.2 to 15.0, after reboot my system reported 14.1 (i.e. apparent downgrade) while still showing 15.0 as available. Retried update to 15.0. When rebooted system reports 14.2 again, still showing 15.0 as available. System runs smoothly. Decided to skip 15.0 for now and wait for 15.1 to arrive.
RPi4-64-8GB with SSD 128GB (92% free), running HaOS. Core 2025.3.3.

sairon, 2025-03-20:
> What's now the next step? We have to fix the damaged slot first somehow? Is there any way to update to haos15?
> I can find many posts about boot fail after update but I didn't find any official guide of what we should do.

Nothing is damaged, it just means it failed to boot into that slot. You don't need to take any other steps when it's booted back into the working slot.

At this point I'm still waiting for kernel logs as asked above, getting them would help tremendously. I have a device with Intel N100 as well, yet it's running fine on that one. So likely it's caused by a driver for hardware used on some of those mini PCs.

parisdalas, 2025-03-20:
> Nothing is damaged, it just means it failed to boot into that slot. You don't need to take any other steps when it's booted back into the working slot.
>
> At this point I'm still waiting for kernel [...]

I would be happy to provide you what you need to help us. If you can help me take that kernel log for you would be good. I just don't know how to do it...
Can you explain the slot thing? Right now it says that slot B is at version 15.0 . In case I try to update again the slot A will also be in version 15.0 and then the boot will be fail in both slots?

eljenso, 2025-03-20:
@parisdalas I think they are referring to the host logs.
See the "host logs" link in their comment here: https://github.com/home-assistant/operating-system/issues/3948#issuecomment-2737477477

WladyTee, 2025-03-20:
Shared feedbacks from my side: using old Intel NUC with only 4GB RAM, HAOS physical installation (no VM)

- Update to OS 15 worked fine, it was applied with CORE 2025.3.1 (even if 2025.3.3 was available).
- Massive slow down was experienced with OS 15, then installed CORE 2025.3.3.
- Several system slowdowns, Companion App from remote keeps on disconnecting, multiple self reboots, database corruption at startup, several integration timeouts.

It seems related to a large memory usage.
I hope it helps to focus the proper fix.

odin6796, 2025-03-20:
Just to let you know that PI4 also fail to install.
Sorry no logs.....

bruce607, 2025-03-20:
Here is my Host log. (Let me know if this helps, or if I need to obtain this log in some other way) This was downloaded from home assistant, using maximum lines allowed in interface

[host_2025-03-20T13-17-40.149Z.log](https://github.com/user-attachments/files/19367764/host_2025-03-20T13-17-40.149Z.log)

reini-1, 2025-03-20:
can I somehow get these host log files directly as via the gui I can only download max. 10000 lines and that is only a small part ...
I looked in the filesystem of the host but could not find them.

parisdalas, 2025-03-20:
I tried to take the host logs as well but it doesn't go back to the point I did the update. It can show only 1-1,5 day log.

reini-1, 2025-03-20:
in the 3-dotts-menu you can activate haos starts then you can select differntent logs

reini-1, 2025-03-20:
it seem 10000 lines is enough ...

previous start: [host_2025-03-20T13-34-03.930Z.log](https://github.com/user-attachments/files/19368095/host_2025-03-20T13-34-03.930Z.log)

2 starts before: [host_2025-03-20T13-35-03.997Z.log](https://github.com/user-attachments/files/19368109/host_2025-03-20T13-35-03.997Z.log)

3 starts before: [host_2025-03-20T13-35-45.362Z.log](https://github.com/user-attachments/files/19368113/host_2025-03-20T13-35-45.362Z.log)

4 starts before: [host_2025-03-20T13-36-03.776Z.log](https://github.com/user-attachments/files/19368124/host_2025-03-20T13-36-03.776Z.log)

5 starts before: [host_2025-03-20T13-36-18.834Z.log](https://github.com/user-attachments/files/19368129/host_2025-03-20T13-36-18.834Z.log)

6 starts before: [host_2025-03-20T13-36-38.905Z.log](https://github.com/user-attachments/files/19368132/host_2025-03-20T13-36-38.905Z.log)

7 starts before: [host_2025-03-20T13-37-06.393Z.log](https://github.com/user-attachments/files/19368140/host_2025-03-20T13-37-06.393Z.log)

8 starts before: [host_2025-03-20T13-37-23.960Z.log](https://github.com/user-attachments/files/19368155/host_2025-03-20T13-37-23.960Z.log)

current log is already bigger and it did not show the beginnen ...

bruce607, 2025-03-20:
Sorry, this should be the correct Host file (selected show startups)

[host_2025-03-20T14-10-38.431Z.log](https://github.com/user-attachments/files/19368707/host_2025-03-20T14-10-38.431Z.log)

bruce607, 2025-03-20:
Here is 2 startups ago (oldest available)

[host_2025-03-20T14-15-29.199Z.log](https://github.com/user-attachments/files/19368795/host_2025-03-20T14-15-29.199Z.log)

bruce607, 2025-03-20:
Yes, as I said previously, it is connected to an HDMI monitor, but I'm not sure what it is you want me to do??

You want me to just restart home assistant & watch the screen ?

WladyTee, 2025-03-20:
Maybe you already found it: with 4GB RAM my lntel NUC goes into out-of-memory error with "python3" process.
Once it happened when I opened the File Editor.
I believe it also occurred with SpeedTest checking because reports are not completed.

025-03-20 09:59:15.773 homeassistant kernel: oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/system.slice/docker-690291b07f6683e4b27faad7e82b20c20b4770245816a72890a1b518dbe6facc.scope,task=python3,pid=6465,uid=0
2025-03-20 09:59:15.773 homeassistant kernel: Out of memory: Killed process 6465 (python3) total-vm:4874208kB, anon-rss:2732660kB, file-rss:5064kB, shmem-rss:0kB, UID:0 pgtables:8900kB oom_score_adj:-300
2025-03-20 09:59:15.774 homeassistant systemd[1]: docker-690291b07f6683e4b27faad7e82b20c20b4770245816a72890a1b518dbe6facc.scope: A process of this unit has been killed by the OOM killer.
2025-03-20 09:59:20.598 homeassistant systemd[1]: docker-690291b07f6683e4b27faad7e82b20c20b4770245816a72890a1b518dbe6facc.scope: Deactivated successfully.
2025-03-20 09:59:20.601 homeassistant systemd[1]: docker-690291b07f6683e4b27faad7e82b20c20b4770245816a72890a1b518dbe6facc.scope: Consumed 12min 39.797s CPU time.
2025-03-20 09:59:20.677 homeassistant dockerd[529]: time="2025-03-20T09:59:20.673942526Z" level=info msg="ignoring event" container=690291b07f6683e4b27faad7e82b20c20b4770245816a72890a1b518dbe6facc module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
2025-03-20 09:59:21.132 homeassistant systemd[1]: var-lib-docker-overlay2-6db1bc3e970dc788819a3ee71c22a744ee95aaf452c23cbf983ad78494f28a5b-merged.mount: Deactivated successfully.
2025-03-20 09:59:21.148 homeassistant systemd[1]: mnt-data-docker-overlay2-6db1bc3e970dc788819a3ee71c22a744ee95aaf452c23cbf983ad78494f28a5b-merged.mount: Deactivated successfully.
2025-03-20 09:59:21.893 homeassistant systemd[1]: Started libcontainer container 690291b07f6683e4b27faad7e82b20c20b4770245816a72890a1b518dbe6facc.

sairon, 2025-03-20:
@bruce607 Keyboard doesn't need to be connected, just network and HDMI. Install the update and watch the output, once it boots into the new system, a kernel error message should appear shortly before the reboot. If it happens and you could capture it with a camera, it would help a lot.

@WladyTee Your problem is unrelated to the issue discussed here, please stop spamming random issues and open dedicated one with proper report what's happening. However, in your case it seems some integration in Core causes the OOM issues.

bruce607, 2025-03-20:
ok, but (forgive my ignorance), since I told it to skip the update, it no longer appears, so, how do I get it to show me the update again ?

Thanks!!

sairon, 2025-03-20:
Three dots -> Show skipped updates on the [Updates](https://my.home-assistant.io/redirect/updates/) page.

endiz, 2025-03-21:
Just came here to say my update to OS 15 failed on x86 hardware, and after 3 or 4 hard reboots, the OS reverted back to 14.2 (thankfully). Will wait for 15.1 to try again. If any devs are curious I can upload logs if you tell me what to grab.

parisdalas, 2025-03-21:
> Three dots -> Show skipped updates on the [Updates](https://my.home-assistant.io/redirect/updates/) page.

If we try to update to v15 again since we have one "bad slot" it will not brick the other slot as well? Is there any chance we get two bad slots?

reini-1, 2025-03-21:
I did some boot videos and fotos and copied text output.

i hope that helps fixing the problem.

HAOS 15.0 failed boot: https://youtu.be/rWcpJ9spLL0
HAOS 14.2 boot: https://youtu.be/I8JEd9HRlUY

After the update and the first boot I could for a short time enter commands at the console and connected via ssh, after a while it got crazy.

Here is some captured output:

[image]

[image]

[image]

```shell
$ ssh homeassistant-os

Welcome to Home Assistant OS.

Use `ha` to access the Home Assistant CLI.
# dm
dmesg     dmeventd  dmsetup   dmstats
# dmesg
[    0.000000] Linux version 6.12.18-haos (builder@644e17d3fe20) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -g3914f8ca) 12.4.0, GNU ld (GNU Binutils) 2.40) #1 SMP PREEMPT_DYNAMIC Mon Mar 17 11:16:39 UTC 2025
[    0.000000] Command line: BOOT_IMAGE=(hd0,gpt2)/bzImage root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootwait zram.enabled=1 zram.num_devices=3 systemd.machine_id=45a4d61ab26b42e3895f2bd890b6c59d fsck.repair=yes console=tty1 rauc.slot=A
[    0.000000] x86/split lock detection: #AC: crashing the kernel on kernel split_locks and warning on user-space split_locks
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007263ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000072640000-0x000000007573ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000075740000-0x0000000075814fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000075815000-0x00000000758d4fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000758d5000-0x0000000075ffefff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000075fff000-0x0000000075ffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000076000000-0x0000000079ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007aa00000-0x000000007abfffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007b000000-0x00000000803fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000c0000000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000047fbfffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] APIC: Static calls initialized
[    0.000000] efi: EFI v2.8 by American Megatrends
[    0.000000] efi: ACPI=0x75814000 ACPI 2.0=0x75814014 TPMFinalLog=0x75882000 SMBIOS=0x75d8e000 SMBIOS 3.0=0x75d8d000 MEMATTR=0x6f6a1298 ESRT=0x6d915b98 TPMEventLog=0x75766018
[    0.000000] efi: Remove mem67: MMIO range=[0xc0000000-0xcfffffff] (256MB) from e820 map
[    0.000000] e820: remove [mem 0xc0000000-0xcfffffff] reserved
[    0.000000] efi: Not removing mem68: MMIO range=[0xfe000000-0xfe010fff] (68KB) from e820 map
[    0.000000] efi: Not removing mem69: MMIO range=[0xfec00000-0xfec00fff] (4KB) from e820 map
[    0.000000] efi: Not removing mem70: MMIO range=[0xfed00000-0xfed00fff] (4KB) from e820 map
[    0.000000] efi: Not removing mem72: MMIO range=[0xfee00000-0xfee00fff] (4KB) from e820 map
[    0.000000] efi: Remove mem73: MMIO range=[0xff000000-0xffffffff] (16MB) from e820 map
[    0.000000] e820: remove [mem 0xff000000-0xffffffff] reserved
[    0.000000] [Firmware Bug]: Corrupted DMI table, offset 5263 (only 77 entries processed)
[    0.000000] SMBIOS 3.5.0 present.
[    0.000000] DMI:  /, BIOS 5.26 09/26/2023
[    0.000000] DMI: Memory slots populated: 4/8
[    0.000000] tsc: Detected 800.000 MHz processor
[    0.000000] tsc: Detected 806.400 MHz TSC
[    0.000219] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[... 2485 lines cut ...]
[  400.924363] systemd-journald[115]: Failed to write entry to /var/log/journal/45a4d61ab26b42e3895f2bd890b6c59d/system.journal (28 items, 1143 bytes) despite vacuuming, ignoring: Input/output error (Dropped 12 similar message(s))
[  401.936001] systemd-journald[115]: Failed to write entry to /var/log/journal/45a4d61ab26b42e3895f2bd890b6c59d/system.journal (28 items, 1143 bytes) despite vacuuming, ignoring: Input/output error (Dropped 9 similar message(s))
[  402.947533] systemd-journald[115]: Failed to write entry to /var/log/journal/45a4d61ab26b42e3895f2bd890b6c59d/system.journal (28 items, 1143 bytes) despite vacuuming, ignoring: Input/output error (Dropped 22 similar message(s))
[  403.957885] systemd-journald[115]: Failed to write entry to /var/log/journal/45a4d61ab26b42e3895f2bd890b6c59d/system.journal (28 items, 1143 bytes) despite vacuuming, ignoring: Input/output error (Dropped 9 similar message(s))
[  404.969155] systemd-journald[115]: Failed to write entry to /var/log/journal/45a4d61ab26b42e3895f2bd890b6c59d/system.journal (28 items, 1143 bytes) despite vacuuming, ignoring: Input/output error (Dropped 12 similar message(s))
[... cut ...]
[  416.092840] systemd-journald[115]: Failed to write entry to /var/log/journal/45a4d61ab26b42e3895f2bd890b6c59d/system.journal (28 items, 1143 bytes) despite vacuuming, ignoring: Input/output error (Dropped 9 similar message(s))
#
#
#
# ha host reboot
Error response from daemon: open /mnt/data/docker/overlay2/8bc642fa8e80f237b520538da57713ba0ffd0ec91123573491967cd3f34b79a5/lower: input/output error
# reboot
# Connection to homeassistant closed by remote host.
Connection to homeassistant closed.
```

endiz, 2025-03-21:
> Three dots -> Show skipped updates on the [Updates](https://my.home-assistant.io/redirect/updates/) page.

@sairon does updating through the GUI include the latest changes in the repo? or we have to wait for a release for the built in updater to reflect the latest PR changes (like https://github.com/home-assistant/operating-system/pull/3957)

eljenso, 2025-03-24:
@endiz Updating through the GUI will give you the latest release, which is currently 15.0 (see https://github.com/home-assistant/operating-system/releases).
You have to wait till 15.1 is release to get the current changes from the repo.

I'm currently waiting for that release, just to be safe and not run into this issue here.

sairon, 2025-03-24:
@reini-1 Thank you, that helps a lot. The changed LPM policy can be very much likely the cause, as @comtel2000 pointed out.

Can you (or anyone else having these issues) try running the 15.0 slot with `libata.force=nolpm` parameter? The procedure to add it temporarily via the GRUB menu is same as [here](https://github.com/home-assistant/operating-system/issues/3944#issuecomment-2734343133).

sairon, 2025-03-24:
@parisdalas I see you have the same SSD (N900-512). It appears to be some random Chinese SSD and it's been known to be problematic for quite a while already, e.g. here's someone who ran into it in June on Arch: <https://bbs.archlinux.org/viewtopic.php?id=296463>

For @bruce607, it's quite possible it's the same cause, as his PC appears to be using a SSD that's branded `512 GB SSD`. Adding the kernel parameter per the instructions above might resolve it too.

Regarding others, at least output of `ha hardware info` from the running slot would be needed.

bruce607, 2025-03-24:
SO, Do I need to replace the ssd with a better one?  Will that resolve the issue, or would it be better to replace the entire NUC with something better??  Any advice would be appreciated!!

BeJaHome, 2025-03-24:
@bruce607 I have the same problem with RPi4 with good quality SSD. So I would keep the NUC and SSD, and wait for 15.1 to arrive. And keep your fingers crossed that the programmers have solved the problems… At least that’s what I do.

dkrizanek, 2025-03-27:
> The switch to version 15 from 14.2 failed. The error message was 'home assistant cli not starting jump into emergency mode'. The machine is a hp thinclient with 8GB ram, and 128GB SSD.

Has the same issue, had to go back to 14.2

AMD APU, 8GB RAM
Sandisk SSD
Sonoff Zigbee 3.0 dongle

BeJaHome, 2025-03-31:
@bruce607 probably not your SSD that is flaky. My RPi4 with brand new SSD just failed updating to 15.1 from 14.2 (was fallen back after failing 15.0 update), now reporting it’s running 14.1, with a notification that 15.1 is available. 
Presumably the earlier 15.0 update attempt broke something, so I might be stuck at 14.1 forever, because I don’t expect that a near future 15.2 and beyond will solve things for me.
The current 14.1 version seems to run fine for now, so I leave it for the time being. Nice job for a rainy day: format my SSD and load a fresh 15.1 image from Github, and then restore my Core backup.…

bruce607, 2025-03-31:
Mine is showing 14.2, SO, just ordered a new nvme to replace the sata ssd & will try cloning with Balena etcher & see if that resolves the issue.

avandorp, 2025-03-31:
My old Intel NUC finally managed to install 15.1, but it took several attempts and I don't know why it finally worked.

Eraser127, 2025-04-01:
I can confirm that HA OS 15.1 is working just fine on my HP T520.

BartRoels, 2025-04-01:
Confirming I have been able to install HA OS 15.1 without any issues on my Dell Wyse 5070

comtel2000, 2025-04-01:
I think 2 different problems are being mixed here.

1. amdgpu SI and CIK support (fixed by OS 15.1 #3957)
2. LPM policy error (...still open?)

The probability is high that the LPM policy issue can be eliminated with a new nvme/SSD. This problem mainly affects certain SSDs with faulty LPM implementation.
@bruce607  please try [`libata.force=nolpm`](https://github.com/home-assistant/operating-system/issues/3948#issuecomment-2747656402) before scrapping your exist hardware.

WladyTee, 2025-04-01:
I can confirm that HA OS 15.1 works fine on my old Intel NUC 5CPYH - 4GB RAM and 128GB SSD Kingston.
No VM.

Starting from HA OS 15, I experienced continuous out-of-memory crashes (python3 in the task list) with reboots every 10-25 mins for the last 2 weeks.
I disabled several add-ons / integrations to debug the issue without success (never restored any backup), I just downgraded to previous CORE 2025.3.1 [ha core update --version 2025.3.1] while working with OS 14.2 in the failover partition. 
Yesterday applied HA OS 15.1 followed by CORE 2025.3.4.

I managed to make the system stable after [ha supervisor repair] that finally stopped the rolling crash down.

Still issues with BLUETOOTH, will be probably fixed with incoming CORE 2025.4.
Checking some issues with "Timeout on call http://172.30.32.1:8123/api/core/state".
In my case, I believe that a combined set of small issues caused some unstable behavings.

I hope it helps.

yanivf38, 2025-04-01:
I was able to update the OS by adding libata.force=nolpm during boot, but that won't work if we need to do it every time.

@bruce607, here are the steps:
1. Connect your Beelink to a screen and keyboard.
2. In home assistant, click to update the OS (as you would normally do)
3. Keep an eye on the screen connected to your Beelink until the machine reboots. During this time, repeatedly press the **Del** key to access the GRUB menu.
4. [Bonus step] Note the ***** in the list of partitions. This tells you which partition it's going to boot into. (This is needed as if something goes wrong you can repeat this steps but select the 'other partition' and boot, which will put you back into the working home assistant)
5. Press **e** to edit the boot options.
6. In the Linux line, add libata.force=nolpm at the end of the line (_not in new line_), as shown in @reini-1's comment (https://github.com/home-assistant/operating-system/issues/3948#issuecomment-2749226877).
7. After adding the command, press **F10** to continue loading and updating the OS.
8. When it will done booting hopefully it will put you in the new OS

sairon, 2025-04-02:
@yanivf38 If you confirmed the LPM is indeed the source of the trouble, you can persist the change in `cmdline.txt` by following steps in [this post from user pda](https://community.home-assistant.io/t/home-assistant-unreachable-after-update-to-haos-15/865003/106?u=sairon), just substituting `modprobe.blacklist=amdgpu` with `libata.force=nolpm`.

For some SSDs a persistent fix could be submitted to the kernel, however, if it's a knockoff brand with a bogus identifier, the chance it will be accepted in the kernel is next to none. **If you happen to have these issues, send what `data_disk` line says in `ha os info`, we can investigate further from there.**

yanivf38, 2025-04-02:
Thank you @sairon, I'll follow the steps. My data_disk: 512GB-SSD-CM43ABH2300479

Do you have thoughts of why the power saving flag work-around the issue? I have been running HA on this hardware for 2 years and had no problem until OS 15.

bruce607, 2025-04-02:
My data_disk is : 512GB-SSD-CM379BH1211077

Have not tried the libata.force=nolpm yet, is there any point, or should I just wait & restore my backup to the new Odroid H4 when it arrives ?

reini-1, 2025-04-02:
My cmdline.txt looks now like this:

```shell
# cat /mnt/boot/cmdline.txt
console=tty1 libata.force=nolpm
```

that results in the following overall cmdline:

```shell
# cat /proc/cmdline
BOOT_IMAGE=(hd0,gpt4)/bzImage root=PARTUUID=a3ec664e-32ce-4665-95ea-7ae90ce9aa20 rootwait zram.enabled=1 zram.num_devices=3 systemd.machine_id=45a4d61ab26b42e3895f2bd890b6c59d fsck.repair=yes console=tty1 libata.force=nolpm rauc.slot=B
```

with this I can boot 15.x:

```shell
# ha os info
board: generic-x86-64
boot: B
boot_slots:
  A:
    state: inactive
    status: good
    version: "15.0"
  B:
    state: booted
    status: good
    version: "15.1"
data_disk: N900-512-AA000000000000003217
update_available: false
version: "15.1"
version_latest: "15.1"
```

yanivf38, 2025-04-02:
I faced a similar dilemma—whether to upgrade to a new NUC for better performance and an NVMe drive instead of SATA. Ultimately, I decided to stick with my current setup for now since the NUC is already powerful enough. If the issue is just about preventing the kernel from putting the drive into power-saving mode, then that's fine. My only question, as I mentioned earlier, is whether we optimized something in OS 15 that caused this power-saving behavior.

NemesisRE, 2025-04-04:
Same problem here some china sata m.2 ssd upgrade fails, cloned onto transcend nvme and it works like a charm.

Tobster77, 2025-04-04:
I suffer from similar symptoms on a HP T 620 machine, running HASS 14.2. The update to 15.1 fails, but after three hardware reboots, the machine becomes available again and offers me the 15.1 update again. I reported this also [here](https://github.com/home-assistant/operating-system/discussions/3990)  and [here](https://community.home-assistant.io/t/update-hass-14-2-to-15-1-fails-on-x86-hp-t620-machine/873507)

May I hope this is resolved in 15.2 and simply wait?

Thanks!

yanivf38, 2025-04-04:
@Tobster77 follow the steps (they are easier than you think)  -
https://github.com/home-assistant/operating-system/issues/3948#issuecomment-2770837367

Once this works, you can make a permanent change by following the steps here - 
https://community.home-assistant.io/t/home-assistant-unreachable-after-update-to-haos-15/865003/106?u=sairon

Adminius, 2025-04-09:
I'm using Fujitsu S920, simmilar to HP T620.
15.0 can't start because of amdgpu-driver thing -> 15.1
15.1 stops at Swap something while booting. 
cmdline trick didn't help.

I'll stay with 14.2 that works perfectly until all bugs will be fixed.

Tobster77, 2025-04-10:
> [@Tobster77](https://github.com/Tobster77) follow the steps (they are easier than you think) - [#3948 (comment)](https://github.com/home-assistant/operating-system/issues/3948#issuecomment-2770837367)
> 
> Once this works, you can make a permanent change by following the steps here - https://community.home-assistant.io/t/home-assistant-unreachable-after-update-to-haos-15/865003/106?u=sairon

@yanivf38 Sorry, I am basically able to work with a linux command line, but the container-based thing of HA is beyond my experience.

I would need something starting from HA GUI. I have just installed the SSH add-on ... now what? Any chance to simply wait for 15.2 within the next few weeks?

pekkeri, 2025-06-05:
I had also the boot loop problem after update from 14.2 to 15.x.
I was finally able to boot stable to 15.2 with the mentioned `libata.force=nolpm` added to `cmdline.txt`

This is a Beelink U59 mini PC, Celeron N5105, 16GB RAM:
`board: generic-x86-64`
`data_disk: 512GB-SSD-CM42ABH1801565`

Nab0y, 2025-06-07:
> I had also the boot loop problem after update from 14.2 to 15.x. I was finally able to boot stable to 15.2 with the mentioned `libata.force=nolpm` added to `cmdline.txt`
> 
> This is a Beelink U59 mini PC, Celeron N5105, 16GB RAM: `board: generic-x86-64` `data_disk: 512GB-SSD-CM42ABH1801565`

The same hardware, the same problem.

closed 2025-09-14 by github-actions[bot]

BeJaHome, 2025-09-14:
I solved this boot problem by replacing my SATA-USB adapter.
It appeared that using the Patriot P220 SSD with JMicron JMS578 adapter on a RPi4 is not the best combination.
Replacing the JMS578 by the StarTech SATA–USB adapter solved all my boot problems. It runs (and boots) like a charm now.
It seems that the JMS578 adapter might also give boot problems with other SSDs on RPi(4).

Groet,
John Janken

(sent from mobile)

Closed #3948<https://github.com/home-assistant/operating-system/issues/3948> as not planned.