home-assistant/operating-system #3950: Network Storage SMB/CIFS compatibility issue/not working with some servers HAOS 14.2/15

radulyz, 2025-03-18:
### Describe the issue you are experiencing

I noticed only after updating to HAOS 15: I'm running into trouble connecting to my Mikrotik NAS: SMB/CIFS connection from the Storage/Network Storage setting of the HA...
Since the update even tho the Mikrotik server logs show a successful login from HA, the network storage doesn’t work. (CIFS protocol, server, share, user and password are all good, the server is confirming that)
Current system: OS version: 15; HA core: 2025.3.3

From the logs:
```
2025-03-17` 22:00:12.932 homeassistant systemd[1]: Mounting Supervisor cifs mount: MikrotikSMB…
2025-03-17 22:00:12.946 homeassistant kernel: CIFS: Attempting to mount [//192.168.30.8/backup-ha] https://192.168.30.8/backup-ha)
2025-03-17 22:00:13.074 homeassistant kernel: CIFS: VFS: cifs_read_super: get root inode failed
2025-03-17 22:00:13.081 homeassistant mount[20512]: mount error(13): Permission denied
2025-03-17 22:00:13.081 homeassistant mount[20512]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
2025-03-17 22:00:13.085 homeassistant systemd[1]: mnt-data-supervisor-mounts-MikrotikSMB.mount: Mount process exited, code=exited, status=32/n/a
2025-03-17 22:00:13.086 homeassistant systemd[1]: mnt-data-supervisor-mounts-MikrotikSMB.mount: Failed with result ‘exit-code’.
2025-03-17 22:00:13.086 homeassistant systemd[1]: Failed to mount Supervisor cifs mount: MikrotikSMB.
```
After downgrading OS to 14.2 the problem still exists...
I can't exactly pinpoint when the problem started, I know it worked with the same server before maybe half a year ago, I tested it, but I chose not to use the functionality at that point.
Interestingly, the Samba Backup AddOn is working with no issues with the same credentials and the same share/Mikrotik NAS.

What I noticed is that If in the Windows Group Policy [Network security: LAN Manager authentication level]
is set to any of the following 
```
- Send LM & NTLM responses
- Send LM & NTLM - use NTLMv2 session security if negotiated
- Send NTLM responses only
```
The Mikrotik NAS will behave the same way as with the HA when it tries to connect to it (unsuccessfully)
but when the [Network security: LAN Manager authentication level] is set to either of the following:
```
- Send NTLMv2 responses only
- Send NTLMv2 responses only. Refuse LM
- Send NTLMv2 responses only. Refuse LM & NTLM
- Not Defined
```
The windows machine will connect with no issues to the Mikrotik NAS...
So the HA might also have something to do with these communication protocol settings.

Any suggestion where I could change any of these settings in the HA to get the Network Storage working again?

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

15

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

PI4, 8GB RAM, 120GB Kingston SSD is used for the operating system (HAOS)

### Steps to reproduce the issue

1. Tried downgrading HAOS to 14.2, but not sure from which version the problem started I just know it worked 2024 November
2. Tried to connect from the same system with Samba Backup AddOn: that's working with no issues, but I cannot use that in the system as an automatic backup option
3. Tested the NAS in various scenarios to see what the issue could be, it seems like a communication protocol issue.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2025-03-17` 22:00:12.932 homeassistant systemd[1]: Mounting Supervisor cifs mount: MikrotikSMB…
2025-03-17 22:00:12.946 homeassistant kernel: CIFS: Attempting to mount [//192.168.30.8/backup-ha] https://192.168.30.8/backup-ha)
2025-03-17 22:00:13.074 homeassistant kernel: CIFS: VFS: cifs_read_super: get root inode failed
[... 122 lines cut ...]
current_recorder_run | March 18, 2025 at 00:34
estimated_db_size | 595.42 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

<details><summary>Sonoff</summary>

version | 3.8.2 (c4b6fda)
-- | --
cloud_online | 7 / 10
local_online | 1 / 1

</details>

### Additional information

NAS Server LOGs on successful login when using Samba Backup AddOn (when the backup was performed: ~450MB):

[image]

NAS Server LOGs on UNsuccessful connection when using the Network Storage (even tho the login is OK, still error from the HA on 2 separate ocations):

[image]

supperka, 2025-03-18:
also have problems with samba connection, to my share disk.

```
Logger: homeassistant.components.hassio
Zdroj: components/hassio/websocket_api.py:141
integrace: Home Assistant Supervisor (dokumentace, problémy)
První výskyt: 20:41:33 (12 výskyty)
Naposledy logováno: 20:55:40
```
```
Failed to to call /mounts/share - Mounting share did not succeed. Check host logs for errors from mount or systemd unit mnt-data-supervisor-mounts-share.mount for details.
```

Failed to to call /mounts/share - Mounting share did not succeed. Check host logs for errors from mount or systemd unit mnt-data-supervisor-mounts-share.mount for details. Hour before update all works great.

brugk000, 2025-03-19:
I have the same error. After an unsuccesfull update to 15.0 there was a problem connecting a samba share to my Unraid server.
I am having this errors now:
2025-03-19 10:55:22.705 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)

TrilIiard, 2025-03-20:
Same errors on rpi4 4gb with 15.0, smb stopped working. Sadly it's not even possible to go back to previous version with existing backups. No matter how old the backup is, a day (14.3) or a month (14.1) it doesn't restore to that version and always reboots to 15.0

supperka, 2025-04-01:
Version 15.1 isn't fix my problem.

radulyz, 2025-04-01:
Yeah, the issue persists. I don't know where it comes from or if it's from HA or HAOS. As I said, with the same server credentials (and server), Samba Backup (v 5.2.0) works without issues. 90% is from the [Network security: LAN Manager authentication level] communication mismatch...

1kemp, 2025-04-17:
After the upgrade to HAOS 15.0 my network storage CIFS to a WIN11 machine stopped working. If I try to re-add the storage I get this error

2025-04-17 16:23:13.544 homeassistant systemd[1]: Mounting Supervisor cifs mount: TapoRecordings...
2025-04-17 16:23:13.547 homeassistant kernel: CIFS: Attempting to mount //192.168.0.106/TapoRecordings
2025-04-17 16:23:13.605 homeassistant kernel: CIFS: VFS: cifs_read_super: get root inode failed
2025-04-17 16:23:13.608 homeassistant mount[209966]: mount error(95): Operation not supported
2025-04-17 16:23:13.608 homeassistant mount[209966]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
2025-04-17 16:23:13.609 homeassistant systemd[1]: mnt-data-supervisor-mounts-TapoRecordings.mount: Mount process exited, code=exited, status=32/n/a
2025-04-17 16:23:13.609 homeassistant systemd[1]: mnt-data-supervisor-mounts-TapoRecordings.mount: Failed with result 'exit-code'.
2025-04-17 16:23:13.610 homeassistant systemd[1]: Failed to mount Supervisor cifs mount: TapoRecordings.

Mounting the same with the same credentials works on my Proxmox Server works without issue.

HarriVa, 2025-05-19:
> After the upgrade to HAOS 15.0 my network storage CIFS to a WIN11 machine stopped working. If I try to re-add the storage I get this error
> 

After HAOS upgrade I have the same problem with Win 11 machine, but strangely it only happens when I share an OneDrive folder from Windows. Non-OneDrive folder shares are accessible from HAOS.

techjunkie91, 2025-07-11:
> After HAOS upgrade I have the same problem with Win 11 machine, but strangely it only happens when I share an OneDrive folder from Windows. Non-OneDrive folder shares are accessible from HAOS.

I get the same issue.
Any folder outside of the OneDrive location works fine.