home-assistant/operating-system #3952: OS 15.0 no longer displays IP Addresses on Welcome Page

GSVil, 2025-03-18:
### Describe the issue you are experiencing

Even though my HAOS installation on Intel NUC still works, the welcome page no longer displays the IP addresses. My Network config has hardcoded IP/Gateway/DNS configured in HA Network. It always displayed the IP addresses, even though I have hardcoded them.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

15

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Intel NUC, with Zigbee Dongle plugged into the USB, as well as Bluetooth Dongle running on Ethernet-Port. Standard HAOS installation.

### Steps to reproduce the issue

Nothing that i can see

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing that i can see, just the normal info records.
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing looks out of the ordinary
```

### Additional information

Everything appears to work, its just the IP's that are not showing up.

sairon, 2025-03-19:
Was it welcome page on the HDMI console or when accessed remotely? Does it still not show even if you type `ha banner` when the network is connected?

GSVil, 2025-03-19:
> Was it welcome page on the HDMI console or when accessed remotely? Does it still not show even if you type `ha banner` when the network is connected?

It does **NOT** show on the monitor connected via HDMI after boot-up.
it does show the IP-Addresses when I SSH in remotely.
If type in command 'ha banner' it also shows the IP-Addresses.

sairon, 2025-03-19:
That would mean the device received the IP lease after the CLI has started. Usually it should not be the case, as the NetworkManager should initialize the network interface earlier in the boot. Why does it take longer in your case would need full [Host logs](https://my.home-assistant.io/redirect/logs/?provider=host) for the boot (e.g. scroll up to the point where the device starts to boot and then clicking the download to download whatever the number of lines it shows). It would be also worth checking it wasn't a fluke and it's readily reproducible.

GSVil, 2025-03-19:
Could not reproduce, closed.

closed 2025-03-19 by GSVil