home-assistant/operating-system #3953: OS 14.0+ Raspberry Pi 5 doesn't boot after power failure

Ardivia, 2025-03-19:
### Describe the issue you are experiencing

I don't know if it's related to #3842 but after a power cycle Pi 5 won't boot, only containers started in docker are observer and supervisor, was happening with v14.2 too.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

15.0

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

RPI 5

### Steps to reproduce the issue

1. Power cycle a Pi 5

### Anything in the Supervisor logs that might be useful for us?

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/udev.sh
[09:03:16] INFO: Using udev information from host
cont-init: info: /etc/cont-init.d/udev.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun supervisor (no readiness notification)
services-up: info: copying legacy longrun watchdog (no readiness notification)
[09:03:16] INFO: Starting local supervisor watchdog...
s6-rc: info: service legacy-services successfully started
2025-03-19 09:03:18.582 INFO (MainThread) [__main__] Initializing Supervisor setup
2025-03-19 09:03:18.661 INFO (MainThread) [supervisor.coresys] Setting up coresys for machine: raspberrypi5-64
2025-03-19 10:03:18.667 INFO (MainThread) [supervisor.docker.supervisor] Attaching to Supervisor ghcr.io/home-assistant/aarch64-hassio-supervisor with version 2025.03.3
2025-03-19 10:03:18.686 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state initialize
2025-03-19 10:03:18.688 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2025-03-19 10:03:18.689 INFO (MainThread) [__main__] Setting up Supervisor
2025-03-19 10:03:18.781 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
s6-rc: info: service legacy-services: stopping
[09:32:26] INFO: Watchdog restart after closing
s6-svwait: fatal: timed out
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

### Anything in the Host logs that might be useful for us?

```txt
can't access
```

### System information

can't access

### Additional information

docker ps output:

```
00a18b8554c1   ghcr.io/home-assistant/aarch64-hassio-observer:2023.06.0   "/usr/bin/observer"   12 hours ago   Up 16 minutes   0.0.0.0:4357->80/tcp, [::]:4357->80/tcp   hassio_observer
0815529325c2   ghcr.io/home-assistant/aarch64-hassio-supervisor:latest    "/init"               12 hours ago   Up 16 minutes                                             hassio_supervisor

```

Ardivia, 2025-03-19:
Happens even with a reboot so problem #3842 not resolved.

sairon, 2025-03-19:
The issue you refer to is about getting stuck in the bootloader, that's not what you're seeing. Is there anything suspicious in `systemctl --failed` or `dmesg -l warn+`?

sairon, 2025-03-19:
That seems to be the same problem as seen here: <https://github.com/home-assistant/operating-system/issues/3943#issuecomment-2735997431>

I am trying to reproduce these issues here at no avail, what is your bootloader version and RPi model? The bootloader version is shown on the display when you run the Pi without any boot media but it can be also obtained in less-readable form from `xxd /proc/device-tree/chosen/bootloader/build-timestamp`. Eventually, when you update it from RPi OS with `rpi-eeprom-update -a`, it's shown there too.

Anyway, it seems that updating the bootloader helps but knowing what version you are running would be helpful.

sairon, 2025-03-19:
Yep, that's indeed timestamp for 2024-06-05. I'll try checking how that version behaves here, but you should definitely update it.

Asking about the RPi model, I also meant what RAM size it has, as higher RAM models (or newer makes with BCM2712D0 revision) had some issues with older bootloaders too.

Ardivia, 2025-03-19:
I have 2GB version.

sairon, 2025-03-26:
Should be fixed by #3972 (or by updating the bootloader).

closed 2025-03-26 by sairon