home-assistant/operating-system #3960: Compreface add-on for HAOS broken after 15.0 update

evandroid, 2025-03-20:
### Describe the issue you are experiencing

I was using double take with compreface add-on (single container) just fine.

After 15.0 update, compreface add-on stopped working. 

CORE
2025.3.3

SUPERVISOR
2025.03.3

HA OS
15.0

FRONTEND:
20250306.0

It appears to be a problem with ProcessorMetrics bean of Micrometer. Tomcat cant properly initialize on SpringBoot.

HAOS 15.0 probably is not authorizing compreface to access such data (cpu metrics) anymore from compreface single container version in its actual version (?)

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

15.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Using proxmox 8.3.5
not LXC, using VM. 
sonoff zigbee 3.0 plugged

### Steps to reproduce the issue

1. start compreface add-on
2. watch logs
3. spring boot enters in loop regarding processorMetrics bean

### Anything in the Supervisor logs that might be useful for us?

```txt
in supervisor logs, there's nothing about it.
```

### Anything in the Host logs that might be useful for us?

```txt
nothing on host logs regarding this issue
```

### System information

## System Information

version | core-2025.3.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.12.18-haos
arch | x86_64
timezone | America/Sao_Paulo
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1663
Downloaded Repositories | 11

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.0
-- | --
update_channel | stable
supervisor_version | supervisor-2025.03.3
agent_version | 1.7.2
docker_version | 28.0.1
disk_total | 1007.2 GB
disk_used | 91.8 GB
healthy | true
supported | failed to load: Unsupported
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Spotify Connect (0.14.0), Advanced SSH & Web Terminal (20.0.2), Assist Microphone (1.3.0), openWakeWord (1.10.0), Studio Code Server (5.18.3), Samba share (12.5.0), Grocy (0.24.0), Mosquitto broker (6.5.0), Node-RED (19.0.2), Double Take (1.13.1), Zigbee2MQTT (2.1.3-1), Exadel CompreFace (1.1.0), Frigate Proxy (1.5), ESPHome Device Builder (2025.3.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 5
-- | --
resources | 7
views | 3
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp0s18 (enabled, default, auto), docker0 (disabled), hassio (disabled), veth196631b (disabled), veth369abde (disabled), veth4dcde95 (disabled), veth3e8d3b7 (disabled), vethfb3e3ef (disabled), veth8d489a0 (disabled), veth57a707e (disabled), vethabc613c (disabled), veth633ffeb (disabled), vethe36d411 (disabled), vethdcfbf96 (disabled), veth70c4230 (disabled), veth9a674ed (disabled), vethf5579fb (disabled), veth444a991 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp0s18 (192.168.159.154/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), veth196631b (), veth369abde (), veth4dcde95 (), veth3e8d3b7 (), vethfb3e3ef (), veth8d489a0 (), veth57a707e (), vethabc613c (), veth633ffeb (), vethe36d411 (), vethdcfbf96 (), veth70c4230 (), veth9a674ed (), vethf5579fb (), veth444a991 ()
ipv6_addresses | lo (::1/128), enp0s18 (fe80::fd25:da92:ec29:812b/64), docker0 (fe80::7469:67ff:fe55:cf8/64), hassio (fe80::8c1d:3eff:fe0a:a3a3/64), veth196631b (fe80::1017:efff:fea6:d5fe/64), veth369abde (fe80::3c6f:8eff:fe6e:4aa5/64), veth4dcde95 (fe80::541e:e9ff:fe4e:79c6/64), veth3e8d3b7 (fe80::2cfe:b0ff:fe1e:20e9/64), vethfb3e3ef (fe80::4cd3:4cff:fedb:a892/64), veth8d489a0 (fe80::cc90:47ff:fe85:6a36/64), veth57a707e (fe80::748e:f3ff:fe38:f879/64), vethabc613c (fe80::98e1:f8ff:feb8:c695/64), veth633ffeb (fe80::882a:5aff:fea5:8390/64), vethe36d411 (fe80::488f:8fff:fe7f:a8b0/64), vethdcfbf96 (fe80::8cf7:ffff:fe12:4995/64), veth70c4230 (fe80::e822:68ff:febc:5a70/64), veth9a674ed (fe80::ccf3:eeff:fe3e:825/64), vethf5579fb (fe80::3c16:5cff:fe99:3276/64), veth444a991 (fe80::1c9d:46ff:fe06:8ee3/64)
announce_addresses | 192.168.159.154, fe80::fd25:da92:ec29:812b

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | March 8, 2025 at 03:57
-- | --
current_recorder_run | March 19, 2025 at 14:56
estimated_db_size | 294.93 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

chheiss, 2025-03-20:
We see the problem as well after upgrade to OS 15.

chheiss, 2025-03-21:
Downgrading to HA OS 14.2 solves the issue. So the problem is with the HA 15 upgrade.

DAVIZINH0, 2025-03-21:
same problem. the compreface enter in a bootloop mode with a hight cpu usage. 
happends after update to HAOS 15

in the logs of compreface apears info like this:

`2025-03-21 20:53:33.751 ERROR 1970 --- [           main] o.s.b.web.embedded.tomcat.TomcatStarter  : Error starting Tomcat context. Exception: org.springframework.beans.factory.UnsatisfiedDependencyException. Message: Error creating bean with name 'webMvcMetricsFilter' defined in class path resource [org/springframework/boot/actuate/autoconfigure/metrics/web/servlet/WebMvcMetricsAutoConfiguration.class]: Unsatisfied dependency expressed through method 'webMvcMetricsFilter' parameter 0; nested exception is org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'simpleMeterRegistry' defined in class path resource [org/springframework/boot/actuate/autoconfigure/metrics/export/simple/SimpleMetricsExportAutoConfiguration.class]: Initialization of bean failed; nested exception is org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'processorMetrics' defined in class path resource [org/springframework/boot/actuate/autoconfigure/metrics/SystemMetricsAutoConfiguration.class]: Bean instantiation via factory method failed; nested exception is org.springframework.beans.BeanInstantiationException: Failed to instantiate [io.micrometer.core.instrument.binder.system.ProcessorMetrics]: Factory method 'processorMetrics' threw exception; nested exception is java.lang.NullPointerException
2025-03-21 20:53:33.774  INFO 1970 --- [           main] o.apache.catalina.core.StandardService   : Stopping service [Tomcat]
2025-03-21 20:53:33.786  WARN 1970 --- [           main] ConfigServletWebServerApplicationContext : Exception encountered during context initialization - cancelling refresh attempt: org.springframework.context.ApplicationContextException: Unable to start web server; nested exception is org.springframework.boot.web.server.WebServerException: Unable to start embedded Tomcat
2025-03-21 20:53:33.802  INFO 1970 --- [           main] ConditionEvaluationReportLoggingListener : 
Error starting ApplicationContext. To display the conditions report re-run your application with 'debug' enabled.
2025-03-21 20:53:33.827 ERROR 1970 --- [           main] o.s.boot.SpringApplication               : Applicat`

SomeoneSinister, 2025-03-22:
above is same issue for me.
compreface stuck in boot loop with high CPU usage. reverting to 14.2 solved the problem.

renaatdb, 2025-03-23:
Sae here

adultano, 2025-03-23:
same here
Core
2025.3.4
Supervisor
2025.03.3
Operating System
15.0
Frontend
20250306.0

macje1968, 2025-03-24:
Same problem here

sairon, 2025-03-24:
@evandroid is correct, it appears to be a problem in the 1.1.0 version of Compreface. It even appears when I run it on my desktop running kernel 6.13.8 when I start the vanilla compreface container: `docker run --rm -ti -p8000:80 exadel/compreface:1.1.0`. Version 1.2.0 doesn't exhibit the same behavior but the [jakowenko/double-take-hassio-addons](https://github.com/jakowenko/double-take-hassio-addons) repo lacks bump to this version. This is definitely something that needs to be fixed there. FWIW here's the complete stack trace of the crash, on which it loops, but I'm not willing to delve into it a lot as the newer version obviously fixes the incompatibility.

```
org.springframework.context.ApplicationContextException: Unable to start web server; nested exception is org.springframework.boot.web.server.WebServerException: Unable to start embedded Tomcat
	at org.springframework.boot.web.servlet.context.ServletWebServerApplicationContext.onRefresh(ServletWebServerApplicationContext.java:163)
	at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:577)
	at org.springframework.boot.web.servlet.context.ServletWebServerApplicationContext.refresh(ServletWebServerApplicationContext.java:145)
	at org.springframework.boot.SpringApplication.refresh(SpringApplication.java:775)
	at org.springframework.boot.SpringApplication.refreshContext(SpringApplication.java:448)
	at org.springframework.boot.SpringApplication.run(SpringApplication.java:339)
	at org.springframework.boot.SpringApplication.run(SpringApplication.java:1365)
	at org.springframework.boot.SpringApplication.run(SpringApplication.java:1354)
	at com.exadel.frs.FrsApplication.main(FrsApplication.java:28)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.springframework.boot.loader.MainMethodRunner.run(MainMethodRunner.java:49)
	at org.springframework.boot.loader.Launcher.launch(Launcher.java:108)
	at org.springframework.boot.loader.Launcher.launch(Launcher.java:58)
	at org.springframework.boot.loader.JarLauncher.main(JarLauncher.java:88)
Caused by: org.springframework.boot.web.server.WebServerException: Unable to start embedded Tomcat
	at org.springframework.boot.web.embedded.tomcat.TomcatWebServer.initialize(TomcatWebServer.java:142)
	at org.springframework.boot.web.embedded.tomcat.TomcatWebServer.<init>(TomcatWebServer.java:104)
	at org.springframework.boot.web.embedded.tomcat.TomcatServletWebServerFactory.getTomcatWebServer(TomcatServletWebServerFactory.java:456)
	at org.springframework.boot.web.embedded.tomcat.TomcatServletWebServerFactory.getWebServer(TomcatServletWebServerFactory.java:204)
	at org.springframework.boot.web.servlet.context.ServletWebServerApplicationContext.createWebServer(ServletWebServerApplicationContext.java:182)
	at org.springframework.boot.web.servlet.context.ServletWebServerApplicationContext.onRefresh(ServletWebServerApplicationContext.java:160)
	... 16 common frames omitted
Caused by: org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'webMvcMetricsFilter' defined in class path resource [org/springframework/boot/actuate/autoconfigure/metrics/web/servlet/WebMvcMetricsAutoConfiguration.class]: Unsatisfied dependency expressed through method 'webMvcMetricsFilter' parameter 0; nested exception is org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'simpleMeterRegistry' defined in class path resource [org/springframework/boot/actuate/autoconfigure/metrics/export/simple/SimpleMetricsExportAutoConfiguration.class]: Initialization of bean failed; nested exception is org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'processorMetrics' defined in class path resource [org/springframework/boot/actuate/autoconfigure/metrics/SystemMetricsAutoConfiguration.class]: Bean instantiation via factory method failed; nested exception is org.springframework.beans.BeanInstantiationException: Failed to instantiate [io.micrometer.core.instrument.binder.system.ProcessorMetrics]: Factory method 'processorMetrics' threw exception; nested exception is java.lang.NullPointerException
	at org.springframework.beans.factory.support.ConstructorResolver.createArgumentArray(ConstructorResolver.java:800)
	at org.springframework.beans.factory.support.ConstructorResolver.instantiateUsingFactoryMethod(ConstructorResolver.java:541)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.instantiateUsingFactoryMethod(AbstractAutowireCapableBeanFactory.java:1352)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBeanInstance(AbstractAutowireCapableBeanFactory.java:1195)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.doCreateBean(AbstractAutowireCapableBeanFactory.java:582)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBean(AbstractAutowireCapableBeanFactory.java:542)
	at org.springframework.beans.factory.support.AbstractBeanFactory.lambda$doGetBean$0(AbstractBeanFactory.java:335)
[... cut ...]
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.springframework.beans.factory.support.SimpleInstantiationStrategy.instantiate(SimpleInstantiationStrategy.java:154)
	... 94 common frames omitted
```

gfn256, 2025-03-24:
I can confirm after testing, that my increased CPU load with Home Assistant VM in Proxmox, was not caused by the HAOS update, but rather the newly applied qemu 9.2 on the server. Anyone rebooting (as in STOPPING & then STARTING) the VM under the new qemu version - will "enjoy" this added CPU load! See explanation & workaround [here](https://forum.proxmox.com/threads/pl%C3%B6tzlich-h%C3%B6here-cpu-last-bei-allen-linux-vms.163694/post-756878).

I refer this also to my original [post](https://github.com/home-assistant/operating-system/issues/3941#issuecomment-2735749946).

MichaelStruck, 2025-03-24:
I am not sure if this will help or confuse the subject, but I have more peaks of high CPU after the upgrade as shown here. Obviously, the 'noise' began after 15.0...All functions of HA are working, but did notice this.

[image]

adultano, 2025-03-25:
i dont think the qemu approach resolves the add-on issue, or rather i wasnt able to fix this. neither with the downgrade of qemu nor with the arg flags hpet off. i still get the error:
**Error creating bean with name 'processorMetrics' defined in class path resource** [org/springframework/boot/actuate/autoconfigure/metrics/SystemMetricsAutoConfiguration.class]: Bean instantiation via factory method failed; nested exception is org.springframework.beans.BeanInstantiationException: Failed to instantiate [io.micrometer.core.instrument.binder.system.ProcessorMetrics]: Factory method 'processorMetrics' threw exception; nested exception is java.lang.NullPointerException

so if anyone was able to run the compreface addon in homeassistant - please share your knowledge.

thanks!

sairon, 2025-03-26:
@adultano *\*sigh\** Yeah, this is why posting unrelated comments to various issues just stirs confusion, and can even result in people doing unnecessary steps that make things just worse. I marked the posts as off-topic.

The bug lies in Compreface image/add-on itself, as I explained in my last post above. This **needs to be fixed by the add-on**, as it's an incompatibility of Compreface 1.1.0 Docker image with newer Linux kernels. If the original add-on doesn't update to 1.2.0, a new fork/maintainer must be found. I'm not going to do that, as I'm not using the add-on and I'm afraid of any fallback the version bump can potentially have.

closed 2025-03-26 by sairon

adultano, 2025-03-26:
Thank you very much for the clarification! I thought I was going mad - but this definitely helps. Thanks again! I'll switch to a desicated lxc version of compreface - although I am anyway not fully satisfied with the detectio results. Thanks again for investing your time!