home-assistant/operating-system #3961: Home assistant Yellow PoE RPi4 -> RPi5 upgrade

stau, 2025-03-20:
### Describe the issue you are experiencing

source: https://community.home-assistant.io/t/home-assistant-yellow-gets-cm5-support-in-haos-14/800621/47?u=sau1

Long story short, the on-board Zigbee module has become invisible after upgrading the compute module from RPi4 to RPi5 on the Home Assistant Yellow PoE version. There seems to be some anecdotal evidence that the non-PoE version is fine. 

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

14.2

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Home assistant yellow PoE + CM5 

### Steps to reproduce the issue

1. Follow the guide to plug in CM5
2. Start HomeAssistant Yellow (the zigbee is already not visible here)
3. Restore backup
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Nope, it looks like the device is not visible to the system at all.
```

### Anything in the Host logs that might be useful for us?

```txt
Nope
```

### System information

## System Information

version | core-2025.3.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.6.62-haos-raspi
arch | aarch64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
[... 777 lines cut ...]
[   92.735660] audit: type=1325 audit(1742259149.132:239): table=nat:88 family=2 entries=1 op=nft_register_rule pid=3143 subj=unconfined comm="iptables"
[   92.735668] audit: type=1300 audit(1742259149.132:239): arch=c00000b7 syscall=211 success=yes exit=600 a0=3 a1=7fd64777a8 a2=0 a3=1 items=0 ppid=631 pid=3143 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-nft-multi" subj=unconfined key=(null)
[   92.735674] audit: type=1327 audit(1742259149.132:239): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D74006E6174002D4100504F5354524F5554494E47002D7000746370002D73003137322E33302E33332E33002D64003137322E33302E33332E33002D2D64706F7274003232002D6A004D415351554552414445
[   92.737927] audit: type=1325 audit(1742259149.136:240): table=filter:89 family=2 entries=1 op=nft_register_rule pid=3145 subj=unconfined comm="iptables"
[   92.930057] eth0: renamed from veth646c7c8
[   92.966258] hassio: port 9(veth1ed4820) entered blocking state
[   92.966265] hassio: port 9(veth1ed4820) entered forwarding state
[   93.089790] hassio: port 10(vethf47b041) entered blocking state
[   93.089798] hassio: port 10(vethf47b041) entered disabled state
[   93.089808] vethf47b041: entered allmulticast mode
[   93.089874] vethf47b041: entered promiscuous mode
[   93.090018] hassio: port 10(vethf47b041) entered blocking state
[   93.090022] hassio: port 10(vethf47b041) entered forwarding state
[   93.286131] eth0: renamed from veth8eb3d53
[   93.689257] hassio: port 11(vethc8c801e) entered blocking state
[   93.689265] hassio: port 11(vethc8c801e) entered disabled state
[   93.689276] vethc8c801e: entered allmulticast mode
[   93.689343] vethc8c801e: entered promiscuous mode
[   93.689758] hassio: port 11(vethc8c801e) entered blocking state
[   93.689763] hassio: port 11(vethc8c801e) entered forwarding state
[   93.710406] hassio: port 11(vethc8c801e) entered disabled state
[   93.862928] eth0: renamed from veth4f62ea8
[   93.887807] hassio: port 11(vethc8c801e) entered blocking state
[   93.887814] hassio: port 11(vethc8c801e) entered forwarding state
[   97.007785] pl011-axi 1f0003c000.serial: no DMA platform data

agners, 2025-03-25:
> [ 0.652489] 1f0003c000.serial: ttyAMA1 at MMIO 0x1f0003c000 (irq = 148, base_baud = 0) is a PL011 AXI

That is the expected output, the serial port is at MMIO 3c000 is the one where the radio is connected to, and that is mapped to `ttyAMA1`.

The most likely case here is that something else is accessing the serial port, e.g. maybe ZHA reconfigured itself and accesses `ttyAMA1` directly? Or some other add-on was set to default start up and accesses `ttyAMA1`?

I do have the PoE version here and can access the radiio using the EmberZNet fiirmware from ZHA, so there is no general issue here. The Multiprotocol should work just as well. That said, Multiprotocol is not recommended at this point, so ideally you flash the Zigbee firmware instead and use a separate radio for Thread.

stau, 2025-03-26:
In the end it looked like the "wrong firmware" was installed, but I am not sure which part of the upgrade would have caused it. Remarkably there was no mention of "wrong firmware" until I uninstalled ZHA.

I followed the next steps for resolution: 

1. Running the ZHA re-configuration, this did not do anything, it times out with an error. 
2. I uninstalled the ZHA integration completely + removed data
3. I re-installed the ZHA integration, this then prompted that the wrong firmware was installed, sent me to https://yellow.home-assistant.io/guides/disable-multiprotocol/#flash-the-silicon-labs-radio-firmware. (It was not possible to follow the steps as the instructions did not match reality) : [image]
Furthermore there is no prompt to remove anything :

[image]

5. I decided to believe the firmware somehow was broken, so I ran the `Silicon Labs Flasher`. 
6. Then I ran the ZHA integration again and it is able to access `ttyAMA1`.