home-assistant/operating-system #3962: Upon updating from 14.x to 15.0, USB devices are enumerated differently, breaking ZHA

laposheureux, 2025-03-21:
### Describe the issue you are experiencing

After this update, I had to reconfigure Zigbee from ttyUSB1 to ttyUSB2. Now my HubZ device has Z-Wave at USB0, zigbee at USB2, and my SkyConnect has inserted itself between them at USB1.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

15.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

RPI5 8GB, originally had HASS on an SD card on a PI4, restored it to the Pi 5 with an SSD as the boot drive and have done upgrades of the OS multiple times since without issue.

Current USB device adapter configuration after the 15.0 update looks like:

[image]

### Steps to reproduce the issue

1. Have both HubZ hybrid Z-Wave and Zigbee stick connected as well as SkyConnect
2. Upgrade from 14.x to 15.0
3. Devices change USB IDs

### Anything in the Supervisor logs that might be useful for us?

```txt
Logs pointed me to the device not being found which led me to looking at the hardware entries and seeing that the device had switched serial port adapter names
```

### Anything in the Host logs that might be useful for us?

```txt
n/a
```

### System information

## System Information

version | core-2025.3.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.6.74-haos-raspi
arch | aarch64
timezone | America/Los_Angeles
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1601
Downloaded Repositories | 1

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.0
-- | --
update_channel | stable
supervisor_version | supervisor-2025.03.3
agent_version | 1.7.2
docker_version | 28.0.1
disk_total | 234.0 GB
disk_used | 17.2 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | ESPHome Device Builder (2025.3.0), File editor (5.8.0), Mosquitto broker (6.5.0), PS5 MQTT (1.4.0), Ring-MQTT with Video Streaming (5.8.0), Samba share (12.5.0), Z-Wave JS (0.10.0), Piper (1.5.2), Whisper (2.4.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 0
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), end0 (enabled, default, auto), hassio (disabled), docker0 (disabled), veth155ffe4 (disabled), veth4f316f6 (disabled), veth74a51b6 (disabled), veth76ffa18 (disabled), veth40d5894 (disabled), veth5e92401 (disabled), veth9ff7ed1 (disabled), veth915e84c (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), end0 (10.0.7.165/24), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), veth155ffe4 (), veth4f316f6 (), veth74a51b6 (), veth76ffa18 (), veth40d5894 (), veth5e92401 (), veth9ff7ed1 (), veth915e84c ()
ipv6_addresses | lo (::1/128), end0 (fd82:b364:726b:149:abb2:aa6c:9ea9:495b/64, fe80::7a0:5e09:18ea:1078/64), hassio (fe80::4400:e5ff:fe48:5d48/64), docker0 (fe80::2000:16ff:fe18:f571/64), veth155ffe4 (fe80::c4f3:dff:fea4:64b9/64), veth4f316f6 (fe80::e4dd:33ff:fefe:a0a6/64), veth74a51b6 (fe80::ec15:cfff:fe82:7d51/64), veth76ffa18 (fe80::58eb:7cff:fec3:5a40/64), veth40d5894 (fe80::8871:38ff:fe32:390b/64), veth5e92401 (fe80::28ab:a4ff:fe20:85b4/64), veth9ff7ed1 (fe80::8c1d:e8ff:fe83:987a/64), veth915e84c (fe80::3055:49ff:fe7e:4578/64)
announce_addresses | 10.0.7.165, fd82:b364:726b:149:abb2:aa6c:9ea9:495b, fe80::7a0:5e09:18ea:1078

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | February 27, 2025 at 9:36 AM
-- | --
current_recorder_run | March 21, 2025 at 1:16 AM
estimated_db_size | 502.86 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

sairon, 2025-03-23:
Please use paths from `/dev/serial/by-id/` in your configuration. These are predictable and do not rely on the order of enumeration or change when device is (un)plugged. This has been a general recommendation for quite a while, if you think it should be clarified somewhere in the docs, let us know.

closed 2025-03-23 by sairon

laposheureux, 2025-03-23:
🤔 Home Assistant should probably do that itself, I don't configure zha with a manual configuration anywhere, it's all done in the UI.