home-assistant/operating-system #3963: After updating to Home Assistant OS 15, z-wave js fails with stick ZME_UZB1

saveriol, 2025-03-21:
### Describe the issue you are experiencing

After updating to Home Assistant OS 15, z-wave js fails with this error:

`[12:28:30] INFO: Successfully send discovery information to Home Assistant.
2025-03-21T11:28:34.354Z DRIVER   
2025-03-21T11:28:34.369Z DRIVER   version 14.3.8
2025-03-21T11:28:34.370Z DRIVER   
2025-03-21T11:28:35.478Z CONFIG   version 14.3.8
2025-03-21T11:28:36.826Z CNTRLR   querying Serial API capabilities...
2025-03-21T11:28:38.295Z CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                                  n 100 ms.
2025-03-21T11:28:39.411Z CNTRLR   Failed to execute controller command after 2/3 attempts. Scheduling next try i
                                  n 1100 ms.
2025-03-21T11:28:41.537Z DRIVER   Failed to initialize the driver: ZWaveError: Timeout while waiting for an ACK 
                                  from the controller (ZW0200)
                                      at Driver.sendMessage (file:///usr/src/node_modules/zwave-js/src/lib/drive
                                  r/Driver.ts:6059:23)
                                      at ZWaveController.queryCapabilities (file:///usr/src/node_modules/zwave-j
                                  s/src/lib/controller/Controller.ts:1144:37)
                                      at Driver.initializeControllerAndNodes (file:///usr/src/node_modules/zwave
                                  -js/src/lib/driver/Driver.ts:1665:46)
                                      at Immediate.<anonymous> (file:///usr/src/node_modules/zwave-js/src/lib/dr
                                  iver/Driver.ts:1466:16)
Error in driver ZWaveError: Failed to initialize the driver: ZWaveError: Timeout while waiting for an ACK from the controller (ZW0200)
    at Driver.sendMessage (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:6059:23)
    at ZWaveController.queryCapabilities (file:///usr/src/node_modules/zwave-js/src/lib/controller/Controller.ts:1144:37)
    at Driver.initializeControllerAndNodes (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:1665:46)
    at Immediate.<anonymous> (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:1466:16) (ZW0100)
    at Immediate.<anonymous> (file:///usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:1486:6) {
  code: 100,
  context: undefined,
  transactionSource: undefined
}
Shutting down`

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

15.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Home Assistant Yellow.
Stick Z-Wave ZME_UZB1

### Steps to reproduce the issue

1. Update to OS 15.0
2. Z-Wave JS not working anymore with Z-Wave ZME_UZB1

### Anything in the Supervisor logs that might be useful for us?

```txt
2025-03-21 12:27:26.216 ERROR (MainThread) [supervisor.addons.addon] Add-on core_zwave_js has invalid options: Device '/dev/serial/by-id/usb-0658_0200-if00' does not exist in Z-Wave JS (core_zwave_js). Got {'device': '/dev/serial/by-id/usb-0658_0200-if00', 'log_level': 'info', 'log_to_file': False, 'log_max_files': 7, 'soft_reset': 'Automatic', 's0_legacy_key': 'xxx', 's2_access_control_key': 'xxxx', 's2_authenticated_key': 'xxx', 's2_unauthenticated_key': 'xxx', 'lr_s2_access_control_key': 'xxx', 'lr_s2_authenticated_key': 'xxx...
```

### Anything in the Host logs that might be useful for us?

```txt
2025-03-21 11:39:24.860 homeassistant kernel: usb 1-1.2: new full-speed USB device number 21 using dwc2
2025-03-21 11:39:25.068 homeassistant kernel: usb 1-1.2: device descriptor read/64, error -32
2025-03-21 11:39:25.384 homeassistant kernel: usb 1-1.2: device descriptor read/64, error -32
2025-03-21 11:39:25.404 homeassistant kernel: Bluetooth: hci0: Received unexpected HCI Event 0x00
2025-03-21 11:39:25.404 homeassistant kernel: Bluetooth: hci0: Frame reassembly failed (-84)
2025-03-21 11:39:25.412 homeassistant kernel: Bluetooth: hci0: Frame reassembly failed (-84)
2025-03-21 11:39:25.688 homeassistant kernel: usb 1-1.2: new full-speed USB device number 22 using dwc2
2025-03-21 11:39:25.880 homeassistant kernel: usb 1-1.2: device descriptor read/64, error -32
2025-03-21 11:39:26.404 homeassistant kernel: usb 1-1.2: device descriptor read/64, error -32
2025-03-21 11:39:26.512 homeassistant kernel: usb 1-1-port2: attempt power cycle
2025-03-21 11:39:27.216 homeassistant kernel: usb 1-1.2: new full-speed USB device number 23 using dwc2
2025-03-21 11:39:27.240 homeassistant kernel: usb 1-1.2: device descriptor read/8, error -32
2025-03-21 11:39:27.372 homeassistant kernel: usb 1-1.2: device descriptor read/8, error -32
2025-03-21 11:39:27.660 homeassistant kernel: usb 1-1.2: new full-speed USB device number 24 using dwc2
2025-03-21 11:39:27.684 homeassistant kernel: usb 1-1.2: device descriptor read/8, error -32
2025-03-21 11:39:27.816 homeassistant kernel: usb 1-1.2: device descriptor read/8, error -32
2025-03-21 11:39:27.928 homeassistant kernel: usb 1-1-port2: unable to enumerate USB device
```

### System information

## System Information

version | core-2025.3.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.6.74-haos-raspi
arch | aarch64
timezone | Europe/Rome
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4998
Installed Version | 2.0.5
Stage | startup
Available Repositories | 1601
Downloaded Repositories | 17

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.0
-- | --
update_channel | stable
supervisor_version | supervisor-2025.03.3
agent_version | 1.7.2
docker_version | 28.0.1
disk_total | 27.8 GB
disk_used | 19.6 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | yellow
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (20.0.2), Cloudflared (5.2.11), Mosquitto broker (6.5.0), ESPHome Device Builder (2025.3.0), Network UPS Tools (0.14.1), Zigbee2MQTT (2.1.3-1), MariaDB (2.7.2), Whisper (2.4.0), Piper (1.5.2), File editor (5.8.0), Z-Wave JS (0.10.0), HomeConnect2MQTT (v0.4.3), Zigbee2MQTT Edge (edge), HomeConnect2MQTT (v0.1.7), Node-RED (19.0.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 7
views | 5
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), end0 (enabled, default, auto), hassio (disabled), docker0 (disabled), vethd968666 (disabled), veth953c409 (disabled), vethf8d53ec (disabled), veth1103bc1 (disabled), veth81d93ac (disabled), veth31a7d22 (disabled), veth1d08fde (disabled), vetha3cf3f0 (disabled), veth28aaa41 (disabled), vethdd8a663 (disabled), veth3661326 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), end0 (192.168.127.127/24), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), vethd968666 (), veth953c409 (), vethf8d53ec (), veth1103bc1 (), veth81d93ac (), veth31a7d22 (), veth1d08fde (), vetha3cf3f0 (), veth28aaa41 (), vethdd8a663 (), veth3661326 ()
ipv6_addresses | lo (::1/128), end0 (fe80::7c46:37b1:1d3b:5695/64), hassio (fe80::1c59:e4ff:fe7c:4fd6/64), docker0 (fe80::84cf:5cff:fef8:2102/64), vethd968666 (fe80::cfd:d9ff:fe68:e154/64), veth953c409 (fe80::2496:7aff:fe24:18a2/64), vethf8d53ec (fe80::f0ff:15ff:fe71:298f/64), veth1103bc1 (fe80::28b8:55ff:fe9c:510e/64), veth81d93ac (fe80::9cd8:6dff:fe78:b0a5/64), veth31a7d22 (fe80::6ca0:b5ff:feeb:6caa/64), veth1d08fde (fe80::10f1:e7ff:fe70:2d4d/64), vetha3cf3f0 (fe80::30c2:d7ff:fe1c:3fb/64), veth28aaa41 (fe80::7ce0:7fff:fe98:8ec4/64), vethdd8a663 (fe80::4856:bfff:fe19:250b/64), veth3661326 (fe80::ec8e:dff:febb:b9db/64)
announce_addresses | 192.168.127.127, fe80::7c46:37b1:1d3b:5695

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | September 16, 2024 at 21:18
-- | --
current_recorder_run | March 21, 2025 at 11:49
estimated_db_size | 5239.72 MiB
database_engine | mysql
database_version | 10.11.6

</details>

eb1git, 2025-03-23:
Same problem, downgrading to 14.2 solves it.

I suppose it has to do with this one "Remove USB stack patches working around obsoleted Z-Wave devices issues #3854"

saveriol, 2025-03-23:
Well this stick is [https://www.home-assistant.io/docs/z-wave/controllers/](officially supported) (under "500 series controllers">"Z-Wave.Me UZB1 stick"), that's why I bought it, so I guess this should be fixed...

eb1git, 2025-03-23:
Can't disagree with that and it's one of the most common ones too so probably allot of users affected.

saveriol, 2025-03-23:
@sairon can we expect a solution for this issue? Or do we need to stay on 14.2 until we change sticks? Thanks 🙏

sairon, 2025-03-23:
Sorry, @eb1git is right, the referred change is the reason for breakage and the reasoning for this regression appearing again was already given there.

When the regression appeared the first time last year with Linux 6.6 kernel, we identified the offending patch and discussed potential solution with Linux USB stack maintainer and the Z-Wave.me manufacturer. A software solution in the kernel is not feasible mostly because of the non-compliant behavior of the device, and Z-Wave.me [declares the hardware EOL](https://github.com/home-assistant/operating-system/issues/2995#issuecomment-2059766457) with no fix on the firmware side up to this date.

What we can do is to clarify the documentation. There's [this note](https://www.home-assistant.io/docs/z-wave/controllers/#aeotec-z-stick) already. What no one has realized at the time of writing is that the Z-Wave-Me UZB1 stick is essentially the same hardware and suffers from the same hardware flaw, and Raspberry Pi 4 is not the only platform affected - it depends on more factors (e.g. it may work in some ports) and generally breaks in newer Linux not only in HAOS or RPi OS (it doesn't work with my laptop either).

Also, what's worth saying, is that using a powered USB hub is a possible solution here as well, so using that can be a stop-gap solution if you don't want to migrate to a different controller yet.

closed 2025-03-23 by sairon