home-assistant/operating-system #3970: FAN running all time at intel based minipc

muhkuh666, 2025-03-25:
### Describe the issue you are experiencing

Since OS 15 when I restard my NUC the fan is running all time. 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

15

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

[Mini PC, Intel Celeron N3450 8GB DDR4, 128GB M.2 SATA SSD, Mini Desktop Computer, AWOW AL34 Micro PC, Windows 10 Home, 4K HD, Dual HDMI 1.4, 2.4G/5G Dual WiFi, Gigabit Ethernet, 4 X USB, BT 4.2](https://www.amazon.de/dp/B08F73X85Y?ref_=ppx_hzod_title_dt_b_fed_asin_title_0_0)

### Steps to reproduce the issue

1.Shutdown
2.Start
3. FAN running all the time
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2025-03-24 05:54:11.250 INFO (MainThread) [supervisor.resolution.check] System checks complete
2025-03-24 05:54:11.250 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
2025-03-24 05:54:11.554 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2025-03-24 05:54:11.555 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2025-03-24 05:54:11.555 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2025-03-24 05:56:31.154 INFO (MainThread) [supervisor.store.git] Update add-on /data/addons/core repository from https://github.com/home-assistant/addons
2025-03-24 05:56:31.156 INFO (MainThread) [supervisor.store.git] Update add-on /data/addons/git/5c53de3b repository from https://github.com/esphome/home-assistant-addon
2025-03-24 05:56:31.158 INFO (MainThread) [supervisor.store.git] Update add-on /data/addons/git/bf7963f0 repository from https://github.com/b3nn0/hassio-addon-ism7mqtt
2025-03-24 05:56:31.159 INFO (MainThread) [supervisor.store.git] Update add-on /data/addons/git/45df7312 repository from https://github.com/zigbee2mqtt/hassio-zigbee2mqtt
2025-03-24 05:56:31.180 INFO (MainThread) [supervisor.store.git] Update add-on /data/addons/git/193f5383 repository from https://github.com/mercenaruss/zigstar_addons
2025-03-24 05:56:31.182 INFO (MainThread) [supervisor.store.git] Update add-on /data/addons/git/a0d7b954 repository from https://github.com/hassio-addons/repository
2025-03-24 05:56:31.192 INFO (MainThread) [supervisor.store.git] Update add-on /data/addons/git/d5369777 repository from https://github.com/music-assistant/home-assistant-addon
2025-03-24 05:56:31.196 INFO (MainThread) [supervisor.store.git] Update add-on /data/addons/git/8ce4abae repository from https://github.com/egguy/grott-home-assistant-add-on
2025-03-24 05:56:33.903 INFO (MainThread) [supervisor.store] Loading add-ons from store: 92 all - 0 new - 0 remove
2025-03-24 05:56:33.904 INFO (MainThread) [supervisor.store] Loading add-ons from store: 92 all - 0 new - 0 remove
2025-03-24 05:56:34.253 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2025-03-24 05:59:43.242 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
2025-03-24 06:26:34.607 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2025-03-24 06:37:09.247 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'growatt'
2025-03-24 06:37:09.681 INFO (MainThread) [supervisor.auth] Successful login for 'growatt'
2025-03-24 06:54:11.595 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
2025-03-24 06:54:11.595 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
2025-03-24 06:54:11.596 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2025-03-24 06:54:11.596 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
2025-03-24 06:54:11.605 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
2025-03-24 06:54:11.679 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
2025-03-24 06:54:11.680 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
2025-03-24 06:54:11.680 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
2025-03-24 06:54:11.681 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
2025-03-24 06:54:11.681 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
2025-03-24 06:54:11.684 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
2025-03-24 06:54:11.684 INFO (MainThread) [supervisor.resolution.check] System checks complete
2025-03-24 06:54:11.684 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
2025-03-24 06:54:11.901 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2025-03-24 06:54:11.901 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2025-03-24 06:54:11.902 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2025-03-24 06:56:34.993 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2025-03-24 07:10:31.690 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/mmcblk0 - /dev/disk/by-id/mmc-SC32G_0x54bd25db
2025-03-24 07:10:31.697 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/mmcblk0p1 - /dev/disk/by-id/mmc-SC32G_0x54bd25db-part1
2025-03-24 07:11:46.948 INFO (MainThread) [supervisor.host.control] Initialize host reboot using logind
[... 255 lines cut ...]
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/udev.sh
[06:21:47] INFO: Using udev information from host
cont-init: info: /etc/cont-init.d/udev.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun supervisor (no readiness notification)
services-up: info: copying legacy longrun watchdog (no readiness notification)
[06:21:47] INFO: Starting local supervisor watchdog...
s6-rc: info: service legacy-services successfully started
2025-03-24 06:21:50.796 INFO (MainThread) [__main__] Initializing Supervisor setup
2025-03-24 06:21:50.993 INFO (MainThread) [supervisor.coresys] Setting up coresys for machine: generic-x86-64
2025-03-24 07:21:51.014 INFO (MainThread) [supervisor.docker.supervisor] Attaching to Supervisor ghcr.io/home-assistant/amd64-hassio-supervisor with version 2025.03.3
2025-03-24 07:21:51.071 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state initialize
2025-03-24 07:21:51.078 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
```

### Anything in the Host logs that might be useful for us?

```txt
Sorry no
```

sairon, 2025-03-25:
The OS doesn't have any control over the fans, this is usually done by the BIOS (if I simplify it). Increased fan speed can be an indicator of increased CPU usage, can you check that?

muhkuh666, 2025-03-26:
The Problem occurs since last HASS OS update, before no problem!

muhkuh666, 2025-03-26:
No Bios changes were made....

newtsy, 2025-03-27:
I upgraded last night, walked into the office this morning and my fan is running full speed as well.  CPU temp is 39c and usage 6%

muhkuh666, 2025-03-28:
So why do 2 people have the same Problem after update, then ????

sfields-forever, 2025-03-28:
I also use generic-x86-64 (Generic UEFI capable x86-64 systems). And since the update to HA OS 15 it uses A LOT (three times) more cpu, maybe this results in your fans running all the time. System even is slow and lags sometimes which was never the case on 14.

thamesseb2, 2025-04-01:
same trouble here with 15.1 and nuc

storelovely, 2025-05-30:
Same problem here after upgrade to 15.2, workaround is `os boot-slot` command to reboot. Reboot or shutdown any other way causes the fan to turn back on full throttle.

closed 2025-09-06 by github-actions[bot]