home-assistant/operating-system #3975: Continuously logs stating terminating processes without coredump

jhalldn, 2025-03-26:
### Describe the issue you are experiencing

After successful update from 14.1 to 15.0, logs in System/Logs/Host are continuously show different pids terminating without coredumps. Systemd is the logger.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

15.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

RPI5, 4GB, nvme disk.

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing related
```

### Anything in the Host logs that might be useful for us?

```txt
2025-03-26 21:21:38.425 homeassistant systemd-coredump[754000]: Process 753977 (python3) of user 0 terminated abnormally without generating a coredump.
2025-03-26 21:21:38.427 homeassistant systemd[1]: systemd-coredump@22363-753999-0.service: Deactivated successfully.
2025-03-26 21:21:39.194 homeassistant systemd[1]: Started Process Core Dump (PID 754031/UID 0).
2025-03-26 21:21:39.402 homeassistant systemd-coredump[754032]: Process 754009 (python3) of user 0 terminated abnormally without generating a coredump.
2025-03-26 21:21:39.404 homeassistant systemd[1]: systemd-coredump@22364-754031-0.service: Deactivated successfully.
2025-03-26 21:21:40.214 homeassistant systemd[1]: Started Process Core Dump (PID 754063/UID 0).
2025-03-26 21:21:40.413 homeassistant systemd-coredump[754064]: Process 754041 (python3) of user 0 terminated abnormally without generating a coredump.
2025-03-26 21:21:40.416 homeassistant systemd[1]: systemd-coredump@22365-754063-0.service: Deactivated successfully.
2025-03-26 21:21:41.201 homeassistant systemd[1]: Started Process Core Dump (PID 754095/UID 0).
2025-03-26 21:21:41.405 homeassistant systemd-coredump[754096]: Process 754073 (python3) of user 0 terminated abnormally without generating a coredump.
2025-03-26 21:21:41.406 homeassistant systemd[1]: systemd-coredump@22366-754095-0.service: Deactivated successfully.
2025-03-26 21:21:42.181 homeassistant systemd[1]: Started Process Core Dump (PID 754127/UID 0).
2025-03-26 21:21:42.383 homeassistant systemd-coredump[754128]: Process 754105 (python3) of user 0 terminated abnormally without generating a coredump.
2025-03-26 21:21:42.385 homeassistant systemd[1]: systemd-coredump@22367-754127-0.service: Deactivated successfully.
2025-03-26 21:21:43.209 homeassistant systemd[1]: Started Process Core Dump (PID 754159/UID 0).
2025-03-26 21:21:43.418 homeassistant systemd-coredump[754160]: Process 754137 (python3) of user 0 terminated abnormally without generating a coredump.
2025-03-26 21:21:43.421 homeassistant systemd[1]: systemd-coredump@22368-754159-0.service: Deactivated successfully.
2025-03-26 21:21:44.209 homeassistant systemd[1]: Started Process Core Dump (PID 754210/UID 0).
2025-03-26 21:21:44.416 homeassistant systemd-coredump[754211]: Process 754188 (python3) of user 0 terminated abnormally without generating a coredump.
2025-03-26 21:21:44.418 homeassistant systemd[1]: systemd-coredump@22369-754210-0.service: Deactivated successfully.
2025-03-26 21:21:45.233 homeassistant systemd[1]: Started Process Core Dump (PID 754242/UID 0).
2025-03-26 21:21:45.435 homeassistant systemd-coredump[754243]: Process 754220 (python3) of user 0 terminated abnormally without generating a coredump.
2025-03-26 21:21:45.437 homeassistant systemd[1]: systemd-coredump@22370-754242-0.service: Deactivated successfully.
2025-03-26 21:21:46.213 homeassistant systemd[1]: Started Process Core Dump (PID 754274/UID 0).
2025-03-26 21:21:46.387 homeassistant systemd-coredump[754275]: Process 754252 (python3) of user 0 terminated abnormally without generating a coredump.
2025-03-26 21:21:46.389 homeassistant systemd[1]: systemd-coredump@22371-754274-0.service: Deactivated successfully.
2025-03-26 21:21:47.213 homeassistant systemd[1]: Started Process Core Dump (PID 754315/UID 0).
2025-03-26 21:21:47.410 homeassistant systemd-coredump[754316]: Process 754286 (python3) of user 0 terminated abnormally without generating a coredump.
2025-03-26 21:21:47.412 homeassistant systemd[1]: systemd-coredump@22372-754315-0.service: Deactivated successfully.
2025-03-26 21:21:48.213 homeassistant systemd[1]: Started Process Core Dump (PID 754347/UID 0).
2025-03-26 21:21:48.422 homeassistant systemd-coredump[754348]: Process 754325 (python3) of user 0 terminated abnormally without generating a coredump.
2025-03-26 21:21:48.424 homeassistant systemd[1]: systemd-coredump@22373-754347-0.service: Deactivated successfully.
2025-03-26 21:21:49.198 homeassistant systemd[1]: Started Process Core Dump (PID 754379/UID 0).
2025-03-26 21:21:49.401 homeassistant systemd-coredump[754380]: Process 754357 (python3) of user 0 terminated abnormally without generating a coredump.
[... 150 lines cut ...]
adapters | lo (disabled), end0 (enabled, default, auto), docker0 (disabled), hassio (disabled), veth7c1cea6 (disabled), vethaebfb6b (disabled), veth5e7566a (disabled), veth4df8bad (disabled), vethe9fdd96 (disabled), veth6a14472 (disabled), veth9dcc798 (disabled), vethe6b9870 (disabled), veth5351edc (disabled), veth4f12dce (disabled), vethdf4de54 (disabled), vethcda78a0 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), end0 (192.168.10.88/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), veth7c1cea6 (), vethaebfb6b (), veth5e7566a (), veth4df8bad (), vethe9fdd96 (), veth6a14472 (), veth9dcc798 (), vethe6b9870 (), veth5351edc (), veth4f12dce (), vethdf4de54 (), vethcda78a0 ()
ipv6_addresses | lo (::1/128), end0 (fde9:9b75:68ad:0:afbd:713:ab9f:3a35/64, fe80::700f:dba4:7d0:3e6c/64), docker0 (fe80::bc50:3aff:fe8a:e35/64), hassio (fe80::f0a8:79ff:fe39:ac8c/64), veth7c1cea6 (fe80::48cb:39ff:fe6a:8172/64), vethaebfb6b (fe80::5cdf:ffff:feb6:9266/64), veth5e7566a (fe80::3006:3aff:feb7:5576/64), veth4df8bad (fe80::248c:b1ff:fef7:6250/64), vethe9fdd96 (fe80::6cab:f9ff:fe41:e4fd/64), veth6a14472 (fe80::d493:16ff:feeb:d2ad/64), veth9dcc798 (fe80::380b:1cff:feb4:2619/64), vethe6b9870 (fe80::64ce:35ff:fe94:6721/64), veth5351edc (fe80::ec1d:fff:fece:7746/64), veth4f12dce (fe80::6cb5:39ff:fe27:e18c/64), vethdf4de54 (fe80::8863:b0ff:fe70:876d/64), vethcda78a0 (fe80::7c8a:4eff:fec1:d72c/64)
announce_addresses | 192.168.10.88, fde9:9b75:68ad:0:afbd:713:ab9f:3a35, fe80::700f:dba4:7d0:3e6c

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 15 mars 2025 07:41
-- | --
current_recorder_run | 26 mars 2025 16:08
estimated_db_size | 325.78 MiB
database_engine | mysql
database_version | 10.11.6

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

sairon, 2025-03-28:
This has to be some process in one of the containers. If you can access your HA fine, I'd suggest stopping all the add-ons first to see which one it comes from. Eventually, you could also try reverting to the previous version of the OS using `ha os boot-slot other` in the terminal.

jhalldn, 2025-03-28:
Ah , thanks for Quick and clear info. Yes i can reach my box. Im just glad its not any issue regarding your hard work . Ill continue to check my addons. I you are happy, er can probably close this case.

sairon, 2025-03-28:
No worries. What you're seeing is definitely not normal and the logs filling with the messages are not something you should just ignore. Given that you can reach it, it means that the `python3` process is not the HA Core itself but it's still worth investigating.

jhalldn, 2025-03-29:
Looked at all addons and their logs. Seems like I have an old add on for google assistant sdk. I followed the documentation and it seems to deprecated. I also use HA cloud. I guess I would be ok to uninstall it. After I stopped the addon my logs also stopped.

sairon, 2025-04-02:
Glad you found the culprit, thanks for reporting back.

closed 2025-04-02 by sairon