home-assistant/operating-system #3976: ethernet packet error with OS 15.0

kinggeorge2025, 2025-03-26:
### Describe the issue you are experiencing

im running a windows 2019 server hyper-v on an intel brand server with x722 ethernet with most recent drivers from intel.
14.2 works without the issue, 15.0 even on a fresh virtual built with a freshly downloaded vhd gives me this error as soon as i access HA.
"homeassistant kernel: hv_netvsc 20e35702-9196-4223-97b2-c12098bd5125 eth0: nvsp_rndis_pkt_complete error status: 2"
i did some searching & didnt find a working solution.
thanks

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

15.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

intel zeon silver 4110
64gb ddr4 reg ecc
intel raid 1
intel x722 10gb

### Steps to reproduce the issue

1.ran update onto existing 14.2 immediately causes the issue 
2. new virtual with new vhd of 15 downloaded does it immediately after booting
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
only shows in host log
```

### Anything in the Host logs that might be useful for us?

```txt
"homeassistant kernel: hv_netvsc 20e35702-9196-4223-97b2-c12098bd5125 eth0: nvsp_rndis_pkt_complete error status: 2"
```

sairon, 2025-03-31:
The only option that was disabled was `CONFIG_NET_IPIP`. I have checked the kernel sources and I actually don't understand how it could have any effect, but I'll be glad if you could test it and report back: https://github.com/home-assistant/operating-system/actions/runs/14131479593

kinggeorge2025, 2025-03-31:
i pulled [haos_ova-15.2.dev1743172980.vhdx.zip](https://github.com/home-assistant/operating-system/actions/runs/14131479593/artifacts/2840085052) have the same issue.

kinggeorge2025, 2025-04-03:
also same issue with 15.1

kinggeorge2025, 2025-06-04:
any update on this?
i seen this post claiming there is a fix.
https://bugzilla.kernel.org/show_bug.cgi?id=217503
any chance we will see this soon in haos?
thanks

sairon, 2025-06-17:
@kinggeorge2025 We update OS dev versions to latest kernel stable releases regularly, the issue mentions it should be fixed since 6.12.30 which we updated to couple of weeks ago: https://github.com/home-assistant/operating-system/commit/487ae21e011354cbdac7c554cb6a8cf06f4c210d

Ergo the changes should be also included in yesterday's 16.0.rc1. Could you give it a try?