home-assistant/operating-system #3979: Compression on HAOS released images makes Proxmox installs tougher

kaecyra, 2025-03-28:
### Describe the issue you are experiencing

### TL;DR 

Could the HAOS releases be made to include either uncompressed or simple gz-compressed OVA images, rather than solely xz-compressed ones.

### Why?

I'm currently rebuilding my homelab and taking the opportunity to Terraform the deployment of all my VMs to their respective Proxmox nodes, instead of provisioning them manually. 

Trying to do as much as possible with Terraform and avoid manual configuration, which includes installing available boot images into Proxmox itself. Proxmox has a very cool feature that allows the installation of images by [direct download to the VE](https://pve.proxmox.com/pve-docs/api-viewer/index.html#/nodes/{node}/storage/{storage}/download-url). It even supports checksums and decompression as long as one of the 4 supported algorithms are used: `gz`, `lzo`, `zst`, `bz2`.

The Ubuntu base image was no problem because it is [available uncompressed](https://cloud-images.ubuntu.com/noble/current/noble-server-cloudimg-amd64.img). The HAOS OVA QCow2 image is compressed with `xz` (and I think tar underneath that), which makes it impossible to leverage this Proxmox functionality. I will need to manually download and extract the image locally and then upload the result. This feels fragile and suboptimal.

I know xz offers superior compression ratios to gz, but it is less widely supported and therefore has some drawbacks.

### The ask

**So, any chance the HAOS releases could include uncompressed or gz-compressed OVA images?**

Thank you for thinking about this!

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

15.0

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Virtualized.

Host: Proxmox VE 8.3
Memory: 192Gb
CPU: 2x Xeon E5-2640 v4 @ 2.40GHz
Disk:
  Proxmox: 2x 1.2Tb (RAID 1)
  VMs: 6x 1.2Tb (RAID 10)

Guest: HAOS
Memory: 4Gb
CPU: 2x vCPU
Disk: 32Gb

### Steps to reproduce the issue

1. Begin adding a new ISO image in Proxmox
2. Attempt to "Query URL" against OVA image (https://github.com/home-assistant/operating-system/releases/download/15.0/haos_ova-15.0.qcow2.xz)
3. Note the failed discovery
4. Attempt to download the OVA image anway
5. Failure

### Anything in the Supervisor logs that might be useful for us?

```txt
No, this isn't about HAOS itself.
```

### Anything in the Host logs that might be useful for us?

```txt
No, this isn't about HAOS itself.
```

sairon, 2025-03-28:
I'd suggest taking a step aside here and instead of importing the disk image, importing the actual `.ova` file. As a bonus, that also contains some meta information that simplifies creation of the VM.

We'd like to make this the preferred way of installing HAOS in Proxmox - currently people are usually using the tteck/Community scripts but this method is a bit friendlier and built in the Proxmox.

Getting back to the compression algorithm, I'd rather prefer if Proxmox supported XZ as well. I do not know if there's a reason for not adding that capability but there's quite recent discussion thread about it: https://forum.proxmox.com/threads/cloud-init-using-a-raw-xz-file.158782/

kaecyra, 2025-03-28:
Using the OVA directly means I have to use the "Import" feature, which is not covered by the Terraform provider. FML.

bpg, 2025-03-31:
@kaecyra this workaround might help: https://github.com/bpg/terraform-provider-proxmox/discussions/1212#discussioncomment-12205151

KlfJoat, 2025-04-21:
(I'm not with HA, I'm just a rando that stumbled on this Issue)

Proxmox is less than 1 year older than xz. If Proxmox don't include support for something that has existed for 15 years, and been supported by the Linux kernel for over a decade, then why not?

It seems to me that this puts the onus on the wrong side of the equation. It's unscaleable to think of it as a bug/issue to be fixed by every single possible person or org that creates and publishes a disk image. It's a bug/issue for the consuming application: Proxmox.