home-assistant/operating-system #3983: 15.0 doesn't start properly on rpi5

PennyAfNorberg, 2025-03-29:
### Describe the issue you are experiencing

Observer has both reported ok and disconnected. Working after a rollback

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

14.2

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Rpi 5, sd-card

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
sorry did get that info
```

### Anything in the Host logs that might be useful for us?

```txt
sorry did get that info
```

sairon, 2025-03-31:
Have you checked #3943 if there's something in common?

PennyAfNorberg, 2025-03-31:
It seems like it, need to test that way to be sure however

erkr, 2025-04-01:
@sairon I had reboot issues installing 14.2, but now upgraded successfully to 15.1 🙏 thanks for the improvements.

I didn't update the boot loader firmware yet, as instructed in the referenced issue (don't feel comfortable doing that)

Reason to chime in; you stated Raspberry Pi OS automatically updates the bootloader. 
Any plans that HassOS facilitates updating the bootloader as well?!