home-assistant/operating-system #3984: home assistant not starting anymore after upgrade 14.2 to 15.0

Marteene, 2025-03-30:
### Describe the issue you are experiencing

- after upgrade 14.2 to 15.0, my system does not start up anymore. 
- observer says supervisor disconnected
- when connecting screen and keyboard directly, it says CLI not starting, emergency mode

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

15

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

HP thin client, no devices attached

### Steps to reproduce the issue

- after upgrade 14.2 to 15.0, my system does not start up anymore. 

### Anything in the Supervisor logs that might be useful for us?

```txt
please let me know what you need and where I find it
```

### Anything in the Host logs that might be useful for us?

```txt
please let me know what you need and where I find it
```

DravenSA, 2025-03-30:
I also had this, after a few attempts HA finally rolled back to the previous version, and it tried to update a second time, got the same problem, so I have decided to stay off of v15

hfolkers, 2025-03-30:
I have the same problem. @DravenSA how did HA roll back? Anything I can do for this?

ha-jos, 2025-03-30:
> I have the same problem. [@DravenSA](https://github.com/DravenSA) how did HA roll back? Anything I can do for this?

There are various threads about this issue. https://community.home-assistant.io/t/home-assistant-unreachable-after-update-to-haos-15/865003/26?page=2 for example.

I have the same issue, there are at least 2 options to boot your old 14.2 version.

1. Power cycle untill it comes on again (is not very convient)
2. When you boot you might see a flash of grub menu. You can select a boot slot. In my case boot slot B is corrupt, bootdisk A is OK.

There is far more to talk about, but this should bring your instance back online.

Good luck

DravenSA, 2025-03-30:
> I have the same problem. [@DravenSA](https://github.com/DravenSA) how did HA roll back? Anything I can do for this?

I did they bad way of just power cycling several times

hfolkers, 2025-03-31:
Thank you all, after connecting a screen and keyboard I managed to switch slot. And everything is up and running.

sairon, 2025-03-31:
Looks like a duplicate of the pinned #3944

closed 2025-03-31 by sairon