home-assistant/operating-system #3987: Pb with OS 15.0 HA

Titi9562, 2025-03-31:
### Describe the issue you are experiencing

Difficulty restarting the server after installing Home Assistant OS 15. Error message upon reconnection indicating that the system refused the connection ('[Unable to access home assistant port 8123, connection refused') . Had to clear browser data, and after multiple restarts, it finally returned to normal. The upgrade did not go through; the server remained on version 14.2. It was very stressful because I thought I would never manage to restart the server...

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

14.2

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Generic x86-64
Hard Disk USB
Dongle Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2

### Steps to reproduce the issue

1. Update Home Assistant Operating System 14.2 ==> 15.0
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
## System Information

version | core-2025.3.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.6.73-haos
arch | x86_64
timezone | Europe/Paris
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1625
Downloaded Repositories | 26

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 1 janvier 2018 à 01:00
relayer_connected | false
relayer_region | null
remote_enabled | true
remote_connected | false
alexa_enabled | true
google_enabled | true
cloud_ice_servers_enabled | true
remote_server | null
certificate_status | null
instance_id | 6d0f0660079444a68521162c4fc20488
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.03.3
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 916.8 GB
disk_used | 29.3 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Spotify Connect (0.14.0), go2rtc (1.9.9), Frigate (0.15.0), Terminal & SSH (9.16.0), Studio Code Server (5.18.3), ESPHome Device Builder (dev) (2025.4.0-dev20250324), SomfyProtect2MQTT (2025.3.0), Duck DNS (1.18.0), Cloudflared (5.2.11), Advanced SSH & Web Terminal (20.0.2), MyElectricalData (0.13.2), pnproxy master (master), Mosquitto broker (6.5.0), Linky (1.5.0), Zigbee2MQTT (2.1.3-1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 19
views | 13
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp1s0 (enabled, default, auto), docker0 (disabled), hassio (disabled), vetheb8428a (disabled), vethb4a34e6 (disabled), vethb5afe1c (disabled), veth51c0b60 (disabled), veth9602d8b (disabled), veth2d1bc03 (disabled), veth890ceb8 (disabled), vethadbf6e4 (disabled), vethc57d62e (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp1s0 (192.168.1.10/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), vetheb8428a (), vethb4a34e6 (), vethb5afe1c (), veth51c0b60 (), veth9602d8b (), veth2d1bc03 (), veth890ceb8 (), vethadbf6e4 (), vethc57d62e ()
ipv6_addresses | lo (::1/128), enp1s0 (2a01:e0a:3c4:97c0:da1c:469d:ef65:6/64, fe80::415e:bdc3:83c:34b8/64), docker0 (fe80::42:ddff:fec5:265/64), hassio (fe80::42:64ff:fef7:1607/64), vetheb8428a (fe80::7882:61ff:fe6b:77be/64), vethb4a34e6 (fe80::871:ceff:fe60:aecc/64), vethb5afe1c (fe80::6c09:89ff:fe1e:ffcc/64), veth51c0b60 (fe80::ec50:b5ff:fe55:179d/64), veth9602d8b (fe80::d01c:59ff:fe60:b117/64), veth2d1bc03 (fe80::7c15:d0ff:fe83:6410/64), veth890ceb8 (fe80::8c34:87ff:fe7b:f30b/64), vethadbf6e4 (fe80::b486:2fff:fec6:822d/64), vethc57d62e (fe80::f4a8:bff:fe78:9a53/64)
announce_addresses | 192.168.1.10, 2a01:e0a:3c4:97c0:da1c:469d:ef65:6, fe80::415e:bdc3:83c:34b8

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 9 mars 2025 à 18:37
-- | --
current_recorder_run | 24 mars 2025 à 20:32
estimated_db_size | 52.96 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

<details><summary>Sonoff</summary>

version | 3.8.2 (c4b6fda)
-- | --
cloud_online | 1 / 1
local_online | 1 / 1

</details>
```

### Anything in the Host logs that might be useful for us?

```txt
NO
```

sairon, 2025-03-31:
Please at least share more details about your "generic x86-64" machine.

Titi9562, 2025-03-31:
> Please at least share more details about your "generic x86-64" machine.

What details do you need?

My server is an HP T520 Flexible Series Thin Client TPC-W016 
I have a 1TB WD external USB hard drive

sairon, 2025-03-31:
Then it's a known issue fixed since 15.1.rc1.

closed 2025-03-31 by sairon