home-assistant/operating-system #3989: Yellow reports only 8GB of RAM on CM5 with 16GB

goll, 2025-03-31:
### Describe the issue you are experiencing

I have a HA Yellow with a CM5. Installed HAOS 15.0 using rpiboot without any issues. 

System only reports 8GB of RAM on a 16GB RAM CM5.

```
$ free -m
              total        used        free      shared  buff/cache   available
Mem:           7943        1279        5222           2        1442        6582
Swap:          2621           0        2621
```

```
$ dmesg | grep Memory
[    0.000000] Memory: 8062484K/8384512K available (15232K kernel code, 2294K rwdata, 4824K rodata, 5248K init, 1110K bss, 256492K reserved, 65536K cma-reserved)
```

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

15.0

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

- HA Yellow standard 1.3
- RPi CM5 model CM5116032 (WiFi, 16GB RAM/32GB eMMC) Rev6
- CPU revision e04180 (/proc/cpuinfo)

### Steps to reproduce the issue

1. Install HAOS using rpiboot
2. See 8GB of RAM instead of 16GB

### Anything in the Supervisor logs that might be useful for us?

```txt
Nope
```

### Anything in the Host logs that might be useful for us?

```txt
Nope
```

### System information

## System Information

version | core-2025.3.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.6.74-haos-raspi
arch | aarch64
timezone | Europe/Zagreb
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1621
Downloaded Repositories | 5

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.0
-- | --
update_channel | stable
supervisor_version | supervisor-2025.03.3
agent_version | 1.7.2
docker_version | 28.0.1
disk_total | 28.0 GB
disk_used | 6.5 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | yellow
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.16.0), Piper (1.5.2), Whisper (2.4.0), Mosquitto broker (6.5.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 1
views | 1
mode | storage

</details>

<details><summary>Network Configuration</summary>

REDACTED

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | March 19, 2025 at 3:23 PM
-- | --
current_recorder_run | March 28, 2025 at 11:00 PM
estimated_db_size | 5.46 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

<details><summary>Sonoff</summary>

version | 3.8.2 (c4b6fda)
-- | --
cloud_online | 1 / 1
local_online | 1 / 1

</details>

### Additional information

[image]

sairon, 2025-03-31:
Unfortunately I don't have it to check, but is it detected correctly in RPi OS?

sairon, 2025-03-31:
Exactly, install RPi OS the same way through rpiboot as you installed HAOS. It might be worth trying to install HAOS for RPi 5 to eMMC, but in that case couple of changes will be needed so you can at least check the serial console. I can provide the instructions later if you can test that too.

goll, 2025-03-31:
Used `2024-11-19-raspios-bookworm-arm64-lite.img.xz` and immediately shows the correct amount of RAM:

```
$ free -m
               total        used        free      shared  buff/cache   available
Mem:           16196         265       15927           5         144       15931
Swap:            511           0         511
```

Grabbed the bootlader_version:

```
$ vcgencmd bootloader_version
2025/01/22 00:16:51
version a77530632346c8ac1d579fa3c369f9980401e821 (release)
timestamp 1737505011
update-time 0
capabilities 0x0000007f
```

Handl3vogn, 2025-04-05:
Just upgraded my yellow from a cm4 8gb to a cm5 16gb (cm5116032).
And homeassistant is only reporting 8gb.
So same issue happening here.

superpower10, 2025-04-28:
+1

P.S.: Have the same CM5116032 Board. Mine is getting very warm (~ 60 °C) even though the CPU usage is only ~4%.

Does somebody of you have the same issue?

remonlam, 2025-04-28:
@superpower10 Just checked, and the CM5116000 is around 55 °C with a similar load (around 5%).

PR #4045 (2025-05-01): Fix U-Boot config to access all RAM on 16 GB CM5
U-Boot defconfig used for Yellow checks only 4 DRAM banks, however, CM5 with 16 GB has the memory spread across 8 banks. Add a patch (submitted upstream) to the defconfig to get access to the whole RAM.

Fixes #3989

## Summary by CodeRabbit

- **New Features**
  - Improved support for Raspberry Pi 5 models with 16 GiB RAM, allowing the system to recognize and utilize the full memory capacity.
commit 6ec77b84 (2025-05-01): Fix U-Boot config to access all RAM on 16 GB CM5 (#4045)

* Fix U-Boot config to access all RAM on 16 GB CM5

U-Boot defconfig used for Yellow checks only 4 DRAM banks, however, CM5 with 16
GB has the memory spread across 8 banks. Add a patch (submitted upstream) to
the defconfig to get access to the whole RAM.

Fixes #3989

* Add Upstream tag with link to uboot patchwork
closed 2025-05-01 by sairon