home-assistant/operating-system #3995: 15.1 Update: Failed marking slot kernel.1 non-bootable: Failed marking slot kernel.1 as bad: grub backend: Failed to run grub-editenv

jasper502, 2025-04-01:
### Describe the issue you are experiencing

```
[core-ssh ~]$ ha os update --version 15.1.rc1
Processing... Done.

Error: Unknown error, see supervisor
[core-ssh ~]$ 
```
This is the same log output I had with 15.1:

```
2025-04-01 15:51:16.512 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
2025-04-01 15:51:16.513 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
2025-04-01 15:51:16.513 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
2025-04-01 15:51:16.517 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
2025-04-01 15:51:16.643 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
2025-04-01 15:51:16.643 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
2025-04-01 15:51:16.644 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
2025-04-01 15:51:16.644 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
2025-04-01 15:51:16.645 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
2025-04-01 15:51:16.645 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
2025-04-01 15:51:16.645 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2025-04-01 15:51:16.646 INFO (MainThread) [supervisor.resolution.check] System checks complete
2025-04-01 15:51:16.646 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
2025-04-01 15:51:16.734 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2025-04-01 15:51:16.734 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2025-04-01 15:51:16.734 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2025-04-01 15:52:36.774 INFO (MainThread) [supervisor.host.info] Updating local host information
2025-04-01 15:52:37.096 INFO (MainThread) [supervisor.host.services] Updating service information
2025-04-01 15:52:37.098 INFO (MainThread) [supervisor.host.network] Updating local network information
2025-04-01 15:52:37.154 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
2025-04-01 15:52:37.159 INFO (MainThread) [supervisor.host.manager] Host information reload completed
2025-04-01 15:52:38.387 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2025-04-01 16:22:38.561 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2025-04-01 16:33:42.038 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_ssh
2025-04-01 16:33:42.040 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_ssh
2025-04-01 16:33:42.043 INFO (MainThread) [supervisor.api.middleware.security] /host/info access from core_ssh
2025-04-01 16:33:42.046 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from core_ssh
2025-04-01 16:33:44.350 INFO (MainThread) [supervisor.api.middleware.security] /os/update access from core_ssh
2025-04-01 16:33:44.484 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://os-artifacts.home-assistant.io/15.1.rc1/haos_generic-x86-64-15.1.rc1.raucb
2025-04-01 16:33:51.397 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-15.1.rc1.raucb
2025-04-01 16:33:51.433 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed marking slot kernel.1 non-bootable: Failed marking slot kernel.1 as bad:  grub backend: Failed to run grub-editenv: Child process exited with code 1
```
OS Info:

```
[core-ssh ~]$ ha os info
board: generic-x86-64
boot: A
boot_slots:
  A:
    state: booted
    status: bad
    version: "14.2"
  B:
    state: inactive
    status: bad
    version: null
data_disk: SAMSUNG-MZNTY128HDHP-000L1-S2YRNB0H927859
update_available: true
version: "14.2"
version_latest: "15.1"
```

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

14.2

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Baremetal x86 Intel NUC

### Steps to reproduce the issue

GUI and CLI upgrades fail -tried some Grok ideas - nothing helps.  OS running fine under 14.2 - just can't upgrade.

### Anything in the Supervisor logs that might be useful for us?

```txt
noted in main post
```

### Anything in the Host logs that might be useful for us?

```txt
none
```

jasper502, 2025-04-02:
Grok to the rescue - corrupted grubenv:

Recreate fully:

```
rm /mnt/boot/EFI/BOOT/grubenv
grub-editenv /mnt/boot/EFI/BOOT/grubenv create
 // Set variables:
grub-editenv /mnt/boot/EFI/BOOT/grubenv set ORDER="A B"
grub-editenv /mnt/boot/EFI/BOOT/grubenv set A_OK=1
grub-editenv /mnt/boot/EFI/BOOT/grubenv set A_TRY=1
grub-editenv /mnt/boot/EFI/BOOT/grubenv set B_OK=0
grub-editenv /mnt/boot/EFI/BOOT/grubenv set B_TRY=0
```

sairon, 2025-04-02:
I wonder how exactly got the GRUB environment corrupted (if it was a zero file or if it contained malformed data). Anyway it was probably just some fluke rather than a bug that we could fix. Good job on finding a way to fix the issue, can we consider this resolved?

closed 2025-04-02 by jasper502