home-assistant/operating-system #3997: 15.1 Update: High memory usage (+2GB)

litecross91, 2025-04-02:
### Describe the issue you are experiencing

After 15.1 update the memory usage is higher (+1,5+2 GB)

state at startup:
[image]

I rolled back to 14.2.

state at startup:
[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

15.1

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Proxmox+HASSOS

### Steps to reproduce the issue

1. Upgrade to 15.1 OS

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

### System information

Version | core-2025.3.4

Installation type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
User | root
Virtual environment | false
Python version | 3.13.2
Operating system family | Linux
Operating system version | 6.6.73-haos
CPU architecture | x86_64
Timezone | Europe/Budapest
Configuration directory | /config

sairon, 2025-04-02:
The RAM usage shown outside of the VM is not relevant - there's plenty of reports like this if you search for "Proxmox RAM/memory usage", e.g. this one: #3346.

Also, it seems you're not on 15.1, as the kernel version in your report doesn't match 6.12.20 from 15.1:
> Operating system version | 6.6.73-haos

litecross91, 2025-04-02:
@sairon 
I have used 15.1 since 27.03 to today. RAM usage went up to around 6 GB... 
Now I'm back to 14.2 and opened a ticket after that. No such problem with 14.2. This is not a Proxmox bug.

litecross91, 2025-04-02:
For more than two years it has been showing 2GB of ram on boot. And since 15.1, it has started with 4GB of ram. Why? They must have done something to make this happen. It bothers me.

sairon, 2025-04-02:
Once again, there's no reason to worry, Linux just uses the assigned RAM more effectively for its operation. I've provided you links that explain the details and what steps can be performed to check actual free RAM in the VM. There are tons of discussions like this on the Proxmox forums as well. If you need the RAM for something else, you can simply lower what's assigned to the VM, very much likely it is not needed and you're just overprovisioning it.

closed 2025-04-02 by sairon