home-assistant/operating-system #3998: 15.1 update resulted onboarding screen

back-2-95, 2025-04-02:
Started the update from UI (mac mini 2012) and now I just got onboarding window. Also SSH does not let me in.

ha console shows banner normally and when running commands, all is OK - BUT e.g. I cannot see my backups etc. It's like fresh installation.

Boot slots show

A = 15.1 good, booted
B = 15.0 good, inactive < are all my stuff there?

_Originally posted by @back-2-95 in https://github.com/home-assistant/operating-system/discussions/3990#discussioncomment-12695766_

sairon, 2025-04-02:
Is there anything in `/mnt/data/supervisor/homeassistant/` or is it gone too? What you describe looks like a wiped data partition, however, that is something that can't possibly happen accidentally. Are there any systemd logs from previous boots in `journalctl --list-boots`?

back-2-95, 2025-04-02:
Folder `/mnt/data/supervisor/homeassistant/` has only the fresh installation, nothing I have made. `journalctl --list-boots` shows only the boots I made during updates.

sairon, 2025-04-03:
Unfortunately, I'm running out of ideas 😢 The last possible option is that you have multiple drives mounted and the data partition was moved. In that case you should see partition labeled `hassos-data-dis` in the `blkid` output.

back-2-95, 2025-04-03:
hmm `blkid` does not list anything. I'm as root on the host.

sairon, 2025-04-03:
By "root on the host", you mean connected to the device directly, with keyboard and display? Or through any of the add-ons? Either `lsblk` doesn't report anything?

back-2-95, 2025-04-03:
Via Advanced SSH & Web Terminal

sairon, 2025-04-03:
That won't work - that way you're not really connected to the host, the commands still run in a container. You need to connect directly or using [host SSH access](https://developers.home-assistant.io/docs/operating-system/debugging#enabling-ssh-access-to-the-host).