home-assistant/operating-system #4000: bluetooth connecting quality of realtek rtl8822b bluetooth/wifi adapter is very poor with 15.1 update

raveit65, 2025-04-03:
### Describe the issue you are experiencing

The 15.1 update reduce dramatically the bluetooth connecting quality of my realtek rtl8822b bluetooth/wifi adapter.
In result i lost several switchbot temperature devices which i use to control my trvs. The distance from the adapter to the devices is max. 7-8 m which shouldn't be a problem for BLE

[image]

Also my bluetooth beacon is affected. Normally i saw him up to 40 m outside of my flat. Now i loose the connection inside my flat.
For the record, i do not use wifi from the realtek stick for my Home-server.
When booting HAOS from slot B with 14.2 the bluetooth adapter works well.
Fortunately i can use this slot. I need working temperature devices.
Btw. Update was from 14.2 to 15.1 because i couldn't use 15.0 with my HP t630 thin client.
Let me know if you need more info.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 15.1

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Home-Server: HP t630 Thin Client connected via LAN cable
RAM : 8GB
SSD: 500GB
Wifi/Bluetooth USB Adapter: 1330Mbps Dual band USB Wifi BT5.0 connected via usb-3.0
Zigbee:
   - 1x uzg-01 conected via usb-2.0
   - 2x slzb-06 connected via LAN cable

### Steps to reproduce the issue

1.Booting generic HAOS 15.1
2. Open dmesg -w in terminal
3. Unplug the Wifi/BT stick
4. Plug in the stick again
with 15.1:
```
[  385.404006] usb 3-2: USB disconnect, device number 2
[  393.599808] usb 1-2: new high-speed USB device number 3 using xhci_hcd
[  393.727105] usb 1-2: New USB device found, idVendor=0bda, idProduct=b82c, bcdDevice= 2.10
[  393.727126] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  393.727135] usb 1-2: Product: 802.11ac NIC
[  393.727141] usb 1-2: Manufacturer: Realtek
[  393.727147] usb 1-2: SerialNumber: 123456
[  393.740214] rtw_8822bu 1-2:1.2: Firmware version 27.2.0, H2C version 13
[  393.817025] Bluetooth: hci0: RTL: examining hci_ver=07 hci_rev=000b lmp_ver=07 lmp_subver=8822
[  393.818001] Bluetooth: hci0: RTL: rom_version status=0 version=2
[  393.818020] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_fw.bin
[  393.818186] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_config.bin
[  393.818274] Bluetooth: hci0: RTL: cfg_sz 14, total sz 20270
[  394.642149] Bluetooth: hci0: RTL: fw version 0xab6b705c
[  394.713569] Bluetooth: MGMT ver 1.23
[  396.191169] rtw_8822bu 1-2:1.2: write register 0xc4 failed with -71
[  396.192322] rtw_8822bu 1-2:1.2: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[  396.193282] usb 1-2: USB disconnect, device number 3
[  396.648308] usb 3-2: new SuperSpeed USB device number 3 using xhci_hcd
[  396.664944] usb 3-2: New USB device found, idVendor=0bda, idProduct=b82c, bcdDevice= 3.00
[  396.664966] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  396.664975] usb 3-2: Product: 802.11ac NIC
[  396.664981] usb 3-2: Manufacturer: Realtek
[  396.664987] usb 3-2: SerialNumber: 123456
[  396.677915] rtw_8822bu 3-2:1.2: Firmware version 27.2.0, H2C version 13
[  396.913143] Bluetooth: hci0: RTL: examining hci_ver=07 hci_rev=000b lmp_ver=07 lmp_subver=8822
[... 354 lines cut ...]
views | 6
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp1s0 (enabled, default, auto), docker0 (disabled), hassio (disabled), veth9028ab0 (disabled), veth4d40ee9 (disabled), veth8898ce8 (disabled), veth5de8f65 (disabled), veth312d1d5 (disabled), vethd0ce1f9 (disabled), veth9138bef (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp1s0 (192.168.0.3/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), veth9028ab0 (), veth4d40ee9 (), veth8898ce8 (), veth5de8f65 (), veth312d1d5 (), vethd0ce1f9 (), veth9138bef ()
ipv6_addresses | lo (::1/128), enp1s0 (2a02:2455:821a:ec00:e1fc:af33:303e:caad/64, fd8f:9079:553:0:831f:7ab8:cff0:eb3f/64, fe80::1c6c:bcf:ab06:7b4e/64), docker0 (fe80::68b9:84ff:fe98:c03b/64), hassio (fe80::1c0e:9bff:fe52:b544/64), veth9028ab0 (fe80::cce0:33ff:fe0f:8658/64), veth4d40ee9 (fe80::b045:fcff:fea2:3c23/64), veth8898ce8 (fe80::a400:45ff:fe85:5501/64), veth5de8f65 (fe80::485e:70ff:fe34:543e/64), veth312d1d5 (fe80::f471:d3ff:fe1e:67af/64), vethd0ce1f9 (fe80::a8ae:ccff:feea:357b/64), veth9138bef (fe80::48a:c2ff:fe49:d5db/64)
announce_addresses | 192.168.0.3, 2a02:2455:821a:ec00:e1fc:af33:303e:caad, fd8f:9079:553:0:831f:7ab8:cff0:eb3f, fe80::1c6c:bcf:ab06:7b4e

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | March 21, 2025 at 3:31 PM
-- | --
current_recorder_run | April 3, 2025 at 1:52 PM
estimated_db_size | 470.81 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

contactcr, 2025-04-05:
Have an Intel NUC12 and this is the 2nd time i've been burned by HA OS version causing major bluetooth issues.  Like you it "worked" but not reliably enough for stuff like my Yale bluetooth door lock which was sometimes taking a full minute or more to respond to requests.

The lock was causing a ton of the following errors in the logs:

Disconnected while updating
Timed out updating

I have a second lock which is in the same room as the NUC and it was less problematic (or at least seemed so from the logs, it doesn't get used much).

I reverted to 14.2 ( "ha os update --version 14.2" ) < in terminal or console if anyone finds this issue and wants to revert.

raveit65, 2025-04-05:
> Have an Intel NUC12 (but don't use built in bluetooth I use a dongle with Realtek chip) and this is the 2nd time i've been burned by HA OS version causing major bluetooth issues. Like you it "worked" but not reliably enough for stuff like my Yale bluetooth door lock which was sometimes taking a full minute or more to respond to requests.
> 
> The lock was causing a ton of the following errors in the logs:
> 
> Disconnected while updating Timed out updating
> 
> I have a second lock which is in the same room as the NUC and it was less problematic (or at least seemed so from the logs, it doesn't get used much).
> 
> I reverted to 14.2 ( "ha os update --version 14.2" ) < in terminal or console if anyone finds this issue and wants to revert.

Which chip for bluetooth does your adapter have?
.....i ask because i bought a new 5.3 bluetooth stick at Ali-E.

contactcr, 2025-04-05:
Brand ZEXMTE BT510.  Who knows some 5.1 usb adapter that worked good with HA at the time I bought it.  I replaced this one cause the NUC built in was acting up.  Truth is they are all only as good as the driver and OS implementation as we found out here.

raveit65, 2025-04-05:
I meant the driver name in kernel? Something like rtl.....
` dmesg | grep rtl` will show it.

contactcr, 2025-04-05:
> I meant the driver name in kernel? Something like rtl..... ` dmesg | grep rtl` will show it.
> 
> Edit:
> 
> ```
> ~ $ dmesg | grep rtl
> [    3.894712] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_fw.bin
> [    3.899383] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_config.bin
> [    7.326993] usbcore: registered new interface driver rtl88x2bu
> ```

trl_bt/rtl8761bu_fw.bin

raveit65, 2025-04-05:
OMG, that means whole bluetooth stack or minimum the rtl stack is broken :/

contactcr, 2025-04-05:
Here's my bug when it broke for the entire 10.x and 11.x major releases.  The fix was not incorporated till 12.x so...

https://github.com/home-assistant/operating-system/issues/2612

raveit65, 2025-04-06:
Curiously my rtl8822b chipset is listed as USB 2.0 at https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Chipsets.md but stick manufacturer provide a USB-3.0 connecter.
Also they talk about improvements with kernel-6.12 for rtl8822b chipset. HAOS 15.1 use kernel 6.12.20-haos.
I tried to connect my stick to a USB-2.0 port and bluetooth is proper working again with 15.1 \o/.
I see my ibeacon outside of my flat  again and all switchbot devices are connected.
Well, seems the improvement with kernel-6.12 called "Do not work with usb-3.0 ports". 😄 
But the double connecting issue in logs still exists.
```
[  946.665750] usb 2-1.4: SerialNumber: 123456
[  946.669287] rtw_8822bu 2-1.4:1.2: Firmware version 27.2.0, H2C version 13
[  946.820859] Bluetooth: hci0: RTL: examining hci_ver=07 hci_rev=000b lmp_ver=07 lmp_subver=8822
[  946.821763] Bluetooth: hci0: RTL: rom_version status=0 version=2
[  946.821828] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_fw.bin
[  946.822003] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_config.bin
[  946.822079] Bluetooth: hci0: RTL: cfg_sz 14, total sz 20270

[  947.307903] rtw_8822bu 2-1.4:1.2: write register 0xc4 failed with -71
[  947.324531] rtw_8822bu 2-1.4:1.2: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[  947.424973] usb 2-1.4: USB disconnect, device number 8
[  947.427386] Bluetooth: hci0: RTL: download fw command failed (-19)

[  947.858384] usb 2-1.4: new high-speed USB device number 9 using ehci-pci
[  947.945413] usb 2-1.4: New USB device found, idVendor=0bda, idProduct=b82c, bcdDevice= 2.10
[  947.945434] usb 2-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  947.945443] usb 2-1.4: Product: 802.11ac NIC
[  947.945450] usb 2-1.4: Manufacturer: Realtek
[  947.945455] usb 2-1.4: SerialNumber: 123456
[  947.949931] rtw_8822bu 2-1.4:1.2: Firmware version 27.2.0, H2C version 13
[  948.015658] Bluetooth: hci0: RTL: examining hci_ver=07 hci_rev=000b lmp_ver=07 lmp_subver=8822
[  948.016603] Bluetooth: hci0: RTL: rom_version status=0 version=2
[  948.016626] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_fw.bin
[  948.016798] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_config.bin
[  948.016873] Bluetooth: hci0: RTL: cfg_sz 14, total sz 20270
[  948.599659] rtw_8822bu 2-1.4:1.2 wlp0s18u1u4i2: renamed from wlan0
[  948.758488] Bluetooth: hci0: RTL: fw version 0xab6b705c
[  948.818118] Bluetooth: MGMT ver 1.23
```

raveit65, 2025-04-11:
I tested a new usb-2.0 bluetooth 5.3 only stick with rtl8761bu chipset which works out of box when connected to usb 2.0 or 3.0 port. I didn't noticed any connecting quality issues. But the blutooth range is to large for my ibeacon usecase.

dominikandreas, 2025-04-20:
Wifi with a rtl88XXau USB stopped working for me entirely with 15.2

closed 2026-02-24 by github-actions[bot]