home-assistant/operating-system #4003: haos_rpi4-64-15.1.img.xz doesn't work properly - unable to reach <host>:8123

efferre79, 2025-04-03:
### Describe the issue you are experiencing

I have downloaded the image haos_rpi4-64-15.1.img.xz and installed on a Raspberry Pi4 board by using RPi Imager 1.8.5. I connected the raspberry to a monitor to see the initialization. The system starts correctly and I see the HA console prompt but I cannot reach the webpage at port 8123. The same procedure instead works with previous release image haos_rpi4-64-15.0.img.xz.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

15.1

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

raspberry Pi4 4GB 

### Steps to reproduce the issue

1. download image haos_rpi4-64-15.1.img.xz 
2. burn it with rpi imager
3. insert the SD card and start the raspberry
...

### Anything in the Supervisor logs that might be useful for us?

```txt
unable to see that, HA is still not initialized
```

### Anything in the Host logs that might be useful for us?

```txt
I didn't try connecting via ssh, HA is still not initialized
```

sairon, 2025-04-07:
There are no changes between 15.0 and 15.1 that should possibly affect this - yes, there's a Docker update but that would manifest in runtime as well and possibly more things wouldn't start. To be sure, I flashed an SD card with `haos_rpi4-64-15.1.img.xz` using RPi Imager (1.9.0) as well and everything starts correctly - there must have been some fluke on your side either in the flashing process or in the OS startup. You could check what is (not) running by typing `login` in the HA command line and checking output of `ha supervisor logs`, `docker ps` or `journalctl --list-failed`. Anyway, this doesn't seem to be a systemic bug, I'm closing this issue unless we have any evidence of proving otherwise.

closed 2025-04-07 by sairon

efferre79, 2025-04-08:
I had tried twice when I reported here. Today I have checked the SD card (a regular class 10 card) for badblocks and everything looks fine. Eventually I have just tried once more time tonight and it worked! Probably some glitch during initialization but it definitely not repeatable so no good reason to have this report open.