home-assistant/operating-system #4020: ODROID C2 running from SD card hangs at reboot

soosp, 2025-04-11:
### Describe the issue you are experiencing

If you set up HA OS to SD card on an ODROID C2 and try to reboot your system it stops, but needs power cycle to boot again. It is a known issue, and have solution: https://forum.armbian.com/topic/27137-if-you-have-an-odroid-c2-and-possibly-others-that-fails-to-reboot-heres-your-fix/
It was tested by me, and works.

### What operating system image do you use?

odroid-c2 (Hardkernel ODROID-C2)

### What version of Home Assistant Operating System is installed?

15.1

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

OROID C2
ADATA IDU3A 32GB Industrial SD card

### Steps to reproduce the issue

1. Install HA OS for ODROID C2 to an SD card
2. Boot it
3. Start a reboot

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing
```

### System information

## System Information

version | core-2025.4.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.12.20-haos
arch | aarch64
timezone | Europe/Budapest
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1647
Downloaded Repositories | 6

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.1
-- | --
update_channel | stable
supervisor_version | supervisor-2025.04.0
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 28.9 GB
disk_used | 3.9 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | odroid-c2
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (20.0.2), File editor (5.8.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 3
mode | auto-gen

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), end0 (enabled, default, auto), docker0 (disabled), hassio (disabled), veth45655da (disabled), veth8f8e61e (disabled), veth9cff04d (disabled), veth594929a (disabled), veth61b49b0 (disabled), veth3f7f22d (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), end0 (192.168.44.232/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), veth45655da (), veth8f8e61e (), veth9cff04d (), veth594929a (), veth61b49b0 (), veth3f7f22d ()
ipv6_addresses | lo (::1/128), end0 (fd40:738:2e00:44:d8fd:af6c:2f43:462c/128, fd40:738:2e00:44:4ef7:c897:685b:3765/64, fe80::d888:4ef7:37fc:5fbc/64), docker0 (fe80::c84:42ff:fee7:d1fb/64), hassio (fe80::f4f1:52ff:fe3d:32c1/64), veth45655da (fe80::508b:7dff:fee9:f0e5/64), veth8f8e61e (fe80::d86a:87ff:fef4:eb8d/64), veth9cff04d (fe80::f011:bfff:fef3:b36/64), veth594929a (fe80::5cdb:73ff:feeb:3ec1/64), veth61b49b0 (fe80::d8c5:bfff:fe79:2a36/64), veth3f7f22d (fe80::d8e4:2cff:fe86:2ba0/64)
announce_addresses | 192.168.44.232, fd40:738:2e00:44:d8fd:af6c:2f43:462c, fd40:738:2e00:44:4ef7:c897:685b:3765, fe80::d888:4ef7:37fc:5fbc

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 2025. április 9. 22:11
-- | --
current_recorder_run | 2025. április 11. 22:16
estimated_db_size | 5.43 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

soosp, 2025-04-14:
Meanwhile I got an eMMC card for Odroid C2 and test it. A fresh eMMC install of HA OS reboots without any issue.  Reboot issue appears only if you use SD card. With the modified dtb file both (SD only and eMMC only) configurations reboot without problem.

agners, 2025-05-12:
> It was tested by me, and works.

So without that patch, your ODROID C2 did not boot from SD card? I just tested the current release, and it does boot fine from SD card on my ODROID C2 (REV0.2 2017114). I am using a SanDisk Extreme UHS-II card.

The configuration suggested in the forum post 0x06 would mean open drain essentially:

```c
/* Bit 0 express polarity */
#define GPIO_ACTIVE_HIGH 0
#define GPIO_ACTIVE_LOW 1

/* Bit 1 express single-endedness */
#define GPIO_PUSH_PULL 0
#define GPIO_SINGLE_ENDED 2

/* Bit 2 express Open drain or open source */
#define GPIO_LINE_OPEN_SOURCE 0
#define GPIO_LINE_OPEN_DRAIN 4

/*
 * Open Drain/Collector is the combination of single-ended open drain interface.
 * Open Source/Emitter is the combination of single-ended open source interface.
 */
#define GPIO_OPEN_DRAIN (GPIO_SINGLE_ENDED | GPIO_LINE_OPEN_DRAIN)
#define GPIO_OPEN_SOURCE (GPIO_SINGLE_ENDED | GPIO_LINE_OPEN_SOURCE)
```

When I look at the schematic, the configuration doesn't really make sense to me.

[image]

soosp, 2025-05-13:
> So without that patch, your ODROID C2 did not boot from SD card?

It boots at power up, but the at a soft reboot it is hanging without the patch.