home-assistant/operating-system #4022: Raspberry Pi 4 bluetooth gets stuck in passive scanning mode until a full system reboot

arjanvlek, 2025-04-12:
### Describe the issue you are experiencing

I've recently started using HA on my Raspberry Pi 4. I'm using Home Assistant OS 15.1, but once in a while, my Govee bluetooth thermometer stops updating.
In the logs, at this moment, there is the following error logged:

```
hci0 (DC:A6:32:8B:D7:63): Error stopping scanner: [org.bluez.Error.InProgress] Operation already in progress
hci0 (DC:A6:32:8B:D7:63): Successful fall-back to passive scanning mode after active scanning failed (4/4)
```

I cannot get the bt adapter back into active scanning mode other than fully rebooting the system (just restarting home assistant is not enough)

Initially, I had reported this issue for the bluetooth integration (https://github.com/home-assistant/core/issues/142656) but I was told this is a driver / hardware issue and this needs to be fixed by the OS. So I'm including the details of my problem here.

**Error in HA Core:**

[image]

**Bluetooth entry configuration**

[config_entry-bluetooth-01JQW6JTZA5PCNDJRKWKYNQWT3.json](https://github.com/user-attachments/files/19720354/config_entry-bluetooth-01JQW6JTZA5PCNDJRKWKYNQWT3.json)

**Debug log of HA Core**

Log is too large to upload to GitHub, so I shared it on Onedrive: https://1drv.ms/u/c/2487a0d63a2b7804/EVcL2jQo0fZHjPwfRP0sK_cB4zjaY5en5UZVPNHnWJm6oQ 

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

15.1

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

- Raspberry Pi 4 with 4GB RAM
- 32GB Samsung micro sd card
- Official Raspberry Pi power supply
- Network connection via Wifi (I don't have an ethernet port free on my router, unfortunately, but network connectivity does not seem to be the issue here)

### Steps to reproduce the issue

1. Start Home Assistant
2. After a few days, notice that temperature of Govee stops updating (flat line)
3. Open system logs, notice the entry shared above (org.bluez.error.InProgress)
4. Need to reboot the system to get issue resolved
...

### Anything in the Supervisor logs that might be useful for us?

```txt
-- Supervisor log seems to be in local timezone (CEST)
2025-04-11 15:20:48.137 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2025-04-11 15:20:48.138 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2025-04-11 15:22:04.955 INFO (MainThread) [supervisor.host.info] Updating local host information
2025-04-11 15:22:05.690 INFO (MainThread) [supervisor.host.services] Updating service information
2025-04-11 15:22:05.701 INFO (MainThread) [supervisor.host.network] Updating local network information
2025-04-11 15:22:05.923 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
2025-04-11 15:22:05.934 INFO (MainThread) [supervisor.host.manager] Host information reload completed
2025-04-11 15:22:07.052 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2025-04-11 15:52:07.425 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2025-04-11 16:20:48.150 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
```

### Anything in the Host logs that might be useful for us?

```txt
-- host log seems to use UTC, CEST is UTC+2. So this is at 15:29:21 local time
2025-04-11 13:29:21.989 homeassistant bluetoothd[549]: Path /org/bleak/67/546996808160 reserved for Adv Monitor app :1.47
```

### System information

## System Information

version | core-2025.4.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.6.74-haos-raspi
arch | aarch64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.1
-- | --
update_channel | stable
supervisor_version | supervisor-2025.04.0
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 28.6 GB
disk_used | 5.2 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (20.0.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 1
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), wlan0 (enabled, default, auto), docker0 (disabled), hassio (disabled), veth491a74c (disabled), vetha415e0e (disabled), veth0618f82 (disabled), vethc83864f (disabled), veth0db6a27 (disabled), veth0817a39 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), wlan0 (192.168.178.80/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), veth491a74c (), vetha415e0e (), veth0618f82 (), vethc83864f (), veth0db6a27 (), veth0817a39 ()
ipv6_addresses | lo (::1/128), wlan0 (fe80::f4cb:b469:a6ba:c9c2/64), docker0 (fe80::f038:1aff:feed:cf76/64), hassio (fe80::74d2:45ff:fed9:a3b5/64), veth491a74c (fe80::e83d:a6ff:fe79:e26a/64), vetha415e0e (fe80::cad:21ff:fec0:4568/64), veth0618f82 (fe80::d08e:ff:fe04:c053/64), vethc83864f (fe80::4c20:2eff:febf:5346/64), veth0db6a27 (fe80::68b6:5cff:fe0c:d059/64), veth0817a39 (fe80::a0b9:2dff:fe20:f57d/64)
announce_addresses | 192.168.178.80, fe80::f4cb:b469:a6ba:c9c2

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 26 maart 2025 om 06:56
-- | --
current_recorder_run | 12 april 2025 om 01:36
estimated_db_size | 115.55 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

### Additional information

Other bluetooth devices, such as Aranet4, keep working even after this issue has occurred. They probably don't require active scanning.

bsjouke, 2025-04-18:
Same issue here. Rollback to 2025.4.1 did not help so OS upgrade should be the cause.

arjanvlek, 2025-05-11:
As a workaround, I've added a CSR bluetooth adapter to my system. This seems to fix the issue with the govee thermometer.
I could not disable the internal Pi bluetooth though, since it causes HA to lose all device pairings (e.g. Aranet) and wouldn't re-pair the devices with only the CSR dongle enabled.

See https://www.home-assistant.io/integrations/bluetooth#known-working-high-performance-adapters

closed 2025-09-02 by github-actions[bot]