home-assistant/operating-system #4031: Fritz Nas Network Storage unavailable after Update to OS15.2

Stothed775, 2025-04-16:
### Describe the issue you are experiencing

I have an issue with network Storage via Fritz Nas.
Since Update to OS15.2 the Storage ist not available and also could not be added again.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

15.2

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Beelink S12Pro
one Sky Connect, One ZBT-1

### Steps to reproduce the issue

Problem occurs directly after Upgrading the OS

### Anything in the Supervisor logs that might be useful for us?

```txt
2025-04-16 17:00:57.582 ERROR (MainThread) [supervisor.mounts.mount] Reloading frigate did not succeed. Check host logs for errors from mount or systemd unit mnt-data-supervisor-mounts-frigate.mount for details.
2025-04-16 17:01:01.735 INFO (MainThread) [supervisor.mounts.manager] Creating or updating mount: frigate
2025-04-16 17:01:01.785 ERROR (MainThread) [supervisor.mounts.mount] Reloading frigate did not succeed. Check host logs for errors from mount or systemd unit mnt-data-supervisor-mounts-frigate.mount for details.
2025-04-16 17:02:11.457 INFO (MainThread) [supervisor.mounts.manager] Creating or updating mount: frigate
2025-04-16 17:02:11.500 ERROR (MainThread) [supervisor.mounts.mount] Reloading frigate did not succeed. Check host logs for errors from mount or systemd unit mnt-data-supervisor-mounts-frigate.mount for details.
2025-04-16 17:04:27.301 INFO (MainThread) [supervisor.mounts.manager] Removing mount: Backup
2025-04-16 17:04:27.343 INFO (MainThread) [supervisor.mounts.manager] Creating or updating mount: Backup
2025-04-16 17:04:27.385 ERROR (MainThread) [supervisor.mounts.mount] Mounting Backup did not succeed. Check host logs for errors from mount or systemd unit mnt-data-supervisor-mounts-Backup.mount for details.
```

### Anything in the Host logs that might be useful for us?

```txt
2025-04-16 14:54:42.271 homeassistant systemd[1]: Unmounting Supervisor bind mount: bind_frigate...
2025-04-16 14:54:42.275 homeassistant systemd[1]: mnt-data-supervisor-media-frigate.mount: Mount process exited, code=exited, status=32/n/a
2025-04-16 14:54:42.275 homeassistant systemd[1]: Failed unmounting Supervisor bind mount: bind_frigate.
2025-04-16 14:55:12.300 homeassistant systemd[1]: Unmounting Supervisor bind mount: bind_frigate...
2025-04-16 14:55:12.304 homeassistant systemd[1]: mnt-data-supervisor-media-frigate.mount: Mount process exited, code=exited, status=32/n/a
2025-04-16 14:55:12.304 homeassistant systemd[1]: Failed unmounting Supervisor bind mount: bind_frigate.
2025-04-16 14:55:12.308 homeassistant systemd[1]: Unmounting Supervisor cifs mount: frigate...
2025-04-16 14:55:12.318 homeassistant systemd[1]: mnt-data-supervisor-mounts-frigate.mount: Deactivated successfully.
2025-04-16 14:55:12.318 homeassistant systemd[1]: Unmounted Supervisor cifs mount: frigate.
2025-04-16 14:55:12.345 homeassistant systemd[1]: Mounting Supervisor cifs mount: frigate...
2025-04-16 14:55:12.346 homeassistant mount[136652]: error 2 (No such file or directory) opening credential file /mnt/data/supervisor/.mounts_credentials/frigate
2025-04-16 14:55:12.347 homeassistant systemd[1]: mnt-data-supervisor-mounts-frigate.mount: Mount process exited, code=exited, status=2/INVALIDARGUMENT
2025-04-16 14:55:12.347 homeassistant systemd[1]: mnt-data-supervisor-mounts-frigate.mount: Failed with result 'exit-code'.
2025-04-16 14:55:12.347 homeassistant systemd[1]: Failed to mount Supervisor cifs mount: frigate.
```

hewimeddel, 2025-05-08:
For your Information:
After updating my HomeAssistant Green from 15.1 to 15.2 I do not have any trouble with my Fritz Nas Storage as Backup-Target.
(I do not use Frigate.)

Just wanted to share this information because i waited for weeks with the updated due to this open issue.

HomeAssistantPim, 2025-07-20:
I experience the same on HAOX Green.
It is looking for a mount at /data/mounts/Backup/ which is not there.
In fact /data/mounts is not even there.
I also use frigate to write to the NAS, this works.
Removed the storage in HA and put it back in as a backup storage location: no succes.

Log states:
```
Logger: homeassistant.components.backup
Bron: components/backup/manager.py:1149
integratie: Backup ([documentatie](https://www.home-assistant.io/integrations/backup), [problemen](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+backup%22))
Eerst voorgekomen: 22:50:36 (2 gebeurtenissen)
Laatst gelogd: 22:52:18

    Error creating backup: Backup failed: [{'type': 'BackupError', 'message': 'Cannot open backup file /data/mounts/Backup/Automatic_backup_2025.5.3_2025-07-20_22.50_36513376.tar for writing', 'stage': 'addon_repositories'}]
    Error creating backup: Backup failed: [{'type': 'BackupError', 'message': 'Cannot open backup file /data/mounts/Backup/test_2025-07-20_22.52_18202985.tar for writing', 'stage': 'addon_repositories'}]```