home-assistant/operating-system #4033: nuc nuc6cayh no ecc memory

srw1mpo, 2025-04-20:
### Describe the issue you are experiencing

First of all I have very limited knowledge on Hone Assistant architecture or Linux.

I bought a nuc nuc6cayh in January, and I've not been able to upgrade os.

After upgrading, the system hangs during reboot with error

"EDAC pnd2: Failed to register device with error -22"

I can get system to restart but it reverts back to the old os version.

The only way ive been able to upgrade is to install latest disk image from scratch and restore from backup.

I have read that the EDAC error is something to do with ecc memory.

My NUC doesn't support ecc so I think that's causing the issue.

Is there a way around this, or do I need to get a different NUC that supports ecc?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

15.1

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

I use a SONOFF Universal Zigbee 3.0 USB Dongle.

### Steps to reproduce the issue

1. Update os via fronted or cli
2. System won't reboot to new os version
3. Boots into old os version
...

### Anything in the Supervisor logs that might be useful for us?

```txt
N/a
```

### Anything in the Host logs that might be useful for us?

```txt
N/a
```

### System information

## System Information

version | core-2025.4.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.12.20-haos
arch | x86_64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1666
Downloaded Repositories | 15

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 27

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | April 22, 2025 at 01:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
cloud_ice_servers_enabled | true
remote_server | eu-central-1-21.ui.nabu.casa
certificate_status | ready
instance_id | 49523d7bbbbf43149109fb78262f2b4d
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.1.rc1
-- | --
update_channel | stable
supervisor_version | supervisor-2025.04.0
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 116.7 GB
disk_used | 10.7 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.5.0), Get HACS (1.3.1), Everything Presence Zone Configurator (1.1.2), ESPHome Device Builder (2025.4.0), Home Assistant Google Drive Backup (0.112.1), Advanced SSH & Web Terminal (20.0.2), SQLite Web (4.3.1), Hikvision Doorbell (3.0.19), Samba share (12.5.0), File editor (5.8.0), Studio Code Server (5.19.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 9
-- | --
resources | 7
views | 13
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp3s0 (enabled, default, auto), docker0 (disabled), hassio (disabled), veth68d008e (disabled), vethbdb8b8e (disabled), veth4f831e7 (disabled), vethaa42d12 (disabled), veth551c41c (disabled), veth2c8a28e (disabled), vethb3ea990 (disabled), veth33928a3 (disabled), veth5466c56 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp3s0 (192.168.0.180/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), veth68d008e (), vethbdb8b8e (), veth4f831e7 (), vethaa42d12 (), veth551c41c (), veth2c8a28e (), vethb3ea990 (), veth33928a3 (), veth5466c56 ()
ipv6_addresses | lo (::1/128), enp3s0 (fd78:932b:ccd7:d28b:8930:afba:ba48:1858/64, fe80::1815:1d8f:8f90:18f6/64), docker0 (fe80::5cdf:46ff:feda:b45f/64), hassio (fe80::1066:98ff:fece:c5b0/64), veth68d008e (fe80::1865:5fff:fec1:2f55/64), vethbdb8b8e (fe80::e4d2:5eff:fe09:d0a0/64), veth4f831e7 (fe80::fc3b:a1ff:fe85:b122/64), vethaa42d12 (fe80::1428:8cff:fea3:c54d/64), veth551c41c (fe80::e85c:f5ff:fe5f:ec77/64), veth2c8a28e (fe80::446e:e4ff:fec2:d049/64), vethb3ea990 (fe80::70c4:d2ff:feef:4849/64), veth33928a3 (fe80::d49a:83ff:fe6f:92b4/64), veth5466c56 (fe80::f433:d8ff:fe68:d767/64)
announce_addresses | 192.168.0.180, fd78:932b:ccd7:d28b:8930:afba:ba48:1858, fe80::1815:1d8f:8f90:18f6

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | April 8, 2025 at 15:17
-- | --
current_recorder_run | April 20, 2025 at 10:42
estimated_db_size | 554.55 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

srw1mpo, 2025-04-21:
Update - I have reinstalled using Proxmox, and I can now update os.

I don't know if ecc memory is the problem or not, but I'm not having the issue now. So I would still like any input on the issue in case proxmox doesn't work out for me. Thanks

sairon, 2025-05-06:
The message you refer to should be harmless and unrelated to the issues you saw. Most likely something else caused the updates to fail. But it's hard to tell as you didn't provide any logs. Hopefully Proxmox will work on your hardware better, it would be eventually easier to troubleshoot HAOS issues there.

closed 2025-05-06 by sairon

srw1mpo, 2025-05-06:
Thanks for the reply. I wasn't sure how to get the logs that where relevant and if they hadn't been over written as it had been a while since I'd tried to update.

Proxmox working great now. Wish I'd gone down that route at first.

Regards

Steve