home-assistant/operating-system #4037: Upgrade from 14.0 to 14.1 and 15.2 both fail with boot loop

camsaway, 2025-04-23:
### Describe the issue you are experiencing

Hi,
I'm currently running HASSOS on an Intel Mini-PC
A bit over a month ago I tried to upgrade from 14.0 to 14.1 and my device would keep rebooting about 30 seconds after HASS UI started up - not enough time to get to the logs screen to get any useful info.
I reverted to 14.0 with a directly attached keyboard and monitor with "ha os boot-slot swap" - there was enough time to do this before the device reboots.
I just now tried again but this time to 15.2 and hit the same problem.
Where/How can I get some information/logs to look at and see what might be causing this?
I've done lots of updates over the last 3-4 years and never had a problem before.
Thanks.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

15.2

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Beelink U59 Mini PC 11th Gen 4-Cores N5105, Office Mini Computer 16GB DDR4 500G SSD, 4K UHD Triple Display Dual HDMI Type-C Dual LAN/WiFi5/BT4.0/Auto

I normally have nothing connected, but did plug in a USB Keyboard/Mouse Combo and HDMI monitor in order to revert back to the previous OS

### Steps to reproduce the issue

1. Select Upgrade from the UI
2. Wait for upgrade to complete and reboot
3. System will then go into a boot loop about 30 secs after the UI comes up.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No unless there is a way to get logs from the previous boot / failure. Please tell me how
```

### Anything in the Host logs that might be useful for us?

```txt
No unless there is a way to get logs from the previous boot / failure. Please tell me how
```

### System information

I can't get to this during a failure, but here is the info running on the rolled back version 14

## System Information

version | core-2025.3.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.6.63-haos
arch | x86_64
timezone | Africa/Johannesburg
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4989
Installed Version | 2.0.5
Stage | running
Available Repositories | 1728
Downloaded Repositories | 2

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 24

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | September 23, 2025 at 02:00
relayer_connected | true
relayer_region | us-east-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
cloud_ice_servers_enabled | true
remote_server | us-east-1-7.ui.nabu.casa
certificate_status | ready
instance_id | 086fa9438c264a95a4badd5af1f27bde
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.0
-- | --
update_channel | stable
supervisor_version | supervisor-2025.04.0
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 468.7 GB
disk_used | 35.8 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Studio Code Server (5.19.2), ESPHome Device Builder (2025.4.0), Mosquitto broker (6.5.0), SQLite Web (4.2.1), Samba share (12.5.0), chrony (5.1.0), Paradox Alarm Interface (3.6.1), Apache2 Minimal (2.12.4), OneDrive Backup (2.3.8), Advanced SSH & Web Terminal (20.0.2), syslogng (1.1.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 8
-- | --
resources | 3
views | 20
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp3s0 (enabled, default, auto), docker0 (disabled), hassio (disabled), veth837a4ab (disabled), vethb17e3c8 (disabled), vethfd7c8e6 (disabled), veth1c6a8bb (disabled), vethc8f2ee5 (disabled), veth7335236 (disabled), veth687937d (disabled), vethd68a15a (disabled), veth2c8cd5b (disabled), vethe95531b (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp3s0 (192.168.1.182/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), veth837a4ab (), vethb17e3c8 (), vethfd7c8e6 (), veth1c6a8bb (), vethc8f2ee5 (), veth7335236 (), veth687937d (), vethd68a15a (), veth2c8cd5b (), vethe95531b ()
ipv6_addresses | lo (::1/128), enp3s0 (fe80::9f4:d76d:a94e:6603/64), docker0 (fe80::42:c5ff:fee9:2019/64), hassio (fe80::42:92ff:fe89:1085/64), veth837a4ab (fe80::ccab:b9ff:fe48:6f45/64), vethb17e3c8 (fe80::4ca7:b3ff:fea2:bb48/64), vethfd7c8e6 (fe80::f8a1:2ff:fe5e:2f8b/64), veth1c6a8bb (fe80::a4e2:78ff:fe67:8571/64), vethc8f2ee5 (fe80::e8a4:fff:feea:53c4/64), veth7335236 (fe80::8c97:82ff:fefc:7b7/64), veth687937d (fe80::d423:65ff:fe10:2180/64), vethd68a15a (fe80::42b:cff:fed0:d69f/64), veth2c8cd5b (fe80::540e:ddff:fe95:b7c6/64), vethe95531b (fe80::c4e2:38ff:fe72:a49/64)
announce_addresses | 192.168.1.182, fe80::9f4:d76d:a94e:6603

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | April 7, 2025 at 04:50
-- | --
current_recorder_run | April 23, 2025 at 20:08
estimated_db_size | 2163.84 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

<details><summary>Solcast PV Forecast</summary>

can_reach_server | ok
-- | --

</details>

<details><summary>Sonoff</summary>

version | 3.8.1 (ffa7e22)
-- | --
cloud_online | 3 / 3
local_online | 1 / 1

</details>

### Additional information

I spent hours googling but can't find any documentation on how to get the log history for the failed boots

ansgar-santic, 2025-04-24:
Same, on synology VM with HAos 15.2 getting stuck on supervisor state: setup and cant boot. 

Rollback to 15.1 and watching via Synology VM commandwindow shows at start 15.1 chooses Slot A and 15.2 chooses Slot B. But still same error: 

`homeassistant error returned from supervisor system is not ready with state setup`

admn-monkey, 2025-05-03:
My system is also having an issue upgrading. It will not upgrade from 14.2 to any 15.x version without getting a CRC error on the SATA drive on boot. That results in it failing the drive and either getting stuck waiting on the partition or going into a connect / reset loop on the SATA drive. Drive diagnostics come back perfect; there is nothing wrong with my drive. A fresh install using the Ubuntu USB method of 15.x does the same thing and I tried all the versions (insert Sleestak comment here). I re-installed 14.2 (Ubuntu method), accessed the web page from my PC, and restored from a network backup. Apparently development on Home Assistant has no issues moving forward without fixing major issues with the x86 install. Already on 15.2 and no attention is being paid to this issue or the others that likely relate to the same problem.

admn-monkey, 2025-05-06:
I should have included my hardware details. I'm running on an HP EliteDesk G5R49UT (mini-pc) and an OCZ Agility 2 SATA drive. So yes, OCZ is reasonably well known, but certainly an off brand drive. That said, what manufacturers are on brand? Samsung, Crucial, Western Digital?

camsaway, 2025-05-06:
> [@camsaway](https://github.com/camsaway) This looks very much like [#3948](https://github.com/home-assistant/operating-system/issues/3948) - there were more reports of issues with unbranded SATA SSDs that identified as `512GB SSD` (where usually manufacturer/device model is stated), actually even on some unspecified Beelink PCs. Adding `libata.force=nolpm` to `/mnt/boot/cmdline.txt` could help here as well.
> 
Thanks @sairon  - agree it does sound very similar - I even have exactly the same mini-PC (Beelink U59) as one of the OPs. I'll try the suggestion later in the week.

admn-monkey, 2025-05-10:
Well, I'm happy to say that replacing the OCZ with a Crucial worked perfectly for me. Thanks!

camsaway, 2025-05-11:
@sairon - I can also confirm this seems to be stable (up for almost an hour now). I will keep monitoring, but when can we expect this to be included into a new version? Or is this something I would need to do (grub menu) with every upgrade?

camsaway, 2025-05-19:
@sairon - so I dicovered that the setting doesn't persist across reboots. I tried using "Advanced SSH and Web Terminal" to create a file /mnt/boot/cmdline.txt with just that one command, but it didn't seem to boot nicely after that, so I took it away. (My /mnt folder existed but it was empty so I did mkdir boot first. I also chmod 777 on cmdline.txt file just in case but it was owned by root)

Any suggestions? I'm afraid this one is above my skill level for offering to contribute a fix.

sairon, 2025-05-21:
@camsaway Unfortunately, since the device uses a bogus identification string, it's unlikely the fix for it will be included in the Linux kernel. But you can persist the change by modifying `/mnt/boot/cmdline.txt` either when you directly connect to the host (with a display and keyboard and type `login` to the HA CLI) or through [Developer SSH](https://developers.home-assistant.io/docs/operating-system/debugging#enabling-ssh-access-to-the-host). Nether the standard or the "advanced" SSH add-on can be used for this.

camsaway, 2025-05-21:
Hi @sairon - thank you so much for all the help and feedback. Really appreciated.

closed 2025-05-21 by camsaway