home-assistant/operating-system #4053: /usr/sbin/hassos-cli: line 9: docker: not found

yinchak, 2025-05-06:
### Describe the issue you are experiencing

Sorry for asking some question, I am try to build Raspberry Pi 5 HA OS with source code, but after successful build the image and but it to Raspberry Pi 5, it cause error , /usr/sbin/hassos-cli: line 9: docker: not found . may i ask why i got this error and how to fix it?

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

15.3

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Raspberry Pi 5

### Steps to reproduce the issue

1.git clone https://github.com/home-assistant/operating-system.git
2.sudo scripts/enter.sh
3.make O=output_rpi5_64 rpi5_64

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

yinchak, 2025-05-06:
Hello , I followed https://developers.home-assistant.io/docs/operating-system/getting-started . here is my command step:
1. git clone https://github.com/home-assistant/operating-system/
2. cd operating-system/
3. git submodule update --init
4. sudo scripts/enter.sh
5. export FORCE_UNSAFE_CONFIGURE=1
6. make O=output rpi4_64
But the image still havenot Docker , I see the file Docker-init file is broken , Is there any way to fix or can anyone provide a full step to build with source code? thank you very much

sairon, 2025-05-06:
> export FORCE_UNSAFE_CONFIGURE=1

This should not be needed. Do not run `scripts/enter.sh` as root, rather than as an unprivileged user. 

Also please elaborate on `I see the file Docker-init file is broken`. Eventually, if you have a place where you could upload complete build logs and/or the `.img.xz` file, I could have a look what could be wrong.

Anyway, no bug seen here, Discussions would be a better place for this.