home-assistant/operating-system #4059: Bluetooth (Intel AC 3165) adapter not recognized on HAOS 15.x

bieniu, 2025-05-10:
### Describe the issue you are experiencing

After updating HA OS from version 15.1 to 15.2 Bluetooth adapter stopped being detected. Downgrading system to version 15.1 does not help, the problem still occurs.
This is Intel Dual Band Wireless-AC 3165 802.11ac and Bluetooth adapter https://usb-ids.gowdy.us/read/UD/8087/0a2a
Previously, this adapter was detected and worked properly with Bluetooth integration.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

15.1

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Intel NUC5CPYH
connected devices:

- Home Assistant SkyConnect 1.0, firmware: EmberZNet Zigbee 7.4.4.1
- Aeotec ZW090 Z-Stick Gen5, firmware 1.0

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2025-05-10 13:16:23.487 WARNING (MainThread) [supervisor.hardware.monitor] Ignore device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5.1 / failes to initialize by udev
2025-05-10 13:16:25.534 WARNING (MainThread) [supervisor.hardware.monitor] Ignore device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5.1 / failes to initialize by udev
2025-05-10 13:16:25.546 WARNING (MainThread) [supervisor.hardware.monitor] Ignore device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5.1 / failes to initialize by udev
```

### Anything in the Host logs that might be useful for us?

```txt
2025-05-10 10:34:20.344 nuc kernel: usb 1-5.1: new full-speed USB device number 68 using xhci_hcd
2025-05-10 10:34:20.430 nuc kernel: usb 1-5.1: New USB device found, idVendor=8087, idProduct=0a2a, bcdDevice= 0.01
2025-05-10 10:34:20.441 nuc kernel: usb 1-5.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
2025-05-10 10:34:20.444 nuc kernel: sdhci-pci 0000:00:12.0: SDHCI controller found [8086:2296] (rev 21)
2025-05-10 10:34:20.504 nuc systemd[1]: Reached target Bluetooth Support.
2025-05-10 10:34:22.202 nuc kernel: usb 1-5.1: USB disconnect, device number 68
2025-05-10 10:34:22.209 nuc kernel: Bluetooth: hci0: sending initial HCI reset failed (-19)
2025-05-10 10:34:22.228 nuc systemd[1]: Stopped target Bluetooth Support.
2025-05-10 10:34:22.385 nuc kernel: usb 1-5.1: new full-speed USB device number 69 using xhci_hcd
2025-05-10 10:34:22.469 nuc kernel: usb 1-5.1: New USB device found, idVendor=8087, idProduct=0a2a, bcdDevice= 0.01
2025-05-10 10:34:22.484 nuc kernel: usb 1-5.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
2025-05-10 10:34:22.489 nuc kernel: sdhci-pci 0000:00:12.0: SDHCI controller found [8086:2296] (rev 21)
2025-05-10 10:34:22.542 nuc systemd[1]: Reached target Bluetooth Support.
2025-05-10 10:34:24.249 nuc kernel: usb 1-5.1: USB disconnect, device number 69
2025-05-10 10:34:24.263 nuc kernel: Bluetooth: hci0: sending initial HCI reset failed (-19)
2025-05-10 10:34:24.272 nuc systemd[1]: Stopped target Bluetooth Support.
2025-05-10 10:34:24.427 nuc kernel: usb 1-5.1: new full-speed USB device number 70 using xhci_hcd
2025-05-10 10:34:24.505 nuc kernel: usb 1-5.1: New USB device found, idVendor=8087, idProduct=0a2a, bcdDevice= 0.01
2025-05-10 10:34:24.506 nuc kernel: usb 1-5.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
2025-05-10 10:34:24.519 nuc kernel: sdhci-pci 0000:00:12.0: SDHCI controller found [8086:2296] (rev 21)
2025-05-10 10:34:24.526 nuc kernel: sdhci-pci 0000:00:12.0: SDHCI controller found [8086:2296] (rev 21)
2025-05-10 10:34:24.574 nuc systemd[1]: Reached target Bluetooth Support.
2025-05-10 10:34:26.299 nuc kernel: usb 1-5.1: USB disconnect, device number 70
2025-05-10 10:34:26.306 nuc kernel: Bluetooth: hci0: sending initial HCI reset failed (-19)
2025-05-10 10:34:26.336 nuc systemd[1]: Stopped target Bluetooth Support.
```

### System information

## System Information

version | core-2025.5.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.3
os_name | Linux
os_version | 6.12.20-haos
arch | x86_64
timezone | Europe/Warsaw
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1729
Downloaded Repositories | 12

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 13

</details>

<details><summary>Airly</summary>

can_reach_server | ok
-- | --
requests_remaining | 865
requests_per_day | 1000

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 31 maja 2025 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | false
cloud_ice_servers_enabled | true
remote_server | eu-central-1-12.ui.nabu.casa
certificate_status | ready
instance_id | 13b6d728053d4402929e6a5f7262fca8
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.1
-- | --
update_channel | stable
supervisor_version | supervisor-2025.04.1
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 116.7 GB
disk_used | 16.5 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.5.0), Z-Wave JS (0.13.1), MariaDB (2.7.2), Vaultwarden (Bitwarden) (0.24.1), Cloudflared (5.3.0), ESPHome Device Builder (2025.4.1), Samba share (12.5.1), Terminal & SSH (9.17.0), chrony (5.1.0), Matter Server (7.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 21
-- | --
resources | 5
views | 25
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp3s0 (enabled, default, auto), docker0 (disabled), hassio (disabled), veth494046f (disabled), vethbc7ad50 (disabled), vethcf7c32d (disabled), veth652c58c (disabled), veth129eace (disabled), veth6993965 (disabled), veth9662484 (disabled), vetha51542d (disabled), vethb39b7f5 (disabled), veth409cdf5 (disabled), veth95e25f6 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp3s0 (192.168.2.15/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), veth494046f (), vethbc7ad50 (), vethcf7c32d (), veth652c58c (), veth129eace (), veth6993965 (), veth9662484 (), vetha51542d (), vethb39b7f5 (), veth409cdf5 (), veth95e25f6 ()
ipv6_addresses | lo (::1/128), enp3s0 (fe80::6d63:d225:6290:94c6/64), docker0 (fe80::782b:50ff:febe:32c/64), hassio (fe80::984f:60ff:fe5d:d9c2/64), veth494046f (fe80::b48a:4eff:fe99:2ed2/64), vethbc7ad50 (fe80::c015:d5ff:fe37:9015/64), vethcf7c32d (fe80::d82e:1dff:fe84:45ff/64), veth652c58c (fe80::2830:34ff:fee0:12ed/64), veth129eace (fe80::70e7:f1ff:fe54:30dd/64), veth6993965 (fe80::20bd:4aff:fe1a:ab15/64), veth9662484 (fe80::a87b:adff:fe29:51ad/64), vetha51542d (fe80::c41b:82ff:fec0:c567/64), vethb39b7f5 (fe80::306d:a4ff:fe1c:fbda/64), veth409cdf5 (fe80::5c94:5cff:fea0:ee2d/64), veth95e25f6 (fe80::a020:88ff:fe71:db15/64)
announce_addresses | 192.168.2.15, fe80::6d63:d225:6290:94c6

</details>

<details><summary>NextDNS</summary>

can_reach_server | ok
-- | --

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 1 maja 2025 10:18
-- | --
current_recorder_run | 10 maja 2025 12:34
estimated_db_size | 953.86 MiB
database_engine | mysql
database_version | 10.11.6

</details>

### Additional information

[supervisor_2025-05-10T11-16-33.057Z.log](https://github.com/user-attachments/files/20138921/supervisor_2025-05-10T11-16-33.057Z.log)

[host_2025-05-10T11-06-27.775Z.log](https://github.com/user-attachments/files/20138922/host_2025-05-10T11-06-27.775Z.log)

sairon, 2025-05-21:
Sometimes a complete shutdown (i.e. a cold boot) is needed when the device ends up in a weird state. Have you tried that?

bieniu, 2025-05-21:
LOL that's a bit embarrassing. I've restarted the device several times, but never turned it off completely. That helped! Thanks.

closed 2025-05-21 by bieniu