home-assistant/operating-system #4068: Home Assistant Yellow with CM5 (16GB RAM) detects only 8GB

madarnis, 2025-05-15:
### Describe the issue you are experiencing

### Hardware:
- Home Assistant Yellow
- Raspberry Pi Compute Module 5 (CM5) with 16GB RAM

### Software:
- Home Assistant OS 15.2 (latest as of May 2025)
- Supervisor and Core up to date

### Problem:
After installing CM5 with 16GB RAM, only ~8GB is detected by the system. The memory reported by `/proc/meminfo` and `dmesg` shows ~8133268 kB, i.e., ~8 GB usable.

Output from `dmesg | grep -i memory`:
[ 0.000000] Memory: 8062484K/8384512K available

Output from `/proc/meminfo`:
MemTotal: 8133268 kB

There are no kernel panics or errors in the boot log indicating hardware or allocation problems. It seems the DTB or kernel currently does not fully support 16GB RAM on CM5.

### Steps tried:
- Updated OS to latest version (`ha os update`)
- Checked `dmesg` and memory usage
- Verified CM5 hardware is working properly
- CM5 16GB is physically present and known to work

### Request:
Could support for full 16GB RAM on CM5 be added or confirmed? If already planned, happy to test beta or dev builds. Let me know if additional logs or dumps are helpful.

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 15.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Home Assistant Yellow, CM5116064

### Steps to reproduce the issue

1. Prepare a brand new Home Assistant Yellow device.
2. Install a Raspberry Pi Compute Module 5 (CM5) with 16GB RAM.
3. Flash Home Assistant OS 15.2 (latest version as of May 2025).
4. Restore a previous Home Assistant installation from backup (originally run on a laptop).
5. Complete setup and boot Home Assistant normally.
6. Access system via SSH or terminal.
7. Run the following command to check memory:
   cat /proc/meminfo
8. Result shows only ~8133268 kB of RAM (about 8 GB) detected.
9. Optionally, check kernel memory log:
   dmesg | grep -i memory
10. Note that no errors appear, but the system only utilizes ~8 GB RAM instead of 16 GB.

### Anything in the Supervisor logs that might be useful for us?

```txt
No relevant errors or warnings in the Supervisor logs related to memory or hardware detection.
The system runs normally — the only issue is that it detects only ~8 GB of RAM on a CM5 with 16 GB.
```

### Anything in the Host logs that might be useful for us?

```txt
No relevant entries in the Host logs related to RAM or hardware detection.
System starts normally without errors or warnings. Memory is limited to ~8 GB, but no issues are logged regarding hardware or kernel failures.
```

ThUnD3r-Gr33n, 2025-05-16:
Absolute same issue here. Ready to test and to surf on the bleeding edge :) I did buy the 16GB module to get the max out of my Home assistant yellow :)

abjugard, 2025-05-18:
This has actually been fixed already (#3989), but there hasn't been a release of Home Assistant OS in ages so currently there's nothing we can do to get access to it but hope that the maintainers make a new build soon.

philolilo, 2025-05-18:
Hello. I have got the same issue. 
Home Assistant Yellow with CM5116064 (which means 16Go of RAM, and 64Go of eMMC embedded storage). No NWMe. No flash memory card connected.
Start from a new home assistant setup (no backup restored).

Are you sure the issue is linked with missing support of CM5 with  16Go RAM ?

closed 2025-05-21 by sairon