home-assistant/operating-system #4080: [HAOS 15.2] "unable to find /run in /proc/mounts" warning on Lenovo ThinkCentre M715Q Tiny

AlexNissa, 2025-05-28:
### Describe the issue you are experiencing

Depuis une mise à jour récente de Home Assistant OS, mon système affiche un avertissement persistant au démarrage concernant l'impossibilité de trouver `/run` dans `/proc/mounts`. Cet avertissement suggère que le conteneur est susceptible de planter ("The container is likely to crash soon"), ce qui me préoccupe quant à la stabilité à long terme de mon installation.

L'avertissement exact est :
`/package/admin/s6-overlay/libexec/preinit: warning: unable to find /run in /proc/mounts, check that your container manager pre-mounts /proc, and that /run is a tmpfs. The container is likely to crash soon, if /run is (incorrectly) mounted noexec.`

Cet avertissement apparaît dans les journaux système au début du démarrage de l'OS.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

15.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

* Modèle : `Lenovo ThinkCentre M715Q Tiny`
* Architecture : `x86_64`

### Steps to reproduce the issue

1.1. Démarrer/redémarrer Home Assistant OS 15.2 sur ma machine.
2. Accéder aux journaux de l'OS (par exemple, via `ha su logs` ou l'interface UI si le log est disponible).
3. L'avertissement ci-dessus est visible.

**Comportement attendu :**
L'avertissement `/package/admin/s6-overlay/libexec/preinit: warning: unable to find /run in /proc/mounts` ne devrait pas apparaître, indiquant un montage correct de `/run`.

**Comportement actuel :**
L'avertissement apparaît à chaque démarrage, signalant une potentielle instabilité.

2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
/package/admin/s6-overlay/libexec/preinit: info: read-only root
/package/admin/s6-overlay/libexec/preinit: info: writable /run. Checking for executability.
/package/admin/s6-overlay/libexec/preinit: warning: unable to find /run in /proc/mounts, check that your container manager pre-mounts /proc, and that /run is a tmpfs. The container is likely to crash soon, if /run is (incorrectly) mounted noexec.
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/udev.sh INFO: Using udev information from host
cont-init: info: /etc/cont-init.d/udev.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun supervisor (no readiness notification)
services-up: info: copying legacy longrun watchdog (no readiness notification) INFO: Starting local supervisor watchdog...
s6-rc: info: service legacy-services successfully started
2025-05-28 09:37:19.554 INFO (MainThread) [main] Initializing Supervisor setup
2025-05-28 09:37:19.607 INFO (MainThread) [supervisor.coresys] Setting up coresys for machine: generic-x86-64
2025-05-28 11:37:19.613 INFO (MainThread) [supervisor.docker.supervisor] Attaching to Supervisor ghcr.io/home-assistant/amd64-hassio-supervisor with version 2025.05.3
2025-05-28 11:37:19.626 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state initialize
2025-05-28 11:37:19.627 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2025-05-28 11:37:19.627 INFO (MainThread) [main] Setting up Supervisor
2025-05-28 11:37:19.717 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
2025-05-28 11:37:19.776 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
2025-05-28 11:37:19.777 INFO (MainThread) [supervisor.dbus.manager] Connected to system D-Bus.
2025-05-28 11:37:19.777 INFO (MainThread) [supervisor.dbus.agent] Load dbus interface io.hass.os
[... 369 lines cut ...]
adapters | lo (disabled), enp3s0f3u3 (enabled, default, auto), hassio (disabled), docker0 (disabled), vethfccc2ff (disabled), vethb702369 (disabled), veth5c958f8 (disabled), vethd3d7852 (disabled), veth6d1fbfe (disabled), vethe054fb7 (disabled), veth95aa823 (disabled), veth9e19774 (disabled), veth7b07cf9 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp3s0f3u3 (192.168.1.143/24), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), vethfccc2ff (), vethb702369 (), veth5c958f8 (), vethd3d7852 (), veth6d1fbfe (), vethe054fb7 (), veth95aa823 (), veth9e19774 (), veth7b07cf9 ()
ipv6_addresses | lo (::1/128), enp3s0f3u3 (2a01:e0a:3cd:7870:ead4:9f3d:8bde:7442/64, fe80::8fd7:f1e8:8ce5:eaa4/64), hassio (fe80::4835:6bff:fefa:6a6b/64), docker0 (fe80::a43f:25ff:fe77:a2ba/64), vethfccc2ff (fe80::8047:28ff:feb9:e2d4/64), vethb702369 (fe80::384f:4eff:fefc:60c9/64), veth5c958f8 (fe80::f49f:93ff:fea8:5d21/64), vethd3d7852 (fe80::802:4ff:fef3:7e6c/64), veth6d1fbfe (fe80::cc79:7bff:fe0a:6185/64), vethe054fb7 (fe80::c0c2:17ff:fe5a:f6aa/64), veth95aa823 (fe80::8894:2bff:fe4c:97c1/64), veth9e19774 (fe80::2c20:d4ff:fe91:35ea/64), veth7b07cf9 (fe80::304b:afff:fe4d:dd93/64)
announce_addresses | 192.168.1.143, 2a01:e0a:3cd:7870:ead4:9f3d:8bde:7442, fe80::8fd7:f1e8:8ce5:eaa4

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 12 mai 2025 à 15:17
-- | --
current_recorder_run | 28 mai 2025 à 11:38
estimated_db_size | 515.72 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

JCalvi, 2025-06-04:
I have the same warning running HA Supervised (Not HAOS) on a RPi5.

scottdbrown, 2025-06-08:
Same problem:
```
/package/admin/s6-overlay/libexec/preinit: info: read-only root
/package/admin/s6-overlay/libexec/preinit: info: writable /run. Checking for executability.
/package/admin/s6-overlay/libexec/preinit: warning: unable to find /run in /proc/mounts, check that your container manager pre-mounts /proc, and that /run is a tmpfs. The container is likely to crash soon, if /run is (incorrectly) mounted noexec.
```

Running HAOS as a VM on Proxmox VE 7.4, Beelink N5105 mini PC. Crash seems to happen 1-2 times a day, seemingly at random. No errors I can see in Proxmox.

j1philli, 2025-06-11:
I restored a backup from an Unraid VM to a CM5 HA Yellow POE and am now seeing the issue. I also added a couple matter over thread devices as well as restoring the backup.

sairon, 2025-06-12:
The log message is a Supervisor issue, and while it has no effect on the stability, it should be gone in newer Supervisor releases after this change: https://github.com/home-assistant/supervisor/pull/5921 - currently on the beta channel only but I believe it should be pushed to beta soon.

If you see any stability issues, they are unrelated to this error message and more logs would be needed, eventually submitted in another issue report.

Also, please file issues in English as it's the primary language used here - even if you need to use translator, it's easier to find related issues in the search rather if the report is written in another language.

closed 2025-06-12 by sairon