home-assistant/operating-system #4087: Can't connect to Wifi ("Failed to change network settings")

zaneclaes, 2025-05-29:
### Describe the issue you are experiencing

Long time user of HA, attempting to install for the first time via HASS (Raspberry Pi imager installation).

After initially configuring via an Ethernet connection, I cannot finish setup (i.e., change over to Wifi).

The error is:
> Failed to change network settings
> Object does not exist at path '/org/freedesktop/NetworkManager/ActiveConnections/6'

The connection number at the end changes (I have tried 2 installs on 2 RPis connecting to 3 different WiFi networks).

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

core-2025.5.3

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Raspberry Pi 4

### Steps to reproduce the issue

1. Install the OS via RPi imager, per https://www.home-assistant.io/installation/raspberrypi
2. Boot up and perform initial setup via ethernet
3. Open HA network settings and attempt to connect to Wifi

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System information

## System Information

version | core-2025.5.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.3
os_name | Linux
os_version | 6.6.74-haos-raspi
arch | aarch64
timezone | America/Denver
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.05.3
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 57.8 GB
disk_used | 5.2 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | 

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 0
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), end0 (enabled, default, auto), docker0 (disabled), hassio (disabled), veth15e87c0 (disabled), veth20c3c8f (disabled), veth824a508 (disabled), veth023595f (disabled), vethe830705 (disabled), vethacca551 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), end0 (192.168.50.14/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), veth15e87c0 (), veth20c3c8f (), veth824a508 (), veth023595f (), vethe830705 (), vethacca551 ()
ipv6_addresses | lo (::1/128), end0 (fe80::ef42:863f:88d9:be19/64), docker0 (fe80::2c37:b6ff:fe74:188e/64), hassio (fe80::90d8:20ff:fe41:1b5d/64), veth15e87c0 (fe80::4c03:b8ff:fe9f:ad9a/64), veth20c3c8f (fe80::404a:c7ff:fe96:3214/64), veth824a508 (fe80::8c60:ecff:fe6b:3db/64), veth023595f (fe80::e44f:c3ff:fe5f:247/64), vethe830705 (fe80::3421:8dff:fe43:cf31/64), vethacca551 (fe80::ccbf:76ff:fe79:3795/64)
announce_addresses | 192.168.50.14, fe80::ef42:863f:88d9:be19

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | May 29, 2025 at 10:09 PM
-- | --
current_recorder_run | May 29, 2025 at 4:09 PM
estimated_db_size | 0.19 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

sairon, 2025-06-17:
Can you check the [Host logs](https://my.home-assistant.io/redirect/logs/?provider=host) for any errors and share it?

sairon, 2025-06-18:
These are Core logs, to find out what's happening with network, we need Host logs. The link I posted should lead you there directly, or just select them in the top left dropdown on the Logs page.

closed 2025-09-25 by github-actions[bot]