home-assistant/operating-system #4112: HAOS randomly shutting down in ProxmoxVE

yousaf465, 2025-06-22:
### Describe the issue you are experiencing

I am running HAOS 2025.6.1 with Operating System 16.0.rc1, it randomly shuts down without any warning 
[image]
[image]

Host logs between shutdown and manual start after 4 hours 

```
025-06-21 19:10:53.887 homeassistant systemd[1]: Started Time & Date Service.
2025-06-21 19:11:23.659 homeassistant systemd[1]: systemd-hostnamed.service: Deactivated successfully.
2025-06-21 19:11:23.935 homeassistant systemd[1]: systemd-timedated.service: Deactivated successfully.
2025-06-22 03:21:43.050 homeassistant kernel: Linux version 6.12.33-haos (builder@280a9cb85da3) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -g30ef1047) 13.3.0, GNU ld (GNU Binutils) 2.43.1) #1 SMP PREEMPT_DYNAMIC Mon Jun 16 09:52:35 UTC 2025

``` 
both times are wrong. 

and relevant core logs

```
2025-06-22 01:55:25.131 ERROR (MainThread) [custom_components.frigate.api] Timeout error fetching information from htxxxxxxxxxxxxxxxxxxxxx/api/stats: 
2025-06-22 01:55:26.167 WARNING (SyncWorker_6) [custom_components.worlds_air_quality_index.sensor] aqi value from json waqi api was undefined ('-' value)
2025-06-22 01:55:27.757 ERROR (SyncWorker_10) [homeassistant.components.influxdb] Cannot connect to InfluxDB due to 'HTTPConnectionPool(host='a0d7b954-influxdb', port=8086): Max retries exceeded with url: /write?db=homeassistant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f586addb5c0>: Failed to establish a new connection: [Errno -5] Name has no usable address'))'. Please check that the provided connection details (host, port, etc.) are correct and that your InfluxDB server is running and accessible. Retrying in 60 seconds.
2025-06-22 01:55:31.284 ERROR (MainThread) [homeassistant.helpers.event] Error while dispatching event for sensor.huawei_igd_uptime to <Job track state_changed event {'sensor.huawei_igd_uptime'} HassJobType.Callback <bound method TrackTemplateResultInfo._refresh of <TrackTemplateResultInfo {Template<template=({% set uptime = states.sensor.huawei_igd_uptime.state | int %} {% set days = (uptime / 86400) | int %} {%- if days > 0 -%}
  {{ days }} days, {{ (uptime - (days * 86400)) | int | timestamp_custom('%H:%M:%S', false) }}
{%- else -%}
  {{ uptime | int | timestamp_custom('%H:%M:%S', false) }}
{%- endif -%}) renders=934>: <RenderInfo Template<template=({% set uptime = states.sensor.huawei_igd_uptime.state | int %} {% set days = (uptime / 86400) | int %} {%- if days > 0 -%}
  {{ days }} days, {{ (uptime - (days * 86400)) | int | timestamp_custom('%H:%M:%S', false) }}
{%- else -%}
  {{ uptime | int | timestamp_custom('%H:%M:%S', false) }}
{%- endif -%}) renders=934> all_states=False all_states_lifecycle=False domains=frozenset() domains_lifecycle=frozenset() entities=frozenset({'sensor.huawei_igd_uptime'}) rate_limit=None has_time=False exception=None is_static=False>}>>>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 661, in state
    numerical_value = int(value)
ValueError: invalid literal for int() with base 10: '20 days, 19:20:30'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 664, in state
    numerical_value = float(value)
ValueError: could not convert string to float: '20 days, 19:20:30'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 355, in _async_dispatch_entity_id_event
    hass.async_run_hass_job(job, event)
    ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 938, in async_run_hass_job
    hassjob.target(*args)
    ~~~~~~~~~~~~~~^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 1318, in _refresh
    self.hass.async_run_hass_job(self._job, event, updates)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 938, in async_run_hass_job
    hassjob.target(*args)
    ~~~~~~~~~~~~~~^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/template/template_entity.py", line 477, in _handle_results
    self.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1019, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1141, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1081, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1025, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 668, in state
    raise ValueError(
    ...<5 lines>...
    ) from err
ValueError: Sensor sensor.huawei_modem_uptime has device class 'duration', state class 'None' unit 'None' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: '20 days, 19:20:30' (<class 'str'>)
2025-06-22 01:55:40.134 ERROR (MainThread) [custom_components.frigate.api] Timeout error fetching information from httxxxxxxxxxxxxx/api/stats: 
2025-06-22 01:55:51.309 INFO (MainThread) [habluetooth.wrappers] xxxxxxxxxx: Found 1 connection path(s), preferred order: hci0 (xxxxxxx) (RSSI=-80) (failures=6) (in_progress=0) (score=-80.0)
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun home-assistant (no readiness notification)
s6-rc: info: service legacy-services successfully started
2025-06-22 08:23:40.273 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration adaptive_lighting which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

```

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

16.0.rc1

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Beelink EQ12 16GB ram 500GB SSD

### Steps to reproduce the issue

1.Update toOperating System 16.0.rc1,
2. HAOS shutdown randomly.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
couldn't find any
```

### Anything in the Host logs that might be useful for us?

```txt
025-06-21 19:10:53.887 homeassistant systemd[1]: Started Time & Date Service.
2025-06-21 19:11:23.659 homeassistant systemd[1]: systemd-hostnamed.service: Deactivated successfully.
2025-06-21 19:11:23.935 homeassistant systemd[1]: systemd-timedated.service: Deactivated successfully.
2025-06-22 03:21:43.050 homeassistant kernel: Linux version 6.12.33-haos (builder@280a9cb85da3) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -g30ef1047) 13.3.0, GNU ld (GNU Binutils) 2.43.1) #1 SMP PREEMPT_DYNAMIC Mon Jun 16 09:52:35 UTC 2025
```

jjavin, 2025-06-25:
I have a similar situation with Proxmox VE and Home Assistant OS 15.2.

sam2b, 2025-06-27:
Same behavior here also since May 2025, Docker running on Debian Linux, AMD CPU laptop.  It started to happen in May, and possibly in April. It is happening more frequently now in June.  What could trigger a total OS shutdown? i.e. the bare metal computer is physically powered off.

itzalgd24, 2025-07-10:
@Impact123 , yessir already checked that. Discard is set and trim every 7 days. Since I did that, there have been no Home Assistant crashes. Keeping fingers crossed.

closed 2025-10-15 by github-actions[bot]