home-assistant/operating-system #4119: HA OS fails to boot on GEEKOM XT1 Mega

RAPHAEL2903, 2025-06-23:
### Describe the issue you are experiencing

After installing everything according to the description [here](https://www.home-assistant.io/installation/generic-x86-64), I tried booting HAOS, but i didn't get past this screen.

[image]

I tried waiting for 2-3 hours but nothing seemed to change.

I also removed the NVME SSD out of the NUC and into an USB enclosure, and then tried to boot it on my PC, and it booted just fine.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

15.2

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

GEEKOM XT1 Mega :
CPU: Intel Ultra U5-125H
RAM: 32 GB DDR5
Memory: 1 TB M.2 NVME SSD Kingston

### Steps to reproduce the issue

1. Installing HAOS on an GEEKOM XT1 Mega NUC
2. Trying to boot

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing
```

### Anything in the Host logs that might be useful for us?

```txt
Booting ‘Slot A (OK=1 TRY=X) X beeing the number of times i tried after installing the HAOS.

System  Will freeze
```

RAPHAEL2903, 2025-06-27:
gonna send the NUC back.

Will try another. I hope will work with other brand.

vehoelite, 2025-09-14:
HaOS isn't compatible with Intel Core Ultra Processors.

closed 2025-09-14 by RAPHAEL2903