home-assistant/operating-system #4127: Error updating Home Assistant Operating System:

CarlosCobos93, 2025-06-27:
### Describe the issue you are experiencing

When I try to update HA OS I receive the following message:

No se pudo realizar la acción update/install. Error updating Home Assistant Operating System: Can't fetch OTA update from https://os-artifacts.home-assistant.io/15.2/haos_green-15.2.raucb: Cannot connect to host github.com:443 ssl:default [Timeout while contacting DNS servers]

What can I do?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

15.1

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

I cannot update OS

### Steps to reproduce the issue

1.Try to update OS

### Anything in the Supervisor logs that might be useful for us?

```txt
Failed to parse frame, ignoring 
 20:25:01 – (ADVERTENCIA) runner.py - el mensaje se produjo por primera vez a las 20:18:21 y aparece 197 veces
 Failed to parse frame gpepIncomingMessageHandler: b'7fcc260037ba04e237ba04e20003000000063ea70175568bb31bff188df1c89f578c9d989fe54c42ba8b65a75a59dbc801dda9b4' 
 20:25:01 – (ADVERTENCIA) /usr/local/lib/python3.13/site-packages/bellows/ezsp/protocol.py - el mensaje se produjo por primera vez a las 20:18:21 y aparece 197 veces
 Error doing job: Exception in callback _SelectorDatagramTransport._read_ready() (None) 
 20:25:01 – (ERROR) Eufy Robovac (integración personalizada) - el mensaje se produjo por primera vez a las 20:17:30 y aparece 181 veces
 [281472860475840] Error during service call to update.install: Error updating Home Assistant Operating System: Can't fetch OTA update from https://os-artifacts.home-assistant.io/15.2/haos_green-15.2.raucb: Cannot connect to host github.com:443 ssl:default [Timeout while contacting DNS servers] 
 20:24:08 – (ERROR) Home Assistant WebSocket API - el mensaje se produjo por primera vez a las 20:17:31 y aparece 5 veces
```

### Anything in the Host logs that might be useful for us?

```txt
6f4fce08b255a050fb68d5cc9bf601f964787d015334fde75598ced49ee44924-merged.mount: Deactivated successfully.
2025-06-27 18:11:40.034 homeassistant systemd[1]: mnt-data-docker-overlay2-6f4fce08b255a050fb68d5cc9bf601f964787d015334fde75598ced49ee44924-merged.mount: Deactivated successfully.
2025-06-27 18:11:40.841 homeassistant systemd[1]: var-lib-docker-overlay2-e52eb8f47a33550116d2bf2da9ee5e0f1f47e10e24f92e402693fc60b5ec734f\x2dinit-merged.mount: Deactivated successfully.
2025-06-27 18:11:40.841 homeassistant systemd[1]: mnt-data-docker-overlay2-e52eb8f47a33550116d2bf2da9ee5e0f1f47e10e24f92e402693fc60b5ec734f\x2dinit-merged.mount: Deactivated successfully.
2025-06-27 18:11:41.104 homeassistant systemd[1]: Started libcontainer container 13aace05d8a88b4e9716d7a13db1aa9a33d958d798eb45a68e2864b51198be03.
2025-06-27 18:16:50.376 homeassistant systemd[1]: docker-13aace05d8a88b4e9716d7a13db1aa9a33d958d798eb45a68e2864b51198be03.scope: Deactivated successfully.
2025-06-27 18:16:50.377 homeassistant systemd[1]: docker-13aace05d8a88b4e9716d7a13db1aa9a33d958d798eb45a68e2864b51198be03.scope: Consumed 2min 5.343s CPU time.
2025-06-27 18:16:50.423 homeassistant dockerd[539]: time="2025-06-27T18:16:50.423025462Z" level=info msg="ignoring event" container=13aace05d8a88b4e9716d7a13db1aa9a33d958d798eb45a68e2864b51198be03 module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
2025-06-27 18:16:50.664 homeassistant systemd[1]: Started libcontainer container 13aace05d8a88b4e9716d7a13db1aa9a33d958d798eb45a68e2864b51198be03.
2025-06-27 18:25:45.318 homeassistant systemd-timesyncd[517]: Network configuration changed, trying to establish connection.
2025-06-27 18:25:45.321 homeassistant systemd-timesyncd[517]: Contacted time server 162.159.200.123:123 (time.cloudflare.com).
2025-06-27 18:25:45.354 homeassistant systemd[1]: Started Journal Gateway Service.
```

CarlosCobos93, 2025-07-12:
any help?

cnecrea, 2025-08-17:
> any help?

Carlos, it’s the DNS. Connect through your mobile (hotspot) and try. Maybe also with a VPN. Another option is to set public DNS (for example 1.1.1.1 or 8.8.8.8). It’s due to your internet provider’s blocks to GitHub.

cnecrea, 2025-08-19:
> Hi,
> 
> I think this is affecting me in more things, any other workaround besides using a hostpot or VPN? to avoid my provider blocking.
> 
> Thank you for your answer!

It’s very likely a **DNS resolution issue** on your current network (your ISP or local network might be blocking or filtering GitHub domains).  
Besides using a hotspot or VPN, you can try these steps:  

1. **Change DNS servers directly in Home Assistant OS**  
   - Go to **Settings → System → Network** (or **Settings → System → Host → IPv4**) and set DNS servers to:  
     ```
     1.1.1.1
     8.8.8.8
     ```
   - Save and **Restart Host**.

2. **If you use Pi-hole or AdGuard**  
   - Ensure they are not blocking `github.com`, `raw.githubusercontent.com`, or `objects.githubusercontent.com`.  
   - Check the upstream DNS servers they use and switch to something like Cloudflare/Google/Quad9.

3. **Check DNS from the CLI (Terminal & SSH add-on)**  
   ```bash
   ha dns info
   ha dns logs
   nslookup github.com
   ```
   If resolution fails, it’s definitely a DNS/network issue.

4. **Force HA DNS to use external resolvers**  
   ```bash
   ha dns options --servers dns://1.1.1.1 --servers dns://8.8.8.8
   ha dns restart
   ```

5. **Disable problematic IPv6 settings** (if applicable) and retry.

Once DNS is fixed, OTA updates and any GitHub-dependent services should work without a hotspot or VPN.

snothub, 2025-10-31:
I'm having this exact same issue. My hassio is setup with a static IP and my own AdGuard DNS running on my router and using Googles DNS upstream. I know I'm not blocking github and can see the IP responding from "nslookup" in the terminal right on HAOS. But the OS won't update and returns this exact error as OP described. I've also tried changing DNS to cloudflare and others and going around my AdGuard, but no cigar.

Raspberry Pi 4
Installation method Home Assistant OS
Core 2025.10.4
Supervisor 2025.10.0
Operating System 11.5
Frontend 20251001.4