home-assistant/operating-system #4143: [WARN] Home Assistant CLI not starting! Jump into emergency console

just-doit, 2025-07-06:
### Describe the issue you are experiencing

After rebooting HA the following message appears on the console:
```txt
Waiting for the Home Assistant CLI to be ready...
[WARN] Home Assistant CLI not starting! Jump into emergency console...
```

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

6.6.74-haos-raspi

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Raspberry Pi 5

### Steps to reproduce the issue

1. Shutdown or reboot gracefully the Home Assistant virtual machine
2. Power it up
3. Get to the console of the Home Assistant virtual machine
4. In the console see the message

### Anything in the Supervisor logs that might be useful for us?

```txt
None
```

### Anything in the Host logs that might be useful for us?

```txt
None
```

sairon, 2025-07-21:
Unless you are seeing any other issues, this message is just a warning and may be harmless. I think we should improve the wording a bit.

just-doit, 2025-07-21:
My problem is that HomeAssistant is not accessible. At least not via the web interface.
I then have to connect a monitor and keyboard to the Raspberry Pi to see if there is a message on the console. And that potentially with every update.

penguin-fr, 2025-08-05:
Same issue today :
- shutting down nicely the VM
- shut down the computer
- power up the computer
- start the VM
"Waiting for the Home Assistant CLI to be ready...
[WARN] Home Assistant CLI not starting! Jump into emergency console..."

I'm using the Oracle VirtualBox on a steroid computer.
Latest version of Home Assistant is up to date.

I saw same issue on several post on google research