home-assistant/operating-system #4156: HA OS 16

chrisgorgo, 2025-07-10:
### Describe the issue you are experiencing

I cannot connect to my HA box after update (getting ERR_CONNECTION_RESET). Surprisingly some automations still work.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

16.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Chrome Box bare metal installation

### Steps to reproduce the issue

1. Upgrade to 16
2. Try to access http://homeassistant.local:8123/

### Anything in the Supervisor logs that might be useful for us?

```txt
Cannot access.
```

### Anything in the Host logs that might be useful for us?

```txt
n/a
```

chrisgorgo, 2025-07-10:
The box has wired connection and static IP.

domectrl, 2025-07-10:
Seems like a lot of people with trouble when upgrading to 16. I have an issue as well, might be related. After the upgrade, my RPi5 does not use the dt_overlays that I had specified in /mnt/boot/config.txt. When I check the config with vi (using ssh to port 22222) the config is still there, but the kernel module that should be loaded (pwm_bcm2835) is no longer loaded with version 16. When I force loading with modprobe, the module is loaded, but still the devices are not created as in v. 15.2. I don't know, but this might be related to all issues people experience...

vlad36N, 2025-07-10:
Same here. RPi4 has wired connection and static IP.

Krojack, 2025-07-10:
I had to restore a backup of my VM.

Mine starts up and runs for about 15 minutes before the web interface just stops working.  I check the console on the VM and it responds fine... my HA log is flooded with the following...

```
OSError: [Errno 24] No file descriptors available
2025-07-10 18:25:40.172 ERROR (MainThread) [homeassistant] Error doing job: socket.accept() out of system resource (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 178, in _accept_connection
    conn, addr = sock.accept()
                 ~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/socket.py", line 295, in accept
    fd, addr = self._accept()
               ~~~~~~~~~~~~^^
```
The log file grew to 1.3 gigs as well.

Some more entries from the log that I haven't seen before this. I've censored out my personal domain name in some spots.

```
2025-07-10 17:46:15.398 WARNING (SyncWorker_6) [urllib3.connectionpool] Connection pool is full, discarding connection: pve1.**********. Connection pool size: 10
```

```
2025-07-10 18:24:07.990 ERROR (MainThread) [homeassistant] Error doing job: socket.accept() out of system resource (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 178, in _accept_connection
  File "/usr/local/lib/python3.13/socket.py", line 295, in accept
OSError: [Errno 24] No file descriptors available
2025-07-10 18:24:07.991 ERROR (MainThread) [homeassistant] Error doing job: socket.accept() out of system resource (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 178, in _accept_connection
  File "/usr/local/lib/python3.13/socket.py", line 295, in accept
OSError: [Errno 24] No file descriptors available
2025-07-10 18:24:07.991 ERROR (MainThread) [homeassistant] Error doing job: socket.accept() out of system resource (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 178, in _accept_connection
  File "/usr/local/lib/python3.13/socket.py", line 295, in accept
```

chrisgorgo, 2025-07-11:
I managed to fix this in my system. I logged in using remote access (via internet) and then enabled wifi interface in system settings.

FreshYop, 2025-07-11:
Same error here

MichalKalita, 2025-07-11:
I can connect to HO from a notebook (I have a static address in DHCP settings in the router),
but HO cannot connect to any device in the network. I need to set up a static address in network settings, and all integrations start working.

OS 16.0, device HO/Green

FreshYop, 2025-07-11:
> A temporary working solultion: Set static IP of eth0 in console. Example:
> 
> > ```
> > ha network update eth0 --ipv4-address 192.168.1.5/24 --ipv4-gateway 192.168.1.1 --ipv4-method static --ipv4-nameserver 192.168.1.1
> > ```
> 
> Then check if it's connected:
> 
> > ```
> > ha network info
> > ```

Even with that I still have lot of bugs 🤔

myhrmans, 2025-07-11:
Same issue here on a fresh install.

FreshYop, 2025-07-11:
How can i go back to 15.2 when 16.0 is already install ?

Krojack, 2025-07-11:
> How can i go back to 15.2 when 16.0 is already install ?

You can try the following if you can get to the command prompt. Never done it myself as I just restore a backup of my entire virtual machine.

`ha os update --version 15.2`

I've seen talk about it here: https://www.reddit.com/r/homeassistant/comments/1lvwuox/failed_upgrade_os_160/

FreshYop, 2025-07-11:
Hum it seems to be risky
I'll wait and hope a new version fixing this bug

Azel4231, 2025-07-12:
Great advice, thanks!
Turns out I had another device that used the PI5 IP. It probably took over when the PI5 restarted after the update. Now I know why you should manage static IPs in one place (DHCP Server).

ptrtech1, 2025-07-14:
Thousands of these error messages:
`2025-07-13 17:31:11.258 ERROR (MainThread) [homeassistant] Error doing job: socket.accept() out of system resource (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 178, in _accept_connection
  File "/usr/local/lib/python3.13/socket.py", line 295, in accept
OSError: [Errno 24] No file descriptors available
2025-07-13 17:31:11.259 ERROR (MainThread) [homeassistant] Error doing job: socket.accept() out of system resource (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 178, in _accept_connection
  File "/usr/local/lib/python3.13/socket.py", line 295, in accept
OSError: [Errno 24] No file descriptors available
2025-07-13 17:31:11.259 ERROR (MainThread) [homeassistant] Error doing job: socket.accept() out of system resource (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 178, in _accept_connection
  File "/usr/local/lib/python3.13/socket.py", line 295, in accept
OSError: [Errno 24] No file descriptors available
2025-07-13 17:31:11.259 ERROR (MainThread) [homeassistant] Error doing job: socket.accept() out of system resource (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 178, in _accept_connection
  File "/usr/local/lib/python3.13/socket.py", line 295, in accept
OSError: [Errno 24] No file descriptors available
2025-07-13 17:31:11.260 ERROR (MainThread) [homeassistant] Error doing job: socket.accept() out of system resource (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 178, in _accept_connection
  File "/usr/local/lib/python3.13/socket.py", line 295, in accept
OSError: [Errno 24] No file descriptors available`

agners, 2025-07-14:
> I managed to fix this in my system. I logged in using remote access (via internet) and then enabled wifi interface in system settings.

@chrisgorgo so you've enabled WiFi and that fixed the issue? Since you initially said that the box has wired connection, with that your system has two connections to the network no?

What type of hardware are you using exactly?

ptrtech1, 2025-07-14:
> @chrisgorgo so you've enabled WiFi and that fixed the issue? Since you initially said that the box has wired connection, with that your system has two connections to the network no?
>
> What type of [...]

FYI, as a point of reference, I'm dual homed (bridged) between two subnets with Ethernet on both (no WiFi at all).

myhrmans, 2025-07-14:
> @chrisgorgo so you've enabled WiFi and that fixed the issue? Since you initially said that the box has wired connection, with that your system has two connections to the network no?
>
> What type of [...]

@agners 
Hit me up on discord if you wanna live debug, I'm in the home Assistant Discord. 

@myhrmans is my tag

shalak, 2025-07-14:
Same issue here. I have 2 NICs in my HassOS, "main" and "iot" - looks like the DHCP for "main" one did not assign the IP address (it did assign DNS and gateway though).

FreshYop, 2025-07-15:
If anyone want to go back to previous version : 
ha os boot-slot other

I've try it and now it seem to be ok. You can test with : 
ha os info 
to see if you still have the 15.2 version.

sairon, 2025-07-15:
Please check logs from the previous boots - once you roll back to the older version or get the system running with a reboot, check [Core](https://my.home-assistant.io/redirect/logs/?provider=core) and [Host](https://my.home-assistant.io/redirect/logs/?provider=host) and using the top right three dots menu, select one of the previous startups where you lost the access.

andyb2000, 2025-07-25:
This is the same thing I'm getting, after a troubled few attempts I've identified what may be the cause
r8169 kernel driver for the ethernet adapter is having problems.
Before the upgrade network connectivity was fine, after the upgrade to os 16.0 (kernel 6.12.35-haos build Wed July 9 14:22:07 UTC 2025

I can't easily copy+paste the output but here is a screenshot.
I've used multiple different network cables and network switchports to confirm it it's a local/cabling issue, so this suggests it's the kernel driver for the r8169 chipset after the kernel upgrade.

[image]

fabiomatavelli, 2025-08-14:
The only way that I could make the upgrade was by doing a backup, installing the v16 from scratch and then restoring the backup

nivek1612, 2025-08-15:
> The only way that I could make the upgrade was by doing a backup, installing the v16 from scratch and then restoring the backup

yeh I'll wait for it to be fixed fully

nivek1612, 2025-09-09:
@sairon, sorry, been away a while. I assume 16.2 hasn't got any changes for this issue, as it's still using the 6.12 kernel. If not, I'll try to pull the host log

mclardass, 2025-09-10:
> EDIT: Tried 16.2. Same issue, HA does not get an IPV4 address. ha os boot-slot other works but of course, i'm back to 15.2

Thanks for the update, I was finally able to get 16.1 installed but same issue on reboot. Your not saves me on struggling with 16.2 and I guess I'll just wait for a resolution (or not) at this point.

sairon, 2025-09-10:
@nivek1612 Right, no changes really, as I don't have the hardware available, I can not investigate the problem myself further.

I tried looking for upstream reports but it doesn't seem to be a known problem at this point, I guess not many people run this board with latest Linux, yet the stable kernel regression is still the most probable cause. Ideally we'll need to identify at least the version which introduced the breakage. We have dev builds for most of the kernel patch releases, so it'd be possible to iterate over dev versions and identify where it became broken, yet it might be a bit tedious.