home-assistant/operating-system #4158: HomeAsssitantOS Network speed (wifi & wired) on Raspberri pi 5 capped at 11MB/s (100Mbit)

joepio, 2025-07-11:
### Describe the issue you are experiencing

The raspberry pi 5 should get gigabit speeds, but I haven't achieved any speed over 100Mbit on my pi5 with HomeAsssistant OS. I tried wired and wireless, but no changes. Other devices on same network cable or wireless connection get 1Gbit/s.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

16.0

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Raspberry Pi 5
Installed OS on 4TB USB HDD

### Steps to reproduce the issue

1. Install HAOS on RBpi5
2. Do any speed check
3. see 100Mbit cap

### Anything in the Supervisor logs that might be useful for us?

```txt
Version	core-2025.7.1
Installation type	Home Assistant OS
Development	false
Supervisor	true
Docker	true
Container architecture	aarch64
User	root
Virtual environment	false
Python version	3.13.3
Operating system family	Linux
Operating system version	6.12.25-haos-raspi
CPU architecture	aarch64
Timezone	Europe/Amsterdam
Configuration directory	/config
Home Assistant Community Store
GitHub API	ok
GitHub Content	ok
GitHub Web	ok
HACS Data	ok
GitHub API Calls Remaining	5000
Installed Version	2.0.5
Stage	startup
Available Repositories	2101
Downloaded Repositories	10
Home Assistant Cloud
Logged In	true
Subscription expiration	January 1, 2018 at 01:00
Relayer connected	false
Relayer region	
Remote enabled	true
Remote connected	false
Alexa enabled	true
Google enabled	true
cloud_ice_servers_enabled	true
Remote server	
Certificate status	
Instance ID	24a080a21b0844a19df3545a11949e8e
Reach certificate server	ok
Reach authentication server	ok
Reach Home Assistant Cloud	ok
Home Assistant Supervisor
Host operating system	Home Assistant OS 16.0
Update channel	stable
Supervisor version	supervisor-2025.07.1
Agent version	1.7.2
Docker version	28.3.0
Disk total	3667.5 GB
Disk used	1109.0 GB
Healthy	true
Supported	true
host_connectivity	true
supervisor_connectivity	true
ntp_synchronized	true
virtualization	
Board	rpi5-64
Supervisor API	ok
Version API	ok
Installed add-ons	Plex Media Server (3.6.5), Terminal & SSH (9.18.0), Matter Server (8.0.0), Mosquitto broker (6.5.1), Frigate (Full Access) (0.15.1), qBittorrent (5.1.2), Jackett NAS (0.22.2117), Samba share (12.5.2), Studio Code Server (5.19.3), Cloudflared (5.3.5), Home Assistant Google Drive Backup (0.112.1), Uptime Kuma (0.14.0), Zigbee2MQTT (2.5.1-1), BattyBirdNET-Pi (2025.06.06), Jellyfin NAS (10.10.7-2)
Dashboards
Dashboards	4
Resources	2
Views	1
Mode	storage
Network Configuration
Adapters	lo (disabled), end0 (enabled, default, auto), docker0 (disabled), hassio (disabled), veth542391b (disabled), veth9119c0b (disabled), veth15e20cc (disabled), veth5fdfe09 (disabled), veth8b80d61 (disabled), vethbfdfaaf (disabled), vethcf4fac6 (disabled), veth97b6cd5 (disabled), veth40551b8 (disabled), veth4f730d2 (disabled)
IPv4 addresses	lo (127.0.0.1/8), end0 (192.168.0.131/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), veth542391b (), veth9119c0b (), veth15e20cc (), veth5fdfe09 (), veth8b80d61 (), vethbfdfaaf (), vethcf4fac6 (), veth97b6cd5 (), veth40551b8 (), veth4f730d2 ()
IPv6 addresses	lo (::1/128), end0 (fd27:4ec7:d9d8:9142:4a50:ceb0:b178:583a/64, fe80::6d2a:9f0e:9d26:1086/64), docker0 (fe80::dc61:a6ff:feb6:a70a/64), hassio (fe80::491:96ff:fe6a:99d4/64), veth542391b (fe80::40c0:ff:fef2:f04/64), veth9119c0b (fe80::2426:5dff:fe31:e63e/64), veth15e20cc (fe80::470:e3ff:fec1:24e8/64), veth5fdfe09 (fe80::ec3c:ccff:fe17:d64b/64), veth8b80d61 (fe80::80be:eff:fe84:f8e0/64), vethbfdfaaf (fe80::281c:43ff:fe64:2a02/64), vethcf4fac6 (fe80::f8a5:b2ff:fe04:3a38/64), veth97b6cd5 (fe80::b8dd:feff:fe88:7221/64), veth40551b8 (fe80::fca9:29ff:fee9:43ac/64), veth4f730d2 (fe80::3c90:26ff:fe7a:afd8/64)
Announce addresses	192.168.0.131, fd27:4ec7:d9d8:9142:4a50:ceb0:b178:583a, fe80::6d2a:9f0e:9d26:1086
Recorder
Oldest run start time	June 25, 2025 at 10:20
Current run start time	July 11, 2025 at 00:49
Estimated database size (MiB)	290.96 MiB
Database engine	sqlite
Database version	3.48.0
Spotify
Spotify API endpoint reachable	ok
Core metrics
Processor usage
0.2 %
Memory usage
10.7 %
Supervisor metrics
Processor usage
0.1 %
Memory usage
1.9 %
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

joepio, 2025-07-11:
Might be related to #3699

dacorde, 2025-11-15:
Home Assistant OS 16.3 has NOT resolved the issue, so I had to reverte to 15.2 again.
With 16.3 download speed is again closer to 600Mbs instead of 900 with 15.2, besides CPU looks a bit busier. 

[image]
[image]

joepio, 2025-11-15:
Update, issue was in fact not caused by homeassistant or my raspberry pi... It was a gigabit switch which had an issue. Sorry!

dacorde, 2025-11-15:
Probably we are mentioning 2 different issues, but I used this to avoid opening a new one.
In my case, with Rpi 4B, it recognizes gigabit speed (1000), but ethernet performance is much lower with new driver from 16.x.