home-assistant/operating-system #4160: HAOS 16.0 fails to boot on RPi 5

analysederix, 2025-07-12:
### Describe the issue you are experiencing

Hello everyone

I acquired a brand new Raspberry Pi 5 8GB I wanted to use with Home Assistant 16.0.

I tried multiple times with the RaspberryPi Imager 1.9.4, as well as BalenaEtcher and Rufus with haos_rpi5-64-16.0.img.xz (file checksum ok), on an SD card and multiple USB drives.

The Pi5 is stuck in the boot menu. Here is the result with an SD card:

[image]

And with a USB drive:

[image]

I then tried with haos_rpi5-64-15.2.img.xz and it worked.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

16.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Raspberry Pi 5 with 8Gb or RAM

### Steps to reproduce the issue

1. Flash HAOS 16.0 on removable media
2. Boot the Pi5 with the removable media
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### System information

No

### Additional information

No. Thank you for your support.

Trippy76, 2025-07-12:
I’m having exactly the same problem

Little-Elephant, 2025-07-13:
+1. exactly the same

analysederix, 2025-07-14:
> > I’m having exactly the same problem
> 
> I gaveup trying to us 16.0, i proved there was nothing wrong with the card or PI5 by installing RaspberryPI OS on the card, everything worked perfectly in that scenario. Downloaded HAOS 15.2 on to the same SD card and then everything worked. Have left my system at 15.2 until this is sorted.

I ended up installing 15.2 and then immediately upgraded to 16.0 and it worked well this way.
I still added the issue for the developers to know.

However if you use Alexa devices you might want to keep 15.2 and wait for the next release as there are a lot of reports here about errors with Alexa integration on 16.0 (https://github.com/home-assistant/core/issues)

sairon, 2025-07-15:
I am unable to reproduce the issue here even after downgrading to the same EEPROM bootloader version. However, my board revision is d04170:

[image]

I currently don't have board with different revision available for experiments but it might be worth trying if bootloader update resolves the issue for you. Since OS 16.0, you can use `rpi-eeprom-update -a` command (when connected directly with keyboard and display, or through the developer SSH at port 22222) to update to the latest EEPROM version we have verified to be working correctly. I wonder if the fresh install works correctly after then.

Trippy76, 2025-07-15:
> I am unable to reproduce the issue here even after downgrading to the same EEPROM bootloader version. However, my board revision is d04170:
> 
> <img alt="Image" width="412" height="102" src="https://private-user-images.githubusercontent.com/211416/466497525-7014ae1d-ca26-4971-b866-7620550020a0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTI2MDE2NzQsIm5iZiI6MTc1MjYwMTM3NCwicGF0aCI6Ii8yMTE0MTYvNDY2NDk3NTI1LTcwMTRhZTFkLWNhMjYtNDk3MS1iODY2LTc2MjA1NTAwMjBhMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNzE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDcxNVQxNzQyNTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xN2UyNjRmZmZkZmFlM2MxYTY2MWMxN2E4ZTY5MWM1ZThmZWVjY2ExYzNiNzZmMzU2MmYxZDk4OTkyNzE2YjRmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.JlEGaBT8wU0iMy92Xo5NY2o1jBgMXVZ2IHwyu3RWkEc">
> I currently don't have board with different revision available for experiments but it might be worth trying if bootloader update resolves the issue for you. Since OS 16.0, you can use `rpi-eeprom-update -a` command (when connected directly with keyboard and display, or through the developer SSH at port 22222) to update to the latest EEPROM version we have verified to be working correctly. I wonder if the fresh install works correctly after then.

I have the latest boot loader and am running a d04171 rev board

[image]

Baxxy13, 2025-07-15:
Hmm, mine is booting without problems.
[image]

HA-OS 16.0 was flashed with Raspberry Pi Imager 1.9.4 under Win10 PC to 32GB PNY sd-card.

TetraGG, 2025-07-17:
> <img alt="Image" width="2000" height="1153" src="https://private-user-images.githubusercontent.com/19843186/467255422-5eac801d-2ffe-4cbf-ab2a-c334c1648d11.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTI3NTE0NTAsIm5iZiI6MTc1Mjc1MTE1MCwicGF0aCI6Ii8xOTg0MzE4Ni80NjcyNTU0MjItNWVhYzgwMWQtMmZmZS00Y2JmLWFiMmEtYzMzNGMxNjQ4ZDExLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzE3VDExMTkxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMwODdkMTFkMmFiZjU0NzIyOGY1ZWFlNGExZjNmYmZiMTIzNzA0ZWM1ZjYxNWU0MGU4NDk5MGYwOGQ2NTY1ZTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.us9DuDhbCksSv2xN2o9ctjHNIXhjYJ9f-8eGl6lXKAs"> <img alt="Image" width="2000" height="1153" src="https://private-user-images.githubusercontent.com/19843186/467255420-62af4b2d-4e1d-4dbb-8edd-9415241c0a64.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTI3NTE0NTAsIm5iZiI6MTc1Mjc1MTE1MCwicGF0aCI6Ii8xOTg0MzE4Ni80NjcyNTU0MjAtNjJhZjRiMmQtNGUxZC00ZGJiLThlZGQtOTQxNTI0MWMwYTY0LlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzE3VDExMTkxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgxZGIwN2U2YTA3YTQxZGQyZDY0YzVkZTA4YjE0MWU1MzAxNmNmZmNjYWFmMzRjM2MyY2Y2NTUyNDBhZDFhMjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.olEIZuxs9y1T8vyIy5IEP9U7o-GaHzNukjPSYb4LAvA">
> +1 with a fresh install (OS + hardware)

To complete, I was using the entry "Other specific-purpose OS > Home Assistant" from Raspberry Pi Imager v1.9.4. I tried to update bootloader to the latest version but the issue is still the same.
But, I used the "Use custom" entry with the OS image downloaded from GitHub release, and it resolved the boot issue.

TechArea, 2025-07-17:
> downloaded from GitHub release, and it resolved the boot issue.

why aren't they identical?

Luwingo, 2025-07-20:
I am having the same exact problem. I bought a brand new Raspberry Pi 5 and the HA OS 16.0 will not boot.

Luwingo, 2025-07-20:
> To complete, I was using the entry "Other specific-purpose OS > Home Assistant" from Raspberry Pi Imager v1.9.4. I tried to update bootloader to the latest version but the issue is still the same. [...]

I just did this and it worked for me as well. Thank you!

zjorz, 2025-07-26:
Bought a RPi v5
Works/boots when using RPi lite x64, RPi desktop x64
Fails/does not boot when using HA OS 16

trex2000, 2025-07-29:
Same happens also with HA Yellow with CM5, no internal EMMC.

sairon, 2025-07-31:
> However, if the DOWNLOAD HA OS 16 from github for RPI5 (haos_rpi5-64-16.0.img.xz) and feed it as a custom image to RPI imager, after the installation, IT DOES BOOT!

This is strange, RPi Imager uses the [very same image](https://github.com/home-assistant/version/blob/046066dc8a6ab1b55cfb7889f9ed48a007c80aef/rpi-imager-haos.json#L6) that you downloaded from GH releases. Unless there's some other factor in play, it shouldn't make difference. But you're at least third to report the same. Except with @analysederix, who tried multiple imaging tools with the downloaded image, so that doesn't fit here.

Anyway, I tried a couple of SD cards I have around and none of them fails, also tried using different utilities and OS's. If you have the issue, can you share as many details as possible? E.g. what tool and OS you have used for flashing the image, what exact SD card model, etc.

closed 2025-08-04 by sairon

alangham20, 2025-08-08:
I got it. I installed a downloaded copy of 16 with the pi installer. It fixed the issue of the boot loader for me

[image]

PR #4199 (2025-08-04): Update genimage to avoid unbootable RPi 5 images when flashed from Windows
Update genimage so the images are not mangled (by the primary GPT relocated) when flashed from Windows. Otherwise, boot media flashed from Windows isn't compatible with bootloader older than 2024-10-10.

This is a regression of #3437. The Buildroot update in #4027 updated genimage to v18, yet the downstream patch that was later replaced by a different one in upstream was not merged to that version yet and the patch was incorrectly removed. In v19 there's another fix in the offset calculation logic that sets the first us [... cut]
commit c39ceed3 (2025-08-04): Update genimage to avoid unbootable RPi 5 images when flashed from Windows (#4199)

Update genimage so the images are not mangled (by the primary GPT relocated)
when flashed from Windows. Otherwise, boot media flashed from Windows isn't
compatible with bootloader older than 2024-10-10.

This is a regression of #3437. The Buildroot update in #4027 updated genimage
to v18, yet the downstream patch that was later replaced by a different one in
upstream was not merged to that version yet and the patch was incorrectly
removed. In v19 there's another fix in the offset calculation logic that sets
the [... cut]