home-assistant/operating-system #4162: HA crashing on Pi 4 after HAOS 16.0 upgrade

chris-y, 2025-07-13:
### Describe the issue you are experiencing

After upgrading to HAOS 16.0 on a Raspberry Pi 4, Home Assistant has been very unstable, I've been unable to connect after a few hours of it running. It looks from some of my graphs that it has been failing before midnight quite consistently. 

I've reverted back to 15.3 and the problem has gone away.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

16.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Raspberry Pi 4 with install on SD card. 

### Steps to reproduce the issue

1. Upgrade
2. Do nothing
3. Inaccessible system
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Couldn't see anything relevant
```

### Anything in the Host logs that might be useful for us?

```txt
Couldn't see anything relevant
```

### System information

This information is from 15.3 after I downgraded:

## System Information

version | core-2025.7.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
container_arch | aarch64
user | root
virtualenv | false
python_version | 3.13.3
os_name | Linux
os_version | 6.6.74-haos-raspi
arch | aarch64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | startup
Available Repositories | 2102
Downloaded Repositories | 28

</details>

<details><summary>Solcast Solar</summary>

can_reach_server | ok
-- | --

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.07.1
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 28.5 GB
disk_used | 17.2 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (21.0.2), File editor (5.8.0), Mosquitto broker (6.5.1), Predbat (1.2.9), Tailscale (0.25.0), Terminal & SSH (9.18.0), predai (1.3.2), motionEye (0.22.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 6
-- | --
resources | 16
views | 12
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), wlan0 (enabled, default, auto), hassio (disabled), docker0 (disabled), tailscale0 (disabled), vethffee12a (disabled), veth21154b6 (disabled), veth8f06da5 (disabled), vetha9a20aa (disabled), vetha4a5244 (disabled), vethde12886 (disabled), vethe1b26ac (disabled), vethe9e09af (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), wlan0 (192.168.0.11/24), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), tailscale0 (100.112.227.14/32), vethffee12a (), veth21154b6 (), veth8f06da5 (), vetha9a20aa (), vetha4a5244 (), vethde12886 (), vethe1b26ac (), vethe9e09af ()
ipv6_addresses | lo (::1/128), wlan0 (fe80::a1e9:fe16:8b44:ce79/64), hassio (fe80::701a:bbff:fe35:5541/64), docker0 (fe80::5083:60ff:feb5:862f/64), tailscale0 (fd7a:115c:a1e0::b701:e30e/128, fe80::a282:40f8:b82f:57b9/64), vethffee12a (fe80::9c21:9bff:feef:6ecd/64), veth21154b6 (fe80::442:daff:fe81:f528/64), veth8f06da5 (fe80::d886:46ff:fee3:9711/64), vetha9a20aa (fe80::84d6:81ff:fe18:e9ec/64), vetha4a5244 (fe80::d0b1:d6ff:fe6c:44da/64), vethde12886 (fe80::a011:95ff:fe4e:b36f/64), vethe1b26ac (fe80::64a9:6eff:fe72:1ea9/64), vethe9e09af (fe80::d8d3:15ff:fe9f:b411/64)
announce_addresses | 192.168.0.11, fe80::a1e9:fe16:8b44:ce79

</details>

<details><summary>NextDNS</summary>

can_reach_server | ok
-- | --

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 2 July 2025 at 23:22
-- | --
current_recorder_run | 13 July 2025 at 09:42
estimated_db_size | 1768.02 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

sairon, 2025-07-15:
Please check logs from the previous boots - once you roll back to the older version or get the system running with a reboot, check [Core](https://my.home-assistant.io/redirect/logs/?provider=core) and [Host](https://my.home-assistant.io/redirect/logs/?provider=host) and using the top right three dots menu, select one of the previous startups where you lost the access.

markusg1234, 2025-08-17:
Had similar issue on two systems at different locations.

Found that the Home Assistant URL for local network was on automatic and assigning an address that wasn't in my network.  

I ended up changing it to the correct local network address via the CLI command prompt using sed but I was unsure how to turn auto off.

I needed to be very quick when the system came online after restarting, I used the GUI and turned off auto. 
Funny thing is if you turn off auto then it turns it on again. You need to quickly turn it off twice and then quickly press save.

After all of this one system was then fine but the other the router needed rebooting also.

you can use this to check what it is currently set to via the root command prompt.

root

cat /mnt/data/supervisor/homeassistant/.storage/core.config

Also you maybe able to turned off auto and set the IP address when you are on 15.3 and then upgrade again. But unsure.

chris-y, 2025-09-10:
My running system is displaying lots of logs like this, which I've not seen before:

````
2025-09-10 14:18:56.262 homeassistant kernel: ieee80211 phy0: brcmf_p2p_send_action_frame: Unknown Frame: category 0xa, action 0x8
2025-09-10 14:23:57.031 homeassistant wpa_supplicant[530]: RRM: Ignoring radio measurement request: Not RRM network
2025-09-10 14:23:58.098 homeassistant wpa_supplicant[530]: wlan0: WNM: Preferred List Available
2025-09-10 14:24:00.638 homeassistant kernel: ieee80211 phy0: brcmf_p2p_send_action_frame: Unknown Frame: category 0xa, action 0x8
2025-09-10 14:28:18.880 homeassistant systemd[1]: Starting Cleanup of Temporary Directories...
2025-09-10 14:28:19.111 homeassistant systemd[1]: systemd-tmpfiles-clean.service: Deactivated successfully.
2025-09-10 14:28:19.112 homeassistant systemd[1]: Finished Cleanup of Temporary Directories.
2025-09-10 14:28:19.128 homeassistant systemd[1]: run-credentials-systemd\x2dtmpfiles\x2dclean.service.mount: Deactivated successfully.
2025-09-10 14:29:01.818 homeassistant wpa_supplicant[530]: RRM: Ignoring radio measurement request: Not RRM network
2025-09-10 14:29:02.793 homeassistant wpa_supplicant[530]: wlan0: WNM: Preferred List Available
2025-09-10 14:29:05.166 homeassistant kernel: ieee80211 phy0: brcmf_p2p_send_action_frame: Unknown Frame: category 0xa, action 0x8
2025-09-10 14:34:06.914 homeassistant wpa_supplicant[530]: RRM: Ignoring radio measurement request: Not RRM network
2025-09-10 14:34:07.839 homeassistant wpa_supplicant[530]: wlan0: WNM: Preferred List Available
2025-09-10 14:34:09.004 homeassistant wpa_supplicant[530]: RRM: Ignoring radio measurement request: Not RRM network
2025-09-10 14:34:09.854 homeassistant kernel: ieee80211 phy0: brcmf_p2p_send_action_frame: Unknown Frame: category 0xa, action 0x8
2025-09-10 14:34:35.398 homeassistant kernel: CIFS: VFS: close cached dir rc -11
2025-09-10 14:39:11.838 homeassistant wpa_supplicant[530]: RRM: Ignoring radio measurement request: Not RRM network
2025-09-10 14:39:12.873 homeassistant wpa_supplicant[530]: wlan0: WNM: Preferred List Available
2025-09-10 14:39:13.888 homeassistant wpa_supplicant[530]: RRM: Ignoring radio measurement request: Not RRM network
2025-09-10 14:39:14.906 homeassistant kernel: ieee80211 phy0: brcmf_p2p_send_action_frame: Unknown Frame: category 0xa, action 0x8
2025-09-10 14:44:16.860 homeassistant wpa_supplicant[530]: RRM: Ignoring radio measurement request: Not RRM network
2025-09-10 14:44:17.906 homeassistant wpa_supplicant[530]: wlan0: WNM: Preferred List Available
2025-09-10 14:44:18.075 homeassistant wpa_supplicant[530]: RRM: Ignoring radio measurement request: Not RRM network
2025-09-10 14:44:19.597 homeassistant wpa_supplicant[530]: RRM: Ignoring radio measurement request: Not RRM network
2025-09-10 14:44:19.942 homeassistant kernel: ieee80211 phy0: brcmf_p2p_send_action_frame: Unknown Frame: category 0xa, action 0x8
````