home-assistant/operating-system #4164: Home assistant reboot autamatically every hour after upgrading to 16.0

purewooder, 2025-07-14:
### Describe the issue you are experiencing

After upgrading HAOS to 16.0, Home Assistant Core reboots approximately every hour, as shown in the picture below. (My setup is HAOS installed as a VM in Proxmox. Last night I turned off the machine completely, so there is a gap there.)
[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

16.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

HAOS is installed as a VM in Proxmox.
The Proxmox version is 8.4.1. It is installed in an Intel(R) N100 machine.

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2025-07-14 10:18:50.837 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2025-07-14 10:40:37.059 WARNING (MainThread) [supervisor.homeassistant.websocket] Connection is closed
2025-07-14 10:40:37.062 INFO (MainThread) [supervisor.api.proxy] [a0d7b954_nodered] Home Assistant WebSocket API closed
2025-07-14 10:40:37.064 INFO (MainThread) [supervisor.api.proxy] [d5369777_music_assistant] Home Assistant WebSocket API closed
2025-07-14 10:40:40.296 WARNING (MainThread) [supervisor.homeassistant.core] Watchdog found Home Assistant failed, restarting...
2025-07-14 10:40:40.312 INFO (SyncWorker_4) [supervisor.docker.manager] Starting homeassistant
```

### Anything in the Host logs that might be useful for us?

```txt
2025-07-14 10:40:08.910 ERROR (MainThread) [custom_components.proxmoxve] Error fetching proxmox_coordinator_lxc_200 data: HTTPSConnectionPool(host='10.0.91.5', port=8006): Max retries exceeded with url: /api2/json/cluster/resources (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f696e18c550>: Failed to establish a new connection: [Errno 24] No file descriptors available'))
2025-07-14 10:40:08.930 WARNING (MainThread) [homeassistant.components.broadlink.updater] Disconnected from 阁楼遥控 (RM4 pro at 192.168.57.22)
2025-07-14 10:40:08.931 ERROR (MainThread) [homeassistant.components.broadlink.updater] Error fetching 阁楼遥控 (RM4 pro at 192.168.57.22) data: [Errno 24] No file descriptors available
2025-07-14 10:40:08.938 WARNING (MainThread) [homeassistant.components.broadlink.updater] Disconnected from 音箱插座 (SP mini 3 at 192.168.57.102)
2025-07-14 10:40:08.938 ERROR (MainThread) [homeassistant.components.broadlink.updater] Error fetching 音箱插座 (SP mini 3 at 192.168.57.102) data: [Errno 24] No file descriptors available
2025-07-14 10:40:08.962 WARNING (MainThread) [homeassistant.components.broadlink.updater] Disconnected from 客厅遥控 (RM4 mini at 192.168.57.20)
2025-07-14 10:40:08.963 ERROR (MainThread) [homeassistant.components.broadlink.updater] Error fetching 客厅遥控 (RM4 mini at 192.168.57.20) data: [Errno 24] No file descriptors available
2025-07-14 10:40:08.972 ERROR (MainThread) [homeassistant.components.systemmonitor.coordinator] Unexpected error fetching System Monitor update coordinator data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 382, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/systemmonitor/coordinator.py", line 135, in _async_update_data
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/components/systemmonitor/coordinator.py", line 180, in update_data
  File "/usr/local/lib/python3.13/site-packages/psutil/__init__.py", line 2036, in swap_memory
  File "/usr/local/lib/python3.13/site-packages/psutil/_pslinux.py", line 488, in swap_memory
  File "/usr/local/lib/python3.13/site-packages/psutil/_common.py", line 766, in open_binary
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
OSError: [Errno 24] No file descriptors available: '/proc/meminfo'
2025-07-14 10:40:09.129 WARNING (MainThread) [homeassistant.components.broadlink.updater] Disconnected from 主卧遥控 (RM mini 3 at 192.168.57.21)
2025-07-14 10:40:09.129 ERROR (MainThread) [homeassistant.components.broadlink.updater] Error fetching 主卧遥控 (RM mini 3 at 192.168.57.21) data: [Errno 24] No file descriptors available
2025-07-14 10:40:09.803 ERROR (MainThread) [custom_components.xiaomi_miot.core.device.yeelink.light.colorb] Got MiioException while fetching the state: [Errno 24] No file descriptors available, mapping: {'light.on': {'siid': 2, 'piid': 1}, 'light.brightness': {'siid': 2, 'piid': 2}, 'light.color_temperature': {'siid': 2, 'piid': 3}, 'light.color': {'siid': 2, 'piid': 4}, 'light.mode': {'siid': 2, 'piid': 5}, 'yeelight.flow_param': {'siid': 3, 'piid': 1}}, max_properties: 6/6
2025-07-14 10:40:09.809 ERROR (SyncWorker_3) [custom_components.xiaomi_miot.core.miio2miot] yeelink.light.lamp15: Got MiioException: [Errno 24] No file descriptors available while get_properties(['power', 'bright', 'ct', 'color_mode', 'bg_power', 'bg_bright', 'bg_ct', 'bg_rgb'])
2025-07-14 10:40:12.842 ERROR (MainThread) [custom_components.proxmoxve] Error fetching proxmox_coordinator_storage_storage/pve/local-lvm data: HTTPSConnectionPool(host='10.0.99.5', port=8006): Max retries exceeded with url: /api2/json/cluster/resources (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f696cfe8cd0>: Failed to establish a new connection: [Errno 24] No file descriptors available'))
2025-07-14 10:40:13.724 WARNING (MainThread) [deebot_client.command] Could not execute command getBattery
[... 83 lines cut ...]
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/json.py", line 228, in save_json
  File "/usr/src/homeassistant/homeassistant/util/file.py", line 66, in write_utf8_file
homeassistant.util.file.WriteError: [Errno 24] No file descriptors available: '/config/.storage/icloud3/tmp7qzhszao'
[02:40:37] INFO: Home Assistant Core finish process exit code 256
[02:40:37] INFO: Home Assistant Core finish process received signal 11
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun home-assistant (no readiness notification)
s6-rc: info: service legacy-services successfully started
```

purewooder, 2025-07-14:
I downgraded the operating system to 15.2. Now there is no issue anymore.

komoricodrutz, 2025-07-14:
Hi,
I am having a similar issue. HA Core restarting roughly every 45-60 minutes after upgrading to HAOS 16.
home-assistant.log.1 goes to 130-180 MiB after each such restart.
A lot of the errors there seem to be HACS related. This didn't happen in 15.2
I restarted in safe mode and it seems to have survived 70 minutes until now, no longer over-inflating the home-assistant.log.1 file, however there are many instances of the
`File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 178, in _accept_connection`
entry
Here are the logs from safe mode:

[home-assistant.log](https://github.com/user-attachments/files/21211977/home-assistant.log)

[home-assistantlog1.log](https://github.com/user-attachments/files/21211980/home-assistantlog1.log)

(HA was restarted in safe mode at 10:01, so the contents of home-assistantlog1.log are from before that restart)

And here are the previous logs involving HACS:

[home-assistant.log.man.txt.zip](https://github.com/user-attachments/files/21211828/home-assistant.log.man.txt.zip)

BTW, a way of downloading logs with anonymised sensitive data like IPs, user names, device names, etc. would be a nice thing to have.

yjamal01, 2025-07-14:
Had the same issue, tried disabling custom integrations but same issue. Downgraded to 15.2 and been stable since

agners, 2025-07-14:
@komoricodrutz as it turns out, the errors in the logs from safe mode are unrelated to HAOS 16.0, see also https://github.com/hacs/integration/issues/4701#issuecomment-3068956949.

However, the non-safe mode logs seem to have the same culprit as the original poster reported, the system is running out of file descriptors:

```
2025-07-14 08:34:34.084 ERROR (MainThread) [homeassistant] Error doing job: socket.accept() out of system resource (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 178, in _accept_connection
  File "/usr/local/lib/python3.13/socket.py", line 295, in accept
OSError: [Errno 24] No file descriptors available
```

Now since this seems not to happen in safe mode, that seems to suggest that one of the custom components open a lot of files. This could be for good reasons (as in, its just a custom component which does need to read lots of files), or it could be due to a bug (it does not close past file descriptors, and hence continously causes more and more file handles to be consumed).

FWIW, for those intrested in the technical details, I did write a bit more background why this changed in HAOS 16.0 in discussion thread https://github.com/home-assistant/operating-system/discussions/4166.

sairon, 2025-07-14:
While this is a bug caused by an OS change, we should address it on the Core side which should adjust its FD limit itself, just like all the other containers/add-ons which now get lower soft limit than before.

Also worth noting, because of https://github.com/golang/go/issues/74608 this can be seen only on x86-64 and ova, at least containers on ARM64-based platforms have soft limit that matches the new hard limit, which is actually a bug.

agners, 2025-07-15:
@purewooder does your system work when using safe mode and HAOS 16.0?

komoricodrutz, 2025-07-15:
I can confirm that the issues stopped after rolling back to HAOS 15.2

sairon, 2025-07-15:
@komoricodrutz This will be fixed on Core side (unless you have issues with some add-ons, which will need [separate handling](https://developers.home-assistant.io/blog/2025/07/14/home-assistant-os-16-open-file-limit)). Core issue is here: 
* https://github.com/home-assistant/core/issues/148806

It'll also help us if you could enlighten what eats up the resources - however, to do that, you need access to shell in the Home Assistant Core container. There are several ways to do that, easiest is the developer SSH at port 22222, alternatively the desired output could be obtained through the Shell Command integration.

closed 2025-07-31 by sairon

madbrain76, 2025-08-04:
> > Had the same issue, tried disabling custom integrations but same issue. Downgraded to 15.2 and been stable since
> 
> [@yjamal01](https://github.com/yjamal01) can you share the logs of the issue when running Home Assistant Core with disabled custom integrations?
> 
> You may also be able to just scroll up to the previous logs of previous boots your current state.

1. how can I use the beta to check if the problem is fixed ?
2. how can I roll back to HAOS 15.2 without rolling back everything else ?
3. I currently 593 devices. The vast majority are with core integrations - TP-Link Kasa, Z-Wave, Wiz. I I am wondering if a 2048 fd limit is sufficient for my setup. Why not make it configurable ?

My system is essentially unusable now. Unfortunately, it does not reboot when the limit is hit. The only workaround is to login to the Windows console and restart the VM from the VMWare GUI.

Logger: homeassistant
Source: /usr/src/homeassistant/homeassistant/runner.py:112
First occurred: 12:16:22 (250818 occurrences)
Last logged: 12:24:33
Error doing job: socket.accept() out of system resource (None)

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 178, in _accept_connection
    conn, addr = sock.accept()
                 ~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/socket.py", line 295, in accept
OSError: [Errno 24] No file descriptors available

yjamal01, 2025-08-04:
> [@yjamal01](https://github.com/yjamal01) can you share the logs of the issue when running Home Assistant Core with disabled custom integrations?
>
> You may also be able to just scroll up to the [...]

Right now im stable on 15.2 but will try and update soon and share logs if I run into issues, thanks

madbrain76, 2025-08-11:
> > how can I use the beta to check if the problem is fixed ?
> 
> You can explicitly update to a beta release, e.g. to update to the currently latest Core beta version use `ha core update --version 2025.8.0b2`

Is there a list of the beta versions available somewhere ?

Also, this is for core and not the OS. Can I update just the OS/kernel without the core also ? Or do they both have to work together ?

> It wasn't configurable in the past, so for now we try to pick a reasonable default. Device count does not mean more file handles are needed, file handles are really a implementation detail which _some_ integrations need. If you need much more than 2k, then likely this is a bug in an integration somehwere, leaking file descriptor. Even if we raise the limit to a much higher limit, your installation will run out of memory eventually. It is better to fix the underlying file descriptor leak in that case.

Thanks. After rolling back the OS to 15.2, and leaving core untouched, everything is fine.

I do not see any notable memory growth in my VM, looking at the hardware details in the settings.
The memory usage is currently at 2.1 GB out of 8 GB. This is similar to what I have seen in the past, even with HAOS 16.

There is unfortunately no visibility as to how many total file descriptors are currently in use in my system, and by which integration. I think it is extremely likely that the number exceeds the 1024 setting set in HAOS 16, though, since I did not make any other change except roll back to 15.2, and the problem was fixed, without evidence of fd leak or RAM growth.

madbrain76, 2025-08-13:
> > Is there a list of the beta versions available somewhere ?
> 
> The list of releases has the pre-releases in it as well: https://github.com/home-assistant/core/releases

Thank you !

> A file descriptor is relatively small, probably a couple of kilobytes max. So if your system opens in the range of thousands, then this is likely only a couple of megabytes in difference. But if it continuously leaks, then this will have an impact over time (weeks)....

My installation could not stay up even a single day on HAOS 16.0 . It required an ACPI restart in VMWare daily.
When I rolled back to HAOS 15.2, everything was fine again, with no VM restart needed.

> What would be good is if you can create a backup (just in case you want to roll back), and update Core to 2025.8.1. Then update HAOS to 16.0. If that still causes problems, than your system requires more than 2048 file descriptors. You can then safely roll back to HAOS 15.2 again (`ha os update --version 15.2`).

I updated to core 2025.8.1 16 hours ago. I was still running HAOS 15.2 .
I just switched to HAOS 16.0 . I'll report if the problem shows up again.

madbrain76, 2025-08-15:
> I updated to core 2025.8.1 16 hours ago. I was still running HAOS 15.2 . I just switched to HAOS 16.0 . I'll report if the problem shows up again.

So far, I have not seen the issue again . I'm on HAOS 16.1 , and the system has been up 1 day, 17:35 .

sairon, 2025-08-15:
@purewooder Everything points to the fact that some custom integration is leaking the file descriptors. In Proxmox it should be fairly easy to run commands which could give you some insights about what proces leaks it and what type of FDs are that. In Proxmox console for the VM, in HAOS CLI, type `login` and then run these commands:
```
docker exec homeassistant /bin/sh -c 'lsof | awk '"'"'{print $2 " " $1}'"'"' | sort | uniq -c | sort -n'
```
This should give you overview of opened file descriptors per process - you can check how fast the numbers grow. Most likely it will be `/usr/local/bin/python3.13` as the main culprit. It would be very interesting if it's something else.
```
docker exec homeassistant /bin/sh -c 'lsof /config/lsof.txt'
```
Will dump listing of opened files to `lsof.txt` in your config folder (i.e. where `configuration.yaml` lives). From there you can download it for further analysis - generally it shouldn't contain any sensitive so you can upload it here. It'd be more useful to do it once you get closer to the mark when a Core crash is expected.

slungs, 2025-08-20:
Hello, I have a similar issue (mentioned here: https://github.com/home-assistant/core/issues/150386) 

How can establish what is causing this? Full output of lsof can be found here:

[lsof_full.txt](https://github.com/user-attachments/files/21890062/lsof_full.txt)

slungs, 2025-08-20:
@sairon thanks for the help and quick reply!

`docker exec homeassistant lsof -Ki | wc -l` now outputs 2153. Not seeing any issues yet, but I'll post the log file here anyway. 

[lsof_full (2).txt](https://github.com/user-attachments/files/21907328/lsof_full.2.txt)

If I don't run in to more issues now, it was most likely the Xiaomi Vacuum Map integration causing issues. However I had the issue in safe mode as well before....

ArtrixTech, 2025-09-06:
Any update on this?