home-assistant/operating-system #4176: HA OS 16 always reverts back to 15.2 on restart

brianm71, 2025-07-16:
### Describe the issue you are experiencing

I've updated to HAOS 16.0 numerous times.  After the update Hassio runs fine, there are no issues.  However, as soon as I restart (to update other components), Hassio boots in Recovery Mode, requiring me to hit CTRL-D to continue, which reverts the OS back to 15.2, my last version.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

15.2

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Windows 10, Virtualbox Version 7.1.10 r169112 (Qt6.5.3)

### Steps to reproduce the issue

1. Update OS to 16.0
2. Restart 
3. CTRL-D to bypass Recovery mode
4. OS back to 15.2
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
The only thing I noticed here is that some of the records are dated one and two days ahead of today.
```

sairon, 2025-07-22:
What does `ha os info` show after the reboot? Can you provide [Host logs](https://my.home-assistant.io/redirect/logs/?provider=host) after the revert to the previous version?

brianm71, 2025-08-03:
I finally figured this out.  The first upgrade attempt failed so Boot-Slot A was "Bad" so it was failing-over to Boot-Slot B, which had v15.2.  After some research, I booted up 15.2, switched to Boot-Slot A, downgraded the OS and reinstalled it, which fixed the issue.  It took me almost two-weeks of research to even figure out there were Boot-Slots. There should be an easier way to figure this out and fix it.

closed 2025-08-03 by brianm71