home-assistant/operating-system #4182: HAOS starts up with 15.2 after updatig to 16.0

marcelhoogantink, 2025-07-21:
### Describe the issue you are experiencing

Trying to update HAOS from 15.2 to 16.0.
But after the reboot when finisched, the version si still 15.2.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

15.2

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Rpi4-64
USB-1: 250 GB SSD
USB-2: APC-USB connection
USB-3: Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001

### Steps to reproduce the issue

1. Start upgrade to 16.) from 15.2
2. wait ....
3. After update, check HAOS version, still 15.2
...
detailed actions:

supervisor time (Host-time) action

18:34 (16:34) restart HA
18:40 (16:40) copied Host-logs 15.2
18:42 (16:42) copied supervisor logs 15.2		
18:43 (16:43) start update to HAOS 16.0 (backup first)
18:49 (16:49) Connection core lost.
18:51 (16:51) connection observer lost
18:52 (16:52) connection observer up again. supervisor connected
18:53 (16:53) Observer : ALL OK
18:54 (16:54) HA Core starting
18:54 (16:54) HA started  HAOS 16.0 is still an update and current OS = 15.2

### Anything in the Supervisor logs that might be useful for us?

```txt
Supervisor Logs after HAOS 15.2 reboot: 
See 'supervisor_15_2.log' in Additional Information

Supervisor Logs after trying to update to HAOS 16.0:
See: 'supervisor_after_faile_update_to_16.0.log' in Additional Information
```

### Anything in the Host logs that might be useful for us?

```txt
HOST Logs after HAOS 15.2 reboot:
See: 'host_15_2.log' in Additional Information

HOST Logs after trying to update to HAOS 16.0:
See: 'host_after_failed_update_to_16.0.log' in Additional Information
```

### System information

## System Information

version | core-2025.7.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
container_arch | aarch64
user | root
virtualenv | false
python_version | 3.13.3
os_name | Linux
os_version | 6.6.74-haos-raspi
arch | aarch64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 2191
Downloaded Repositories | 62

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.07.1
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 229.2 GB
disk_used | 59.6 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | phpMyAdmin (0.12.0), Let's Encrypt (5.4.9), FTP (5.3.2), Advanced SSH & Web Terminal (21.0.2), Samba share (12.5.2), VLC (0.3.0), Glances (0.21.1), Mosquitto broker (6.5.1), Studio Code Server (5.19.3), Local SoX Play (1.0.2), Notifier (1.1.2), Piper (1.6.2), Whisper (2.6.0), CUPS (3.1.0), Nginx Proxy Manager (2.1.0), JupyterLab (0.15.12), ESPHome Device Builder (2025.7.2), MariaDB (2.7.2), Portainer (2.31.3), Everything Presence Zone Configurator (1.2.1), Zigbee2MQTT (2.5.1-1), DbStats (0.5.12), MQTT Explorer (browser-1.0.3), Music Assistant Server (2.5.5), Zigbee2MQTT Edge (edge), chrony (5.2.0), Network UPS Tools (0.14.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 25
-- | --
resources | 45
views | 94
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), end0 (enabled, default, auto), wlan0 (disabled), docker0 (disabled), hassio (disabled), veth407f22a (disabled), veth474374d (disabled), veth8cd3e26 (disabled), veth47ec859 (disabled), veth3ee0919 (disabled), vethfb2a8a7 (disabled), veth77b3bfe (disabled), vethed6fb75 (disabled), vethef261d7 (disabled), vethcf43f48 (disabled), vethcc8c5dd (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), end0 (192.168.2.199/24), wlan0 (192.168.2.200/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), veth407f22a (), veth474374d (), veth8cd3e26 (), veth47ec859 (), veth3ee0919 (), vethfb2a8a7 (), veth77b3bfe (), vethed6fb75 (), vethef261d7 (), vethcf43f48 (), vethcc8c5dd ()
ipv6_addresses | lo (::1/128), end0 (fe80::3898:b26b:1eb9:f21/64), wlan0 (fe80::1e94:d0b1:29ce:74f1/64), docker0 (fe80::d871:4dff:fe4a:8fc1/64), hassio (fe80::a83e:cfff:fe63:cac3/64), veth407f22a (fe80::4c62:bbff:fede:539d/64), veth474374d (fe80::d45b:4bff:fe10:28b6/64), veth8cd3e26 (fe80::a49c:19ff:fe89:d0a8/64), veth47ec859 (fe80::800:33ff:feb1:26be/64), veth3ee0919 (fe80::841f:b3ff:fe5b:2ec7/64), vethfb2a8a7 (fe80::44ac:43ff:feaf:beca/64), veth77b3bfe (fe80::7c9e:8dff:fe38:dafa/64), vethed6fb75 (fe80::7cfb:31ff:fee4:e7e3/64), vethef261d7 (fe80::60ff:2cff:fe15:4a2/64), vethcf43f48 (fe80::bc29:73ff:fe83:38a6/64), vethcc8c5dd (fe80::1845:7fff:fea4:d167/64)
announce_addresses | 192.168.2.199, fe80::3898:b26b:1eb9:f21

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 10 juli 2025 om 10:22
-- | --
current_recorder_run | 21 juli 2025 om 18:53
estimated_db_size | 2109.28 MiB
database_engine | mysql
database_version | 10.11.6

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Additional information

[supervisor_after_faile_update_to_16.0.log](https://github.com/user-attachments/files/21352085/supervisor_after_faile_update_to_16.0.log)
[host_after_failed_update_to_16.0.log](https://github.com/user-attachments/files/21352086/host_after_failed_update_to_16.0.log)
[supervisor_15_2.log](https://github.com/user-attachments/files/21352087/supervisor_15_2.log)
[host_15_2.log](https://github.com/user-attachments/files/21352088/host_15_2.log)

[ha os info] from Terminal:
```
board: rpi4-64
boot: A
boot_slots:
  A:
    state: booted
    status: good
    version: "15.2"
  B:
    state: inactive
    status: good
    version: "15.1"
data_disk: SSD-250GB-14232303200361218
update_available: true
version: "15.2"
version_latest: "16.0"
```

marcelhoogantink, 2025-07-22:
Yes, there is still a SD card in de rpi4.
The first install, a few years agoo, was only with the SD-card. Later I added the USB SSD and moved the system to the SSD with the options provided with HA.
Is it an option to remove the SD-card?

marcelhoogantink, 2025-07-22:
@sairon  Thanks for your reply, I did move the data-disk to  the USB .... so not the OS...

I did a new update try, did work (save as previous) but now (hopefully) with the correct logs:

current.log ( the log directly after update-try  and the last restart)
previous.log (the log before the current)

[previous.log](https://github.com/user-attachments/files/21371992/previous.log)
[current.log](https://github.com/user-attachments/files/21371993/current.log)

jan-gerard, 2025-07-24:
Exactly the same issue here. Home Assistant Blue / Hardkernel ODROID-N2/N2+.
Initially installed everything 2 years ago on an SD card, then used the "move data disk" function on the "config > storage" page to move the data disk to an external SSD drive (that is always connected). 
I assume that the update is installed on the medium that is not used during booting. When I use 'df' I only see mounts on the external drive, while I suppose that is not the medium used during booting. I guess it boots from the SD card, then mounts the SSD partitions. And that the kernel update to 16.0 does not install the new OS version on the SD card, but on the SSD.
I don't know how to solve it, though. It worked all fine during previous OS upgrades.

sairon, 2025-07-24:
There are errors all over the logs which indicate the SD card is corrupt:
```
2025-07-21 16:51:00.301 ha kernel: EXT4-fs error (device mmcblk0p7): ext4_validate_inode_bitmap:105: comm NetworkManager: Corrupt inode bitmap - block_group = 0, inode_bitmap = 271
2025-07-21 16:51:00.359 ha kernel: EXT4-fs error (device mmcblk0p7) in ext4_free_inode:363: Corrupt filesystem
2025-07-21 16:51:00.368 ha kernel: EXT4-fs error (device mmcblk0p7) in ext4_free_inode:363: Corrupt filesystem
2025-07-21 16:51:00.374 ha kernel: EXT4-fs error (device mmcblk0p7) in ext4_free_inode:363: Corrupt filesystem
2024-05-27 21:48:36.683 homeassistant systemd-fsck[274]: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
2024-05-27 21:48:36.945 homeassistant kernel: FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
```

Sometimes the errors can be fixed with `fsck` at the boot time, but in your case it fails:
```
2024-05-27 21:48:35.831 homeassistant systemd: Created slice Slice /system/systemd-fsck.
2024-05-27 21:48:36.621 homeassistant systemd-fsck[274]: Cannot initialize conversion from codepage 850 to ANSI_X3.4-1968: Invalid argument
2024-05-27 21:48:36.621 homeassistant systemd-fsck[274]: Cannot initialize conversion from ANSI_X3.4-1968 to codepage 850: Invalid argument
2024-05-27 21:48:36.621 homeassistant systemd-fsck[274]: Using internal CP850 conversion table
2024-05-27 21:48:36.638 homeassistant systemd-fsck[275]: e2fsck 1.47.0 (5-Feb-2023)
2024-05-27 21:48:36.647 homeassistant systemd-fsck[275]: hassos-overlay: recovering journal
2024-05-27 21:48:36.683 homeassistant systemd-fsck[274]: fsck.fat 4.2 (2021-01-31)
2024-05-27 21:48:36.683 homeassistant systemd-fsck[274]: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
2024-05-27 21:48:36.683 homeassistant systemd-fsck[274]:  Automatically removing dirty bit.
2024-05-27 21:48:36.683 homeassistant systemd-fsck[274]: *** Filesystem was changed ***
2024-05-27 21:48:36.683 homeassistant systemd-fsck[274]: Writing changes.
2024-05-27 21:48:36.683 homeassistant systemd-fsck[274]: /dev/mmcblk0p1: 363 files, 3919/16343 clusters
2024-05-27 21:48:36.694 homeassistant systemd-fsck[275]: Superblock needs_recovery flag is clear, but journal has data.
2024-05-27 21:48:36.694 homeassistant systemd-fsck[275]: Run journal anyway? yes
2024-05-27 21:48:36.700 homeassistant systemd-fsck[275]: fsck.ext4: unable to set superblock flags on hassos-overlay
2024-05-27 21:48:36.701 homeassistant systemd-fsck[275]: hassos-overlay: ********** WARNING: Filesystem still has errors **********
2024-05-27 21:48:36.702 homeassistant systemd-fsck[259]: fsck failed with exit status 12.
2024-05-27 21:48:36.945 homeassistant kernel: FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
2024-05-27 21:48:36.957 homeassistant systemd[1]: systemd-fsck@dev-disk-by\x2dlabel-hassos\x2doverlay.service: Main process exited, code=exited, status=1/FAILURE
2024-05-27 21:48:36.958 homeassistant systemd[1]: systemd-fsck@dev-disk-by\x2dlabel-hassos\x2doverlay.service: Failed with result 'exit-code'.
2024-05-27 21:48:37.226 homeassistant systemd-fsck[349]: e2fsck 1.47.0 (5-Feb-2023)
2024-05-27 21:48:37.325 homeassistant systemd-fsck[349]: hassos-data: clean, 633764/15269888 files, 16606058/61049344 blocks
```

Best thing you should do now is to make a backup (thankfully, your instance's data is on the SSD, so it remains writable and no user data should be corrupted there) and re-install the OS on a new SD card. It should be also possible to simply "adopt" the data disk after you plug it into a freshly installed HAOS, but as always, make a backup first.

sairon, 2025-07-24:
@jan-gerard Please open a new issue with the logs - as you can see, every case can be unique and without logs it's hard to tell what's going on.

closed 2025-07-24 by sairon

marcelhoogantink, 2025-07-24:
@sairon Thank you for your efforts and time. I'll install a new SD card.