home-assistant/operating-system #4183: xt_NETMAP kernel module missing

mateuszdrab, 2025-07-21:
### Describe the issue you are experiencing

I run multiple instances of HAOS, one of which is on a Raspberry Pi and others are standard VMs.
On those instances, I have the Wireguard addon with a custom `post_up` configuration to adds a NETMAP directive to map a subnet to another one due to my routing requirements.

On the Raspberry Pi, this works fine, on the Proxmox VM created by the [helper script](https://raw.githubusercontent.com/community-scripts/ProxmoxVE/main/vm/haos-vm.sh) - the NETMAP module won't load as it's not available and logs the below

```
iptables -t nat -A PREROUTING -d 192.168.67.0/24 -i wg0 -j NETMAP --to 192.168.0.0/24
Warning: Extension NETMAP revision 0 not supported, missing kernel module?
```

I've looked into it and can see that it could potentially be solved by adding `CONFIG_NETFILTER_XT_TARGET_NETMAP` to "https://github.com/home-assistant/operating-system/blob/16.0/buildroot-external/kernel/v6.12.y/hassos.config" but I am not very familiar with buildroot so I'd like to check with you guys.

I am unable to clarify why the module is available on one system but not another - could it be because they run a different kernel? 6.12.25-haos-raspi vs 6.12.35-haos

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

16.0

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Raspberry Pi 4 Model B Rev 1.5

Proxmox VM with OVA image

### Steps to reproduce the issue

Configure Wireguard with the below `post_up`:
```
post_up: >-
  iptables -A FORWARD -i %i -j ACCEPT; iptables -A FORWARD -o %i -j ACCEPT;
  iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE; iptables -t nat -A
  PREROUTING -d 192.168.67.0/24 -i %i -j NETMAP --to 192.168.0.0/24
```

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System information

## System Information

version | core-2025.7.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
container_arch | amd64
user | root
virtualenv | false
python_version | 3.13.3
os_name | Linux
os_version | 6.12.35-haos
arch | x86_64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4942
Installed Version | 2.0.5
Stage | running
Available Repositories | 2110
Downloaded Repositories | 2

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 16.0
-- | --
update_channel | stable
supervisor_version | supervisor-2025.07.1
agent_version | 1.7.2
docker_version | 28.3.0
disk_total | 30.8 GB
disk_used | 5.4 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | failed to load: unreachable
installed_addons | Frigate Proxy (1.6), Advanced SSH & Web Terminal (21.0.2), WireGuard (0.12.0), File editor (5.8.0), Studio Code Server (5.19.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 0
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp6s18 (enabled, default, auto), hassio (disabled), docker0 (disabled), vethd18febb (disabled), veth8b0ab86 (disabled), vethaab44f7 (disabled), vethad34a11 (disabled), vethd1bc874 (disabled), veth1228539 (disabled), vethfbafb4d (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp6s18 (192.168.0.11/24), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), vethd18febb (), veth8b0ab86 (), vethaab44f7 (), vethad34a11 (), vethd1bc874 (), veth1228539 (), vethfbafb4d ()
ipv6_addresses | lo (::1/128), enp6s18 (fe80::8a4f:ba5b:bd57:1b79/64), hassio (fe80::68d4:89ff:fe5f:6dfb/64), docker0 (fe80::20bc:4fff:fe8b:50c9/64), vethd18febb (fe80::94c0:29ff:fe87:cd58/64), veth8b0ab86 (fe80::a0b6:f9ff:fe0c:dc7b/64), vethaab44f7 (fe80::24f9:37ff:fe13:61fc/64), vethad34a11 (fe80::f097:3aff:fe41:64ae/64), vethd1bc874 (fe80::2c67:4eff:fe53:4884/64), veth1228539 (fe80::a4f2:15ff:fe6b:4fd1/64), vethfbafb4d (fe80::2405:eff:fe12:e4f5/64)
announce_addresses | 192.168.0.11, fe80::8a4f:ba5b:bd57:1b79

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 14 July 2025 at 16:51
-- | --
current_recorder_run | 16 July 2025 at 19:31
estimated_db_size | 6.82 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

sairon, 2025-07-22:
It's probably in the upstream config/defconfig of RPi yet not in the config used for the OVA build. It's a valid request to have the netmap module available, shouldn't be a problem to add it to the common kernel config.

mateuszdrab, 2025-07-22:
That would make sense, thanks for creating the PR for this.