home-assistant/operating-system #4190: systemd-journal-gatewayd consumes 100% CPU

danez, 2025-07-22:
### Describe the issue you are experiencing

Sometimes the `systemd-journal-gatewayd` service gets in a weird state and starts consuming 100% CPU.

This happened 3 times now and i noticed this only because the CPU temp went up.

I haven't figured out exactly when this happens, today in the morning it happened after a HomeAssistant restart, couple of minutes ago it happend also after a restart but a couple of minutes after. I also had the supervisor logs open and it returned a 500 error.

So it is not 100% reproducable yet, but I'm trying to reproduce it and keep you posted. In the logs i do not see anything.

This started happening after upgrading to HA 16. And i see there were some backports for this service, so maybe worth looking at them.

```
2667853 systemd+  20   0 3238.1m  90.2m  99.9   1.1  25:22.07 S /usr/lib/systemd/systemd-journal-gatewayd  
```

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

16

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Pi5 8GB RAM, nvme ssd

### Steps to reproduce the issue

1. Not yet...
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
unfortunately no
```

### Anything in the Host logs that might be useful for us?

```txt
unfortunately no
```

### System information

## System Information

version | core-2025.7.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
container_arch | aarch64
user | root
virtualenv | false
python_version | 3.13.3
os_name | Linux
os_version | 6.12.25-haos-raspi
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4950
Installed Version | 2.0.5
Stage | running
Available Repositories | 2113
Downloaded Repositories | 8

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 16.0
-- | --
update_channel | stable
supervisor_version | supervisor-2025.07.1
agent_version | 1.7.2
docker_version | 28.3.0
disk_total | 234.0 GB
disk_used | 61.9 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | Cloudflared (5.3.5), Mosquitto broker (6.5.1), TSUN-Proxy (0.14.1-rel), Terminal & SSH (9.18.0), AdGuard Home (5.3.0), Postgres 17 (17.4-12), Immich (1.135.3), Paperless-ngx (3.0.2), Resticprofile Backup (1.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 4
views | 4
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), end0 (enabled, default, auto), hassio (disabled), docker0 (disabled), veth914e918 (disabled), veth6a6ef79 (disabled), vethd7bb555 (disabled), veth9b46088 (disabled), veth722b975 (disabled), veth0b55564 (disabled), vethfeb10ae (disabled), veth324595a (disabled), vethc90c7d7 (disabled), vethd6981a5 (disabled), veth643a9d1 (disabled), vethb6357c1 (disabled), vethe73a719 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), end0 (192.168.69.3/24), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), veth914e918 (), veth6a6ef79 (), vethd7bb555 (), veth9b46088 (), veth722b975 (), veth0b55564 (), vethfeb10ae (), veth324595a (), vethc90c7d7 (), vethd6981a5 (), veth643a9d1 (), vethb6357c1 (), vethe73a719 ()
ipv6_addresses | lo (::1/128), end0 (fdfb:6b8:70d0:0:61f9:a06a:491c:78d5/64, 2a02:8109:abaf:8200:f5be:7d6f:bf44:eaa5/64, fe80::501a:6fbf:be3b:7304/64), hassio (fe80::40f8:3eff:fef4:715/64), docker0 (fe80::d8a7:1eff:fe10:3398/64), veth914e918 (fe80::d879:94ff:fed3:47f2/64), veth6a6ef79 (fe80::f828:b2ff:fe2f:ac98/64), vethd7bb555 (fe80::a408:4bff:fe08:5f5a/64), veth9b46088 (fe80::4e8:f3ff:fe5e:dab1/64), veth722b975 (fe80::7c4d:e2ff:fe67:e672/64), veth0b55564 (fe80::7c7e:d6ff:fec7:cca6/64), vethfeb10ae (fe80::acc4:fff:fe31:813b/64), veth324595a (fe80::60fe:d8ff:fe25:3a1b/64), vethc90c7d7 (fe80::58a8:b8ff:fe21:1988/64), vethd6981a5 (fe80::b0af:b1ff:fecc:dbb6/64), veth643a9d1 (fe80::d4bd:43ff:fe3c:8fa9/64), vethb6357c1 (fe80::6815:26ff:feac:b2e2/64), vethe73a719 (fe80::c25:ccff:fe43:4e2b/64)
announce_addresses | 192.168.69.3, fdfb:6b8:70d0:0:61f9:a06a:491c:78d5, 2a02:8109:abaf:8200:f5be:7d6f:bf44:eaa5, fe80::501a:6fbf:be3b:7304

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | July 9, 2025 at 09:59
-- | --
current_recorder_run | July 22, 2025 at 21:06
estimated_db_size | 477.15 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

Mike-Allen, 2025-07-24:
Proxmox 8.4  - HAOS in a VM with 2 CPUs & 4GB memory
Home Assistant OS 
Core 2025.7.3
Supervisor 2025.07.1
Operating System 16.0
Frontend  20250702.3
AT&T fiber internet
No problems

Another Proxmox host at a remote location. Same HAOS/VM setup just on different hardware: 100% on both CPUs.
Starlink internet with drops because of obstructions.
systemd-journal-gatewayd.service: Consumed 1d 11h 37min 17.396s CPU time, 176M memory peak.

Supervisor logs:
2025-07-23 22:33:31.915 ERROR (MainThread) [supervisor.host.logs] Could not get a list of boot IDs from systemd-journal-gatewayd
2025-07-23 22:33:31.915 INFO (MainThread) [supervisor.host.logs] Could not get /boots from systemd-journal-gatewayd, using fallback.

Regarding the above - there is a log entry under 2025.7.3 and another entry under 2025.7.2 after the rollback.

Rolled back to 2025.7.2 and everything is okay again (so far.)

I must say, being able to rollback core to a previous version is much nicer than having to restore from a backup and lose changes or logs. Great work!

sairon, 2025-07-24:
> The last lines of the log will be coming in again and again in random order.

@danez Thanks for the detailed report, this thing is strange - does it happen to you since the very beginning when you open the logs? In what interval?

> I think I'm able to reproduce this, by simply opening the supervisor logs in HA and after around 1 minute it starts happening.

Tried to reproduce it by opening it the logs on all possible instances I have available here and I'm not able to run into the same error unfortunately :(

> This started happening after upgrading to HA 16. And i see there were some backports for this service, so maybe worth looking at them.

While we can't rule it out, it's not just those, there's also major Systemd bump "hiding" in the Buildroot update.

Since you have SSH access, could you try adding `debug` to the end of the `/mnt/boot/cmdline.txt` and watch `journalctl -ft systemd-journal-gatewayd` in the console while reproducing the issue?

danez, 2025-07-24:
I added the debug flag and rebooted. 

Now it does not happen that often or easily anymore. I then got it to happen again, but no idea what I did exactly. What i roughly did after reboot:
  - looked at supervisor logs and left open, reloaded a couple of times
  - restarted home assistant 
  - looked at supervisor logs and left open, reloaded a couple of times
  - looked at host logs and left open
  - restarted my restic-backup addon and during restart looked at the logs and left open (https://github.com/danez/addon-resticprofile-backups)
  - looked at supervisor logs again

All of these happened in one single tab, no multiple tabs

<details>

<summary>Here are the debug logs, right before it happened</summary>

```ruby
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: Considering root directory '/run/log/journal'.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: Root directory /run/log/journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: Considering root directory '/var/log/journal'.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: Root directory /var/log/journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: Considering directory '/var/log/journal/f5137dda790f4003828bc65ede1e709f'.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: Directory /var/log/journal/f5137dda790f4003828bc65ede1e709f added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@032c5f7d81cb4576a18d5d7cf656cc0a-00000000001d5612-00063a9bce5d5540.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@171ae589cb3445f68a6819579d0ebae1-0000000000172da7-00063955ec39d547.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@171ae589cb3445f68a6819579d0ebae1-00000000001897f1-0006399bda87b17f.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@171ae589cb3445f68a6819579d0ebae1-0000000000183ed5-0006398f6a50894c.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@171ae589cb3445f68a6819579d0ebae1-000000000018f460-000639b01accb0aa.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@032c5f7d81cb4576a18d5d7cf656cc0a-00000000001e04b5-00063aa9e675fc47.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@171ae589cb3445f68a6819579d0ebae1-000000000016d937-0006393e74106aca.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@171ae589cb3445f68a6819579d0ebae1-0000000000195167-000639c5e5cb8490.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@171ae589cb3445f68a6819579d0ebae1-000000000019ae11-000639dada35524e.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@171ae589cb3445f68a6819579d0ebae1-00000000001a1e6f-000639e9d41a4e97.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@171ae589cb3445f68a6819579d0ebae1-000000000017e523-0006397dd032a664.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@032c5f7d81cb4576a18d5d7cf656cc0a-00000000001aabf2-000639fa80a6a124.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@032c5f7d81cb4576a18d5d7cf656cc0a-00000000001b7fb8-00063a2237e2e3ea.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@032c5f7d81cb4576a18d5d7cf656cc0a-00000000001c9cc5-00063a6cd1409320.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@032c5f7d81cb4576a18d5d7cf656cc0a-00000000001be268-00063a3b5d390b60.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@171ae589cb3445f68a6819579d0ebae1-0000000000157cf7-000638e29f7708dd.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@171ae589cb3445f68a6819579d0ebae1-000000000015d2b6-000638fa086d2662.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@171ae589cb3445f68a6819579d0ebae1-00000000001782f8-0006396cff61b83a.journal added.
Jul 24 12:25:14 homeassistant systemd-journal-gatewayd[11492]: File /var/log/journal/f5137dda790f4003828bc65ede1e709f/system@032c5f7d81cb4576a18d5d7cf656cc0a-00000000001cf8c4-00063a845f13386e.journal added.
[... cut ...]
Jul 24 12:25:34 homeassistant systemd-journal-gatewayd[11492]: Considering root directory '/var/log/journal'.
Jul 24 12:25:34 homeassistant systemd-journal-gatewayd[11492]: Considering directory '/var/log/journal/f5137dda790f4003828bc65ede1e709f'.
```

</details>

Now that it happened once, I can "savely" reproduce again, by opening supervisor logs and leaving it open for 2 minutes.

---

What i also noticed is that in the frontend after 2minutes (sometimes earlier) the blue Live label disappears. Looking at the requests I see this:

[image]

ffafe5, 2025-07-25:
I'm also having this problem.

jsa91, 2025-07-30:
+1 on this issue

kenibarwick, 2025-08-09:
same on a (Alpine Linux 3.19.1 64bit / Linux 6.12.35-haos) Intel(R) Celeron(R) N5095 @ 2.00GHz

CPU%	MEM%	VIRT	RES	PID	USER	TIME+	THR	NI	S	IOR/s	IOW/s	Command
336.5	20.2	40.9G	1.52G	9653	63112	95h 04:45	84	0	S	0	0	systemd-journal-gatewayd

PR #4197 (2025-07-31): Fix systemd-journal-gatewayd 100% CPU issue when watching logs
When follow request for logs is issued that points to/beyond the end of logs, a busy loop in systemd-journal-gatewayd can be triggered which manifests as systemd-journal-gatewayd consuming 100% CPU. Since threads are used for each request, the logs may still work but the CPU will be hogged until the restart of systemd-journal-gatewayd, Supervisor, or the whole system.

Backport the patch submitted upstream that addresses this issue.

Fixes #4190

## Summary by CodeRabbit

* **Bug Fixes**
  * Resolved an issue where following journa [... cut]