home-assistant/operating-system #4202: Add-on 'exec /init: exec format error' on Raspberry Pi 5 with Home Assistant OS 16.0

Mats-dev, 2025-08-02:
### Describe the issue you are experiencing

After upgrading to Home Assistant OS 16.0 on a Raspberry Pi 5, I am unable to start the Spotify Connect add-on. The add-on fails to start immediately with the error exec /init: exec format error repeated continuously in the logs.

Environment Details:
- Home Assistant OS version: 16.0
- Home Assistant Core version: 2025.7.4
- Home Assistant Supervisor version: 2025.07.2
- Hardware: Raspberry Pi 5
- Add-on name & version: Spotify Connect (0.15.0)

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

6.12.25-haos-raspi

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

I boot Home Assistant from an Samsung Evo 870 500GB ssd and I have a Sonoff Zigbee dongle-P attached to an extender cable, I am not using an Ethernet cable.

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2025-08-03 00:47:31.817 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/spotify/aarch64 with version 0.15.0
2025-08-03 00:47:31.823 WARNING (MainThread) [supervisor.addons.addon] Watchdog found addon Spotify Connect is failed, restarting...
2025-08-03 00:47:31.828 INFO (SyncWorker_6) [supervisor.docker.manager] Cleaning addon_a0d7b954_spotify application
2025-08-03 00:47:32.210 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/spotify/aarch64 with version 0.15.0
2025-08-03 00:47:32.214 WARNING (MainThread) [supervisor.addons.addon] Watchdog found addon Spotify Connect is failed, restarting...
```

### Anything in the Host logs that might be useful for us?

```txt
2025-08-02 22:49:10.306 homeassistant systemd[1]: docker-799a55fd24fa2ad7c89a0e812a3a9990263ff30a53639ba98377f777cd4675cf.scope: Deactivated successfully.
2025-08-02 22:49:10.360 homeassistant dockerd[661]: time="2025-08-02T22:49:10.360652025Z" level=info msg="ignoring event" container=799a55fd24fa2ad7c89a0e812a3a9990263ff30a53639ba98377f777cd4675cf module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
2025-08-02 22:49:10.657 homeassistant systemd[1]: Started libcontainer container 868fdbcaacabee7e549d712cd3dfa41c6e9a02fec846c5b77485086513b48003.
2025-08-02 22:49:10.705 homeassistant systemd[1]: docker-868fdbcaacabee7e549d712cd3dfa41c6e9a02fec846c5b77485086513b48003.scope: Deactivated successfully.
2025-08-02 22:49:10.723 homeassistant dockerd[661]: time="2025-08-02T22:49:10.723719013Z" level=info msg="ignoring event" container=868fdbcaacabee7e549d712cd3dfa41c6e9a02fec846c5b77485086513b48003 module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
2025-08-02 22:49:11.116 homeassistant systemd[1]: var-lib-docker-overlay2-019d8554a83406d8b3e05843361f5679879b84ccdf151ae7d921f1583e1ab08d-merged.mount: Deactivated successfully.
2025-08-02 22:49:11.116 homeassistant systemd[1]: mnt-data-docker-overlay2-019d8554a83406d8b3e05843361f5679879b84ccdf151ae7d921f1583e1ab08d-merged.mount: Deactivated successfully.
2025-08-02 22:49:20.928 homeassistant systemd[1]: var-lib-docker-overlay2-d9dc202c82ee39b9b9d7c2f10f731987cdc1e336b5f6c3a0a39488df75863ee5\x2dinit-merged.mount: Deactivated successfully.
2025-08-02 22:49:20.928 homeassistant systemd[1]: mnt-data-docker-overlay2-d9dc202c82ee39b9b9d7c2f10f731987cdc1e336b5f6c3a0a39488df75863ee5\x2dinit-merged.mount: Deactivated successfully.
2025-08-02 22:49:21.078 homeassistant systemd[1]: Started libcontainer container 85adbebbfb1928ba58d8228f496eec7a466505d7e7bb626884ee393ae346e82a.
2025-08-02 22:49:21.141 homeassistant systemd[1]: docker-85adbebbfb1928ba58d8228f496eec7a466505d7e7bb626884ee393ae346e82a.scope: Deactivated successfully.
2025-08-02 22:49:21.161 homeassistant dockerd[661]: time="2025-08-02T22:49:21.161646108Z" level=info msg="ignoring event" container=85adbebbfb1928ba58d8228f496eec7a466505d7e7bb626884ee393ae346e82a module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
2025-08-02 22:49:21.912 homeassistant systemd[1]: var-lib-docker-overlay2-d9dc202c82ee39b9b9d7c2f10f731987cdc1e336b5f6c3a0a39488df75863ee5-merged.mount: Deactivated successfully.
2025-08-02 22:49:21.912 homeassistant systemd[1]: mnt-data-docker-overlay2-d9dc202c82ee39b9b9d7c2f10f731987cdc1e336b5f6c3a0a39488df75863ee5-merged.mount: Deactivated successfully.
2025-08-02 22:49:31.370 homeassistant systemd[1]: var-lib-docker-overlay2-63eb8886cf9f77bd8c3cb177d3f87da817b26c56e9b7e987b84f1498b555fbeb\x2dinit-merged.mount: Deactivated successfully.
2025-08-02 22:49:31.371 homeassistant systemd[1]: mnt-data-docker-overlay2-63eb8886cf9f77bd8c3cb177d3f87da817b26c56e9b7e987b84f1498b555fbeb\x2dinit-merged.mount: Deactivated successfully.
2025-08-02 22:49:31.509 homeassistant systemd[1]: Started libcontainer container dd4cf4efb293b52fb49c63a875b3919a0ce6747a817a271e1d36c27d8dc52d51.
2025-08-02 22:49:31.556 homeassistant systemd[1]: docker-dd4cf4efb293b52fb49c63a875b3919a0ce6747a817a271e1d36c27d8dc52d51.scope: Deactivated successfully.
2025-08-02 22:49:31.576 homeassistant dockerd[661]: time="2025-08-02T22:49:31.576621414Z" level=info msg="ignoring event" container=dd4cf4efb293b52fb49c63a875b3919a0ce6747a817a271e1d36c27d8dc52d51 module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
2025-08-02 22:49:31.877 homeassistant systemd[1]: Started libcontainer container 8f331ef703a0c3b84394f323ee33b8d8ee3991201750e91baa1bd1609bbd1e04.
2025-08-02 22:49:31.923 homeassistant systemd[1]: docker-8f331ef703a0c3b84394f323ee33b8d8ee3991201750e91baa1bd1609bbd1e04.scope: Deactivated successfully.
2025-08-02 22:49:31.943 homeassistant dockerd[661]: time="2025-08-02T22:49:31.943258226Z" level=info msg="ignoring event" container=8f331ef703a0c3b84394f323ee33b8d8ee3991201750e91baa1bd1609bbd1e04 module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
2025-08-02 22:49:32.355 homeassistant systemd[1]: var-lib-docker-overlay2-63eb8886cf9f77bd8c3cb177d3f87da817b26c56e9b7e987b84f1498b555fbeb-merged.mount: Deactivated successfully.
[... 74 lines cut ...]
2025-08-02 22:56:49.657 homeassistant systemd[1]: Started libcontainer container b627a54d98f2f8ba81f7f73f834a9f922e315eb8d535c244f7eb52e180468d08.
2025-08-02 22:56:49.703 homeassistant systemd[1]: docker-b627a54d98f2f8ba81f7f73f834a9f922e315eb8d535c244f7eb52e180468d08.scope: Deactivated successfully.
2025-08-02 22:56:49.721 homeassistant dockerd[661]: time="2025-08-02T22:56:49.721344973Z" level=info msg="ignoring event" container=b627a54d98f2f8ba81f7f73f834a9f922e315eb8d535c244f7eb52e180468d08 module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
```

### System information

<h2 id="title" class="mdc-dialog__title" style="-webkit-font-smoothing: antialiased; font-family: Roboto, sans-serif; font-size: 22.036px; line-height: 28px; font-weight: 400; letter-spacing: 0.27545px; text-decoration: none solid rgb(225, 225, 225); text-transform: none; display: block; position: relative; flex-shrink: 0; box-sizing: border-box; margin: 0px 0px 1px; padding: 12px 12px 0px; color: rgb(225, 225, 225);"><div class="header_title" style="display: flex; align-items: center; direction: ltr;"><span style="overflow: hidden; text-overflow: ellipsis; white-space: nowrap; display: block; padding-left: 4px;">System information</span></div></h2><div id="content" class="mdc-dialog__content" style="-webkit-font-smoothing: antialiased; font-family: Roboto, sans-serif; font-size: 14px; line-height: 21px; font-weight: 400; letter-spacing: 0.4375px; text-decoration: none; text-transform: none; flex-grow: 1; box-sizing: border-box; margin: 0px; overflow: auto; color: rgb(225, 225, 225); padding: 24px; position: relative; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; orphans: 2; text-align: start; text-indent: 0px; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal;"><slot id="contentSlot" style="margin-top: 0px; margin-bottom: 0px;"><div><div>
Version | core-2025.7.4
-- | --
Installation type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
Container architecture | aarch64
User | root
Virtual environment | false
Python version | 3.13.3
Operating system family | Linux
Operating system version | 6.12.25-haos-raspi
CPU architecture | aarch64
Timezone | Europe/Brussels
Configuration directory | /config

</div></div></slot></div>

sairon, 2025-08-06:
Can you post Host and Supervisor logs capturing the uninstall and reinstall process?

Mats-dev, 2025-08-07:
I directly connected the SSD to the Raspberry Pi 5 with no hub.

itayp, 2025-08-09:
I think there is some issue with HAOS16. I had issue first with Node Red, but as I am not really using I disregarded it. Then Zigbee2MQTT started having issue and when I tried to uninstall/reinstall the add on didn't start anymore.
Now it seems everything is broken (tried to upgrade to HAOS 16.01 RC).

alexdsds, 2025-08-11:
After upgrade to 16.0 HA do not boot