home-assistant/operating-system #4213: HA Yellow CM4 Lite to CM5 Lite upgrade doesn't work - no signs of life on known working hardware

cinderblock, 2025-08-05:
### Describe the issue you are experiencing

I have an existing CM4 Lite with NVMe on a HA Yellow. I'd like to upgrade it to a CM5 Lite.

The [instructions](https://support.nabucasa.com/hc/en-us/articles/25608059150237-Migrating-from-CM4-Lite-with-NVMe-to-CM5-Lite-with-NVMe) suggest it should be a seamless hardware swap and the software will just work.

When I try to boot, the red led turns on at startup, along with a brief flash on the blue led. 1~2 seconds later, the red turns off and the green turns on permanently. Device is not accessible on the network. Yellow LED never flashes.

Connecting to the UART via USB-C cable, which I open before connecting power to the HA Yellow, I see no serial output at 115200 baud (nor 9600).

The device is drawing ~2W during startup and then settles at ~3.5W (via 12V barrel). The same behavior happens when powered over PoE.

Trying on the same CM5/HAY hardware with a new empty NVMe, I can flash the latest HAOS without issue, via rpiboot. The serial terminal works fine for CM4 hardware or a fresh install on a new NVMe with CM5. So it's not a hardware issue.

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

16.0

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

HA Yellow with CM4/5 Lite. No USB devices. PoE with CLS4 jumper or 12V DC.

### Steps to reproduce the issue

1. Install HAOS on CM4 Lite with NVMe
2. Replace CM4 Lite with CM5 Lite
3. Boot

### Anything in the Supervisor logs that might be useful for us?

```txt
No. HA doesn't start.
```

### Anything in the Host logs that might be useful for us?

```txt
No. HA doesn't start.
```

### System information

## System Information

version | core-2025.7.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
container_arch | aarch64
user | root
virtualenv | false
python_version | 3.13.3
os_name | Linux
os_version | 6.12.25-haos-raspi
arch | aarch64
timezone | America/Los_Angeles
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 2122
Downloaded Repositories | 4

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 16.0
-- | --
update_channel | stable
supervisor_version | supervisor-2025.07.3
agent_version | 1.7.2
docker_version | 28.3.0
disk_total | 457.7 GB
disk_used | 15.9 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | yellow
supervisor_api | ok
version_api | ok
installed_addons | Studio Code Server (5.19.3), Samba share (12.5.2), Terminal & SSH (9.18.0), chrony (5.2.0), VLC (0.3.0), Piper (1.6.2), Log Viewer (0.17.1), Uptime Kuma (0.14.0), Squeezelite (0.0.21)

</details>

<details><summary>Dashboards</summary>

dashboards | 6
-- | --
resources | 1
views | 4
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), end0 (enabled, default, auto), hassio (disabled), docker0 (disabled), vethe7aa0d9 (disabled), vethcdf90f8 (disabled), vethb5d527e (disabled), vetha34bf62 (disabled), veth2431f9b (disabled), veth1441277 (disabled), veth7f78344 (disabled), vethd4faca1 (disabled), veth1dbd0f7 (disabled), veth7458259 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), end0 (10.255.0.9/20), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), vethe7aa0d9 (), vethcdf90f8 (), vethb5d527e (), vetha34bf62 (), veth2431f9b (), veth1441277 (), veth7f78344 (), vethd4faca1 (), veth1dbd0f7 (), veth7458259 ()
ipv6_addresses | lo (::1/128), end0 (fe80::edd7:4d2:bfc2:dc69/64), hassio (fe80::30b5:ffff:fe90:3fea/64), docker0 (fe80::a0d2:b0ff:fe29:34a1/64), vethe7aa0d9 (fe80::4cac:b4ff:fea9:7361/64), vethcdf90f8 (fe80::6:57ff:feba:759b/64), vethb5d527e (fe80::f85b:ceff:fe8f:e51e/64), vetha34bf62 (fe80::fc2f:6cff:fe75:4225/64), veth2431f9b (fe80::c8a0:bbff:fe72:494d/64), veth1441277 (fe80::9c00:f7ff:fe7a:5159/64), veth7f78344 (fe80::a09f:94ff:fef0:74fb/64), vethd4faca1 (fe80::3c76:16ff:fe78:cb71/64), veth1dbd0f7 (fe80::acbf:dcff:fea8:a1d8/64), veth7458259 (fe80::8cba:36ff:fe65:a658/64)
announce_addresses | 10.255.0.9, fe80::edd7:4d2:bfc2:dc69

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | July 22, 2025 at 1:04 PM
-- | --
current_recorder_run | August 5, 2025 at 3:31 PM
estimated_db_size | 430.93 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

### Additional information

My network reports the CM5 device is assigned an IP... so it's doing *something* on the network... but I cannot get a ping response.

I've got the CM4 booted now. I'm making fresh backups and getting ready to restore a fresh CM5 install from a backup. If that's my only option, that's fine. I've seen others with this issue so it seems like it would be worth figuring out why this simple upgrade doesn't "just work".

I suspect it's because we first setup HAOS on the CM4 before CM5 support had been added and there is some setting that isn't getting upgraded. I had a similar issue with regular Raspberry Pis where an SD card from a Pi4 wouldn't work in a Pi5 but a fresh image worked fine.

### See also

- https://community.home-assistant.io/t/home-assistant-yellow-gets-cm5-support-in-haos-14/800621/94
- https://discord.com/channels/330944238910963714/1393724744339161088
- https://github.com/home-assistant/operating-system/issues/3853
- https://github.com/home-assistant/operating-system/issues/4150#issuecomment-3054268455

sairon, 2025-08-06:
If you still haven't migrated, can you check the content of the boot partition on the NVMe, especially the `config.txt`? There were some migration steps done in the past to make it compatible, if you made any changes there in the past, the migrations might haven't been applied correctly.

cinderblock, 2025-08-06:
I have not made any manual changes to boot partition. I think I used rpiboot to flash the NVMe with HAOS via Raspberry Pi Imager ~1.5 years ago.

[image]

<details>
<summary><code>config.txt</code></summary>

```ini
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# HAOS - don't change it!
disable_splash=1
kernel=u-boot.bin
arm_64bit=1

# Enable primary UART by default
enable_uart=1

# No HDMI on Yellow, but we can't set to 16 since we need the full firmware
# for codecs
gpu_mem=32

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Uncomment this to enable GPIO support for RPI-RF-MOD/HM-MOD-RPI-PCB
#dtparam=i2c_arm=on
#dtoverlay=rpi-rf-mod

# Additional overlays and parameters are documented /boot/overlays/README

[cm4]
device_tree=bcm2711-rpi-cm4-ha-yellow.dtb

[cm5]
device_tree=bcm2712-rpi-cm5-ha-yellow.dtb

[all]
```

</details>
<details>
<summary><code>cmdline.txt</code></summary>

```txt
dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA2,115200n8
```
</details>

Bwooce, 2025-09-10:
I tried this a few months ago, and after booting with rpiboot and the recovery serial terminal discovered it didn't see the controller let alone the NVME, and so installing onto it was never going to work. Stable nvme from the known-working models, so it's hard to imagine it's hardware although I guess the CM is a large part of the hardware overall. My Yellow is pre-button, with the jumper, if that makes any difference.

This, like yours, is a working CM4 Lite install and I have the same symptoms with the LEDs. I've eventually given up (after swapping 4 times) and the CM5 is on an expensive shelf. So I guess this is contentless +1 from me as I don't have logs sadly, but I really would like to move from my CM4 Lite 2GB Wifi to my CM5 Lite 8GB Wifi...so tell me what other info I can provide and I'll try again. It is possible my CM5 is defective but...it booted ok otherwise, I reseated it 3x, and my motivation to buy either a carrier or another CM5 is currently low.

cinderblock, 2025-09-10:
@sairon I am still running my HAOS on CM4 with a Kingston SNV2S/500GB NVMe PCIe 4.0 SSD.

I also have a CM5 IO Board to test with. It boots the CM5 Lite from uSD without issue. It doesn't show any display output trying to boot from NVMe only.

After booting from uSD with NVMe installed, `lsblk` shows a working drive, which I can mount:
```log
pi@travelpi:~ $ sudo lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0  59.6G  0 disk
├─mmcblk0p1 179:1    0   512M  0 part /boot/firmware
└─mmcblk0p2 179:2    0  59.1G  0 part /
nvme0n1     259:0    0 465.8G  0 disk
├─nvme0n1p1 259:1    0    32M  0 part
├─nvme0n1p2 259:2    0    24M  0 part
├─nvme0n1p3 259:3    0   256M  0 part
├─nvme0n1p4 259:4    0    24M  0 part
├─nvme0n1p5 259:5    0   256M  0 part
├─nvme0n1p6 259:6    0     8M  0 part
├─nvme0n1p7 259:7    0    96M  0 part
└─nvme0n1p8 259:8    0 465.1G  0 part
pi@travelpi:~ $ sudo mount /dev/nvme0n1p8 /mnt/temp
pi@travelpi:~ $ ls -l /mnt/temp
total 2683796
drwx--x--- 13 root root       4096 Sep 10 11:40 docker
drwxr-sr-x  3 root kvm        4096 Apr  9  2024 logs
drwx------  2 root root      16384 Mar 13  2024 lost+found
drwxr-xr-x 17 root root       4096 Sep 10 11:41 supervisor
-rw-------  1 root root 2748174336 Oct  8  2024 swapfile
```

ly2xing, 2025-09-25:
same issue here, i got an early revision of yellow, running cm4 with nvme data drive, tried to upgrade to  cm5 lite, nothing, it boot loops, green yellow blink for a while, then red solid, then green solid then back to blinking... tried the same nvme on a new carrier board, HA yellow os won't boot, just uboot logo on top right for couple seconds then off and back on, rinse and repeat.

installed raspbian OS on the same nvme and using the same carrier board it boots fine, everything checks out, just HA Yellow OS won't boot