home-assistant/operating-system #4228: USB Zigbee stick failed to load after upgrade 16.0 -> 16.1

InuSasha, 2025-08-13:
### Describe the issue you are experiencing

After the upgrade from 16.0 to 16.1 the usb stick failed to initialize.

Reboot or unplug+plug-again does not solve the issue.
After the downgrade to 16.0, the usb stick is working, with these log entries.

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

16.1

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

USB-Stick: 
ID 1a86:7523 USB Serial
Zigbee 3.0 USB Dongle Plus ZigBee 3.x.0 TI CC2652P Coordinator CC2652 

### Steps to reproduce the issue

1. Upgrade to HassOS 16.1
2. Plug USB stick

### Anything in the Supervisor logs that might be useful for us?

```txt
not relevant, kernel issue :)
```

### Anything in the Host logs that might be useful for us?

```txt
Oct 08 15:42:32 homeassistant kernel: usbcore: registered new interface driver ch341
Oct 08 15:42:32 homeassistant kernel: usbserial: USB Serial support registered for ch341-uart
Oct 08 15:42:32 homeassistant kernel: ch341 1-1.2:1.0: ch341-uart converter detected
Oct 08 15:42:32 homeassistant kernel: usb 1-1.2: failed to receive control message: -32
Oct 08 15:42:32 homeassistant kernel: ch341-uart ttyUSB0: probe with driver ch341-uart failed with error -32
```

AlexNauta, 2025-08-13:
Same problem here with ZHA to a remote Zigbee coordinator (SLZB-06, ZNP through socket)
Worked like a charm until I updated to HassOS 16.1

InuSasha, 2025-08-13:
maybe the same issue: https://github.com/raspberrypi/linux/issues/6941

aperezva, 2025-08-14:
All yellow?

DjMayone, 2025-08-14:
It would look something like this: https://github.com/home-assistant/operating-system/issues/4223

mmarc, 2025-08-14:
HmIP-RFUSB also fails on Yelllow after upgrade to 16.1.
Downgrade to 16.0 fixes it again.

```Bus 001 Device 003: ID 1b1f:c020 Silicon Labs eQ-3 HmIP-RFUSB```

saschaabraham, 2025-08-14:
> > Same with a DeskPI Pro Case, the fan is controlled via addon and ttyUSB0. After a reboot the fan stays off.
> 
> A similar thing happened to me with a MiniPC (Acemagic T9 Plus) with LEDs on the case controlled by the serial port. A few months ago, after a HAOS update, the script control stopped working.

In your case only a led, in my case the risk of death ;-)

Fredyy90, 2025-08-14:
I had the same issue in my Proxmox VM, after the HAOS update my Connect ZBT-1 was gone and wouldn't come back even after multiple HA restarts, only after a complete reboot of the VM the Stick was detected again.

Tragen, 2025-08-14:
For me it's working. I always do complete reboots.

InuSasha, 2025-08-14:
@sairon: when you mean the ```dtoverlay=dwc2,dr_mode=host```.
Ich tried it, this morning. no effect, it is still broken

sairon, 2025-08-14:
@InuSasha No, there's actually this PR mentioned in the issue that's supposed to fix it: https://github.com/raspberrypi/linux/pull/6936

Trying that would mean to apply the patch and rebuild the kernel - not something you can easily do, but I can build in our CI and you can install the test build.

InuSasha, 2025-08-14:
@sairon: when you send me a patch, ich will try (but the yellow have to boot ;)).
please tell how i can install the new version. i do not know, how to install beta versions

RDF300, 2025-08-14:
Want to add here this is also affecting the desk pi pro Add-on / Its Fan control via USB serial 
https://github.com/adamoutler/HassOSDeskPiPro
My latest post  on https://community.home-assistant.io/t/deskpi-pro-active-cooling-and-front-usb-ports/265136
Reverting to 16.0 resolves the issue. 

Hardware : 
rpi4-64
Deskpi Pro Case 
Booting from SSD Via USB within the desk pi pro case

sairon, 2025-08-14:
I am able to reproduce the issue on Yellow and the patch indeed fixed it. The problems with DeskPi case are almost surely related and will be fixed by the same patch.

The workaround is to downgrade to previous version or 16.0 (i.e. by using `ha os boot-slot other` or `ha os update --version 16.0`).

RDF300, 2025-08-14:
> I am able to reproduce the issue on Yellow and the patch indeed fixed it. The problems with DeskPi case are almost surely related and will be fixed by the same patch.
>
> The workaround is to downgrade [...]

Yup in the end just used ha os boot-slot B in my case.  and all came back as expected. for now will skip 16.1 and lurk to confirm issue closed . Thank you :)

Tragen, 2025-08-14:
@b0ff3n 
In a Proxmox VM with Intel Xeon CPUs.
This war before the titel was renamed from
USB Zigbee stick failed to load after upgrade 16.0 -> 16.1
to
Some USB-Serial adapters broken on Raspberry Pi with OS 16.1

b0ff3n, 2025-08-14:
@Tragen 

I see.

Guess it's fairly widespread then, so I'll hold off on this release for now.

Thank you for replying.

tkusakab, 2025-08-15:
It is HA Yellow, but in my case, it started working with the following command without downgrading.
ha os boot-slot B

sairon, 2025-08-15:
Yesterday's build contains patches for the USB driver. To upgrade to it, run `ha os update --version 16.2.dev20250814`. No need to worry it's a dev version, it's almost identical to 16.1 except those fixes.

InuSasha, 2025-08-15:
@sairon: i can confirm, version "16.2.dev20250814" solved the issue

Hedda, 2025-08-22:
For the record, my friend have just yesterday seen a similar symptom on Home Assistant Blue (ODROID-N2+) with  Aeotec Z-Stick 7:

* https://github.com/home-assistant/core/issues/151024

Not sure if his problem was this exact same root cause as this but he is/was running Home Assistant Operating System 16.1 version.

StefanDierkes, 2025-08-30:
Same issue here with the following hardware:

- Home Assistant Yellow
- USB-Stick: CUL
- USB-Stick: HomeMatic

The Downgrade to HA OS 16.0 solved the problem.

commit 17d811a7 (2025-09-08): Backport patches for Raspberry Pi dwc2 driver (#4231)

Raspberry Pi Linux update to 6.12.34 broken some USB devices, mostly USB-Serial
converters connected to Yellow, but there are reports of some other peripherals
connected to RPi boards too.

This is a known RPi upstream issue [1] fixed by a PR [2] that's not been merged
to RPi stable kernel yet. Applying patches from this PR fixes the issues.

Fixes #4228, refs #4229

[1] https://github.com/raspberrypi/linux/issues/6941
[2] https://github.com/raspberrypi/linux/issues/6936