home-assistant/operating-system #4233: HM-MOD-RPI-PCB no longer working since 16.0

mimorocks, 2025-08-14:
### Describe the issue you are experiencing

Home Assistant Yellow:

Devices `raw-uart` / `eq3loop` vanish after upgrading from 15.2 -> 16.0

HM-MOD-RPI-PCB on internal GPIO no longer detected.

No connect to RF Hardware from RaspberryMatic Addon.

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

16.0

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Home Assistant Yellow
HM-MOD-RPI-PCB on internal GPIO

### Steps to reproduce the issue

1. Upgrade to Home Assistant OS 16.0
2. devices raw-uart / eq3loop vanish
3. no more RF found by RaspberryMatic
...

### Anything in the Supervisor logs that might be useful for us?

```txt
after update to 16.0:

2025-08-13 18:06:46.805 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/mmd_bidcos - None
2025-08-13 18:06:47.171 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/mmd_hmip - None
2025-08-13 18:06:48.824 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/eq3loop - None
```

### Anything in the Host logs that might be useful for us?

```txt
not really
```

### Additional information

after downgrade to **15.2** `eq3loop` and `/dev/raw-uart` are back

```
...
2025-08-14 18:11:44.661 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/raw-uart - None                       
2025-08-14 18:11:44.668 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/snd/timer - None                       
2025-08-14 18:11:44.733 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
2025-08-14 18:11:44.749 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/vc-mem - None
2025-08-14 18:11:44.915 INFO (MainThread) [supervisor.docker.addon] Added cgroup permissions 'c 238:0 rwm' for device /dev/raw-uart to addon_de838cd8_
2025-08-14 18:11:49.507 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_matter_server        
2025-08-14 18:11:50.806 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from ca5234a0_paperless-ngx         
2025-08-14 18:11:52.577 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from ca5234a0_paperless-ngx                
2025-08-14 18:11:53.155 INFO (MainThread) [supervisor.api.middleware.security] /network/interface/end0/info access from ca5234a0_paperless-ngx  
2025-08-14 18:12:04.889 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/eq3loop - None
2025-08-14 18:12:05.018 INFO (MainThread) [supervisor.docker.addon] Added cgroup permissions 'c 236:0 rwm' for device /dev/eq3loop to addon_de838cd8_r
2025-08-14 18:12:08.002 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/mmd_hmip - None                            
2025-08-14 18:12:08.005 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/mmd_bidcos - None                         
2025-08-14 18:12:08.157 INFO (MainThread) [supervisor.docker.addon] Added cgroup permissions 'c 236:1 rwm' for device /dev/mmd_hmip to addon_de838cd8_
2025-08-14 18:12:08.299 INFO (MainThread) [supervisor.docker.addon] Added cgroup permissions 'c 236:2 rwm' for device /dev/mmd_bidcos to addon_de838cd
2025-08-14 18:12:58.486 INFO (MainThread) [supervisor.core] Start Home Assistant Core
....                             
```

bilbo-b, 2025-08-17:
I have the same issue as I described here https://github.com/jens-maus/RaspberryMatic/discussions/3145
Downgrading to HAOS 15.2 made the HM-MOD-RPI-PCB work again (on GPIO of yellow with CM4. It did no longer work with HAOS 16.0, 16.1 and 16.2.dev20250814 (the last one fixed a maybe similar problem for HMIP-RFUSB and 16.1))

bilbo-b, 2025-08-19:
Thank you,

just let me know, which version to test exactly, please

bilbo-b, 2025-08-19:
works like a charm :) 

Thank you very much!

commit 419d4001 (2025-08-20): Fix missing rpi-rf-mod overlay on Yellow (#4244)

Before update to Buildroot 2025.02, the overlays directory on Yellow was
created by rpi-firmware in a condition added confusingly in firmware bump [1].
However, this got lost during Buildroot update, and since Yellow doesn't copy
overlays from the rpi-firmware repo, the directory was never created and the
rpi-rf-mod.dtbo couldn't be copied there in pre-image build hook.

To make things more robust, create the overlays directory for rpi targets
conditionally in the hook instead of relying on rpi-firmware to create it.

[1] https://github.com/hom [... cut]
closed 2025-08-20 by sairon